nodemcu/nodemcu-firmware #2978: After second reboot/reset modules lose their settings obtain by enduser_setup

Michal78S, 2019-12-10:
<------------------------ BUG REPORT -----------------------------------------
### Expected behavior
Setting obtain by Enduser_setup are keeping until wifi.sta.clearsetings() is executed or again Enduser_setup is called.

### Actual behavior
Always after second reset/voltage reebot settings are dropped.

### Test code
function init_wifi() -- ustaw i polacz sie z lokalnym routerem
    ssid, password = wifi.sta.getconfig()
    print(ssid)
    print(password)
    if ssid == "" then
        enduser_setup.start()
    else
        --wifi.setmode(wifi.STATION)
        wifi.sta.connect()
    end
end
### NodeMCU version
It occurs on 3.0.0.0 firmware. Version 2.1.0 is free of this lack (keeps settings). Moreover when I tried back to version 2.1.0 module constantly shows error: *** ERROR ***: unable to format. FS might be compromised.
Reflasing to the oldest firmware which I got helped to bring module back to work.

### Hardware
Tested on ESP12E. I have checked it in two different devices.

8<------------------------ END BUG REPORT -------------------------------------

Michal78S, 2020-01-20:
More details. I have builded another firmware:

> NodeMCU 3.0.0.0 built on nodemcu-build.com provided by frightanic.com
> 	branch: master
> 	commit: 310faf7fcc9130a296f7f17021d48c6d717f5fb6
> 	release: 3.0-master_20190907
> 	release DTS: 201909070945
> 	SSL: false
> 	build type: integer
> 	LFS: 0x0
> 	modules: adc,bme280,bme680,enduser_setup,file,gpio,i2c,net,node,tcs34725,tmr,uart,ucg,wifi,ws2812,ws2812_effects,xpt2046
>  build 2019-12-18 20:30 powered by Lua 5.1.4 on SDK 

Issue still existes. After second restart (or battery remove) firmware lose wifi settings. It is really anoing because require hardcode the settings or add and run enduser_setup each second restart. I think it depends on firmware revision. The 2.1.0 and 2.2.1 are free of this lack; wifi settings are kept no maters how many restarts are done. Probably enduse_setup module erases them? Will check soon.

rvalle, 2020-03-22:
@Michal78S did you made progress on this issue?
I am also experiencing wifi connection issues after setup.
As you say, after a restart the module will not connect to the Wifi.
However I don't think the settings are lost, as I can still see the eus_params.lua in the filesystem with the rightparams.

I am assuming that it is safe to include:
```
enduser_setup.start()
```
in init.lua and wifi will either start or the AP will be started for configuration. 
However the documentation is not clear about what is the intended lifecycle. I cannot find any examples of init.lua were the different cases are handled.

I assume that enduser_setup will attempt to connect with the saved parameters and if not successful will launch the setup portal.

Could this issue just be a timing problem? perhaps my access point is too slow for this?

rvalle, 2020-03-22:
I think the problem is related to calling enduser_setup.start() on an already configured wifi interface while the interface is connecting.

The easiest way to reproduce is to place enduser_setup.start() as first thing on init.lua.

Looks like enduser_setup.start() will gracefully stop when the interface gets its IP but for some reason the interface will not connect on next boot.

I can only reproduce this when using init.lua. If I try later in the interpreter it does not seem to happen.

rvalle, 2020-03-22:
Yes, this can be very easily reproduced with the following steps:

- Create a firmware on the builder: master v3 + wifi manager module
- Flash the ESP12f board
- upload an init.lua with one instruction: "enduser_setup.start()"
- restart the board
- setup wifi on phone
- ping board: works
- restart the board
- ping board: works
- restart the board
- ping board: does not work

I think the problem is that we are assuming that enduser_setup.start() will not break the configuration of an already configured board. but it seems like it does. I cannot find any guide on wifi or enduser_setup with a complete workflow.

Ideally enduser_setup.start() would be able to
- allow adding additional APs to an already configured board or
- wait before starting to see if the board is already configured, if configured just exit.

lorenz24, 2020-03-26:
I agree, enduser_setup.start() is not reliable at all.
Once you start the enduser_setup.start() and follow the instruction your ESP will be do crazy things with SSID and password until you do a cold start. 
My experience:
Setup the SSID and password with enduser_setup.start(), you get a IP and everything looks good.

> print(wifi.sta.getip())
192.168.1.167        255.255.255.0        192.168.1.1
> print(wifi.sta.getconfig()) 
NODEMCU        test-lua        0        f0:9f:c2:a7:c8:54

After one or more reset the setup gets lost. 

> print(wifi.sta.getconfig())
0        f0:9f:c2:a7:c8:54

Writing the SSID and password via wifi.sta.config(station_cfg) it works again. 

> wifi.setmode(wifi.STATION)
> wifi.sta.autoconnect(1)
> station_cfg={}
> station_cfg.ssid="NODEMCU"
> station_cfg.pwd="test-lua"
> station_cfg.save=true
> wifi.sta.config(station_cfg)

Some reset later you get this result:

> print(wifi.sta.getconfig())
0        ff:ff:ff:ff:ff:ff 

OR this result

> print(wifi.sta.getconfig())
0        f0:9f:c2:a7:c8:54

Only a cold start of the module and the SSID setup via wifi.sta.config(station_cfg) helps to get everything to normal. But as soon as you start enduser_setup.start() again the problem starts again.

My conclusion:
run enduser_setup.start() to get the file “eus_params.lua” . 
Cold start the Module and do not start anymore enduser_setup.start() but read the file “eus_params.lua”, generated from enduser_setup.start() , and use the date from this file to start the wifi.sta.config(station_cfg).

I hope there is time left from the expert to solve this problem. I guess there is a pointer going the wrong way in the “enduser_setup.start()” C module.

tanner-rutgers, 2020-07-28:
Also seeing the same issue after second or third power cycle.

```
NodeMCU 3.0.0.0 built on nodemcu-build.com provided by frightanic.com
	branch: master
	commit: 8d091c476edf6ae2977a5f2a74bf5824d07d6183
	release: 3.0-master_20200610
	release DTS: 202006092026
	SSL: false
	build type: float
	LFS: 0x0 bytes total capacity
	modules: enduser_setup,file,gpio,mqtt,node,sjson,tmr,uart,wifi,ws2812,ws2812_effects
 build 2020-07-27 01:56 powered by Lua 5.1.4 on SDK 3.0.1-dev(fce080e)
```

Michal78S, 2021-01-07:
Is there any solution for this issue after over a year? How to avoid lost of settings?

vsky279, 2021-01-07:
I am saving wifi credentials in SPIFFS. I can store several aps parameters. It's a workaround but better than nothing.

closed 2022-04-30 by stale[bot]