nodemcu/nodemcu-firmware #2981: Two Different RSSI  from wifi.sta.getap() & wifi.sta.getrssi()

chathurangawijetunge, 2019-12-21:
NodeMCU 3.0.0.0 built on nodemcu-build.com provided by frightanic.com
	branch: master
	commit: 310faf7fcc9130a296f7f17021d48c6d717f5fb6
	release: 3.0-master_20190907
	release DTS: 201909070945
	SSL: false
	build type: integer
	LFS: 0x20000
	modules: file,gpio,mqtt,net,node,rtctime,sntp,tmr,wifi
 build 2019-12-21 01:37 powered by Lua 5.1.4 on SDK 3.0.1-dev(fce080e)
---------------
ESP8266

i get two different RSSI's by wifi.sta.getap() & wifi.sta.getrssi()

wifi.sta.getap(1,function(t)    
      for bssid,v in pairs(t) do
         print("wifi.sta.getap()",bssid,v)
         print("wifi.sta.getrssi()",wifi.sta.getrssi())       
      end
 end)    

out puts 

wifi.sta.getap()	34:e8:94:04:7a:a0	SCW_Wireless,-56,3,10
wifi.sta.getrssi()	-62
wifi.sta.getap()	d8:d8:66:0b:8c:4a	Dialog 4G 146,-88,3,6
wifi.sta.getrssi()	-62

from wifi.sta.getap()= -56
from wifi.sta.getrssi()= -62

HHHartmann, 2019-12-27:
I can conform this issue.
Although the values fluctuate I only had one out of about 20 tries where the values were the same, in all other cases the value from `wifi.sta.getrssi()` was 3 to 10 worse.

chathurangawijetunge, 2019-12-28:
yes the value from wifi.sta.getrssi() is less then wifi.sta.getap() so which one is the correct...
why ant we getting the same from both methods

closed 2021-01-09 by stale[bot]