nodemcu/nodemcu-firmware #2982: net UDP socket can send a very small string

alexandruantochi, 2019-12-22:
### Expected behavior
UDP datagram has a theoretical limit of ~65,507 bytes. 

`net:socket` can not send more than around 780 characters. Event registers as sent, but no data is sent. (sck:on('sent', function() print('sent') end);

### Actual behavior
Event should not register as 'sent'. The limitation should be written in the doc if not possible to increase/fix. 

### Test code
```
someJson = {}
someJson.item = {}
someJson.item.first = 'first'
someJson.item.second = 'second'
someJson.item.third = 'third'
test = {};
port = 5000
ip = '192.168.0.185'

for i=0,10,1 do 
    test[i] = someJson
end

sck:send(port, ip, sjson.encode(test)) 
```

The code works if i <=10, does not work for i >= 20 (i.e 1400 chars or <2kB);

Tested on both Win device with UDP Sender/Receiver (Win App Store) and Android device with different UDP app.

### NodeMCU version
3.0-master_20190907

### Hardware
 Arch: esp8266 | Version: 3.0.0 | ChipID: 0x8142d8 | FlashID: 0x1625c2

HHHartmann, 2020-12-19:
this should at least be documented.

HHHartmann, 2020-12-22:
Reading the code I found that app\include\lwipopts.h has the following defines
```C
#define IP_REASSEMBLY                   0
#define IP_FRAG                         0
```

which from reading the associated comments turn off fragmentation support, so MTU is the max datagram size then.

Changing them to 1 would enable fragmentation support but given the very limited memory resources of the NodeMCU this might not be a reasonable way to go.

So the way would be to document this.

closed 2022-04-30 by stale[bot]