nodemcu/nodemcu-firmware #2989: Feature request: Machine-readable description of build config options

mk-pmb, 2019-12-31:
### Missing feature
Let's make the comments in `app/include/user_*.h` more machine-readable:
* macro name
* default value
* category
* importance level or sth.
* short description
* (optional) detailed description
* Possible values
  * (optional) short description
  * (optional) detailed description

In case we already have such info, let's put a hint where to find it on top of the header files. And establish a mechanism for keeping them in sync.

### Justification
I'd like to make a config wizard. That wizard, at a minimum, should be able to determine whether it's understanding of available options is still compatible with the currently checked-out version of the firmware. Would be nice if it could also fall back to the maintainer's help texts if it cannot find localized/optimized help texts from a user-preferred source.

### Workarounds
If it turns out we don't have such info, I might have to bake my own.

mk-pmb, 2020-01-01:
Progess: I compiled short descriptions for all documented modules. Preparations for categorization.

HHHartmann, 2020-01-01:
That sounds like a good idea. There is also a possible related issue #386 .
Please also note that in the near future configuration will be changed for esp8266 to work as in esp32, So editing the user_config will be history then.

mk-pmb, 2020-01-01:
Can you/someone estimate how far away that future is?

marcelstoer, 2020-01-01:
> in the near future configuration will be changed for esp8266 to work as in esp32

Cool, who is working on that? It's the first time I hear this. I was also going to propose to @mk-pmb that before he reinvents the wheel he should look into what ESP-IDF does with Kconfig.

HHHartmann, 2020-01-02:
I tried to find where I read it but can't. Seems that I didn't read close enough. It is between https://github.com/nodemcu/nodemcu-firmware/issues/1657#issuecomment-441338203 and a testrun to integrate and build the two variants in one repo by someone (don't remember who) that I got that impression.
So sorry for the noise.

TerryE, 2020-01-31:
> > in the near future configuration will be changed for esp8266 to work as in esp32
> 
> Cool, who is working on that?

This came up in some discussion between Johny and me, but ATM @jmattsson has even  less bandwidth on NodeMCU than I do.

mk-pmb, 2021-01-31:
Any news on this?

jmattsson, 2021-01-31:
It's still on my todo list, just like last year >.<

mk-pmb, 2022-04-16:
Is anyone still working on this, or has it on their todo?