nodemcu/nodemcu-firmware #2997: Compilation fails with //#define I2C_MASTER_OLD_VERSION

KT819GM, 2020-01-03:
### Expected behavior
Compiling with ``` #define I2C_MASTER_OLD_VERSION```  commented out should enable usage of new I2C driver.
### Actual behavior
Code does not compile with error thrown:
```C
CC app/driver/i2c_master.c
i2c_master.c: In function 'i2c_master_setDC':
i2c_master.c:176:62: error: value computed is not used [-Werror=unused-value]
             READ_PERI_REG(RTC_GPIO_IN_DATA) & 1; asm volatile("nop;nop;nop;nop;");
                                                              ^
cc1: all warnings being treated as errors
../../Makefile:422: recipe for target '.output/eagle/debug/obj/i2c_master.o' failed
make[2]: *** [.output/eagle/debug/obj/i2c_master.o] Error 1
make[2]: Leaving directory '/opt/nodemcu-firmware/app/driver'
../Makefile:380: recipe for target '.subdirs' failed
make[1]: *** [.subdirs] Error 2
make[1]: Leaving directory '/opt/nodemcu-firmware/app'
Makefile:380: recipe for target '.subdirs' failed
make: *** [.subdirs] Error 2
make: Leaving directory '/opt/nodemcu-firmware'
```
### Test code
``` //#define I2C_MASTER_OLD_VERSION``` on `user_config.h`

### NodeMCU version
commit 8c08e7648d5a0e98e3e293e36ec7e73b31958874 (HEAD -> dev, origin/dev)

### Hardware
docker nodemcu-build

marcelstoer, 2020-01-03:
@sonaux can we ask you to please take a look at this, thanks.

sonaux, 2020-01-06:
I tried some tricks like casting to void to satisfy the compiler, but new xtensa-gcc optimises this statement out entirely. Also, clock stretching on GPIO16 without open drain mode will not work either (shortcircuiting this pin gives 40mA current on multimeter ) and I think its better to remove it and correct timing constants.
I will create PR bit later.

TerryE, 2020-01-18:
@sonaux, Natalia, thanks for replying on this.  GCC optimizes this out except when `I2C_MASTER_GPIO16_ENABLE` is defined, but it still does semantic checks such as `unused-value` even if the code is later optimised out.  With this enabled and a `(void)( ...)` check to force evaluation of the expression, then it does do a read access to the `RTC_GPIO_IN_DATA` I/O port at `0x6000078C`, etc. but the `& 1` is optimised away, which is only 1 dropped opcode.  I can't comment on the validity of the timing in this GP16 case.

A quick fix for @KT819GM in the meantime is to change the line to
```C
 (void) (READ_PERI_REG(RTC_GPIO_IN_DATA)); asm volatile("nop;nop;nop;nop;");
```
which will compile clean.

KT819GM, 2020-01-20:
@TerryE proposed fix did compile clean. As GPIO16 is not important to me, I will keep it as a solution. Thank you

KT819GM, 2020-07-17:
Somehow I'm missed fix (maybe was done on some bigger patch?), but at least on 5.3 bug does not exist and I'm using i2c devices without any issues.

closed 2020-07-17 by KT819GM