nodemcu/nodemcu-firmware #3000: make[1]: *** No rule to make target 'oldconfig'. Stop.

mk-pmb, 2020-01-03:
### Expected behavior
There's a documented way to build the ESP32 firmware non-interactively.

### Actual behavior
`make` fails with

```text
Console input/output is redirected. Run 'make oldconfig' to update configuration.

make[1]: *** No rule to make target '/opt/nodemcu-firmware/build/include/config/auto.conf', needed by '/opt/nodemcu-firmware/build/bootloader/bootloader.bin'.  Stop.
make[1]: Leaving directory '/opt/nodemcu-firmware'
make: *** [all] Error 2
```

so I try `make oldconfig`, but same problem, ironically:

```text
Console input/output is redirected. Run 'make oldconfig' to update configuration.

make[1]: *** No rule to make target 'oldconfig'.  Stop.
make[1]: Leaving directory '/opt/nodemcu-firmware'
make: *** [oldconfig] Error 2
Makefile:18: recipe for target 'oldconfig' failed
```

Fuller log: https://gist.github.com/mk-pmb/78eeb024d2fca9e8b9a8e3bd2c652cca

### Test code
https://github.com/mk-pmb/nodemcu-firmware-daily-vanilla/runs/372781196

### NodeMCU version
branch dev-esp32, commit bfd2259

### Hardware
The Github cloud

marcelstoer, 2020-01-03:
> There's a documented way to build the ESP32 firmware non-interactively.

I don't know about "documented" but it should be fairly simple

```
$ git clone --branch dev-esp32 --recurse-submodules https://github.com/nodemcu/nodemcu-firmware.git nodemcu-firmware-esp32
$ cd nodemcu-firmware-esp32
$ cp sdkconfig.defaults sdkconfig.old
$ cp my-esp-idf-config-file-I-got-from-wherever sdkconfig
$ python -m pip install --user -r sdk/esp32-esp-idf/requirements.txt #do this once at some point
$ make
```

(This is probably what my cloud builder would do if it offered ESP32 support, mapping from the simply user config to `my-esp-idf-config-file-I-got-from-wherever` yet to be implemented)

mk-pmb, 2020-01-03:
Thanks! I've no idea what `cp sdkconfig.defaults sdkconfig.old` would be good for, i.e. what the SDK would use `sdkconfig.old` for when there's a (ostensibly more current) `sdkconfig` file – but I'll try anyway.

mk-pmb, 2020-01-04:
Tried again, with sdkconfig.defaults copied as both sdkconfig.defaults and sdkconfig.old, also pip run manually instead of via /opt/build. Still no luck. :-( Relevant parts of the log: https://gist.github.com/mk-pmb/370081f91a6ac7600416911af32edacc

mk-pmb, 2020-01-04:
Solved. The error message in the ESP-IDF seems to be misleading (outdated?). `make defconfig` seems to be what I was looking for.

closed 2020-01-04 by mk-pmb