nodemcu/nodemcu-firmware #3003: ESP32: add_rodata_ld.sh not found

artem-dudarev, 2020-01-04:
### Actual behavior
Build fails in Docker:
```
Generating esp32.project.ld
/bin/sh: 1: /opt/nodemcu-firmware/components/base_nodemcu//add_rodata_ld.sh: not found
/opt/nodemcu-firmware/components/base_nodemcu/Makefile.projbuild:13: recipe for target '/opt/nodemcu-firmware/build/base_nodemcu/ld_patched' failed
make[1]: *** [/opt/nodemcu-firmware/build/base_nodemcu/ld_patched] Error 127
make[1]: Leaving directory '/opt/nodemcu-firmware'
Makefile:18: recipe for target 'all' failed
make: *** [all] Error 2
```
```
 Directory of D:\Home\ESP32\nodemcu-firmware\components\base_nodemcu

04.01.2020  12:22    <DIR>          .
04.01.2020  12:22    <DIR>          ..
04.01.2020  01:48               575 add_rodata_ld.sh
04.01.2020  01:48                 0 component.mk
04.01.2020  12:22    <DIR>          include
04.01.2020  01:48             2 347 ip_fmt.c
04.01.2020  01:48               689 Kconfig
04.01.2020  12:22    <DIR>          ld
04.01.2020  01:48             1 845 lextra.c
04.01.2020  01:48             4 342 linit.c
04.01.2020  01:48             6 165 lnodeaux.c
04.01.2020  13:49               519 Makefile.projbuild
04.01.2020  01:48             8 711 uart.c
04.01.2020  01:48             4 831 user_main.c
              10 File(s)         30 024 bytes
               4 Dir(s)  98 434 908 160 bytes free
```
Since the file is there, I thought it's because of the double slash in the path `base_nodemcu//add_rodata_ld.sh` and I wrapped `BASE_NODEMCU_DIR` with `$(abspath)` in the `Makefile.projbuild`:
```
BASE_NODEMCU_DIR:=$(abspath $(dir $(lastword $(MAKEFILE_LIST))))
```
Double slash has gone but the build is still failing:
```
Generating esp32.project.ld
/bin/sh: 1: /opt/nodemcu-firmware/components/base_nodemcu/add_rodata_ld.sh: not found
/opt/nodemcu-firmware/components/base_nodemcu/Makefile.projbuild:13: recipe for target '/opt/nodemcu-firmware/build/base_nodemcu/ld_patched' failed
make[1]: *** [/opt/nodemcu-firmware/build/base_nodemcu/ld_patched] Error 127
make[1]: Leaving directory '/opt/nodemcu-firmware'
Makefile:18: recipe for target 'all' failed
make: *** [all] Error 2
```

### Versions
Windows 10 Enterprise 1909 x64
Docker desktop: 2.1.0.5 (40693)
Git branch: dev-esp32
Latest commit: bfd22591

marcelstoer, 2020-01-04:
With near certainty this should go to a different issue list - just like #3001.

However, as I don't have access to a Windows box atm I can't be sure. @HHHartmann aren't you on Windows?

xvipservice, 2020-01-16:
我知道这个问题，我遇到过，其实是这个文件的编码问题，需要转成unix编码就可以了

xvipservice, 2020-01-16:
dos2unix /opt/nodemcu-firmware/components/base_nodemcu//add_rodata_ld.sh

closed 2021-01-24 by stale[bot]