nodemcu/nodemcu-firmware #3004: espconn lwip wrappers actively harmful

nwf, 2020-01-05:
As far as I can tell, anyway, these things are just obstruction, rather than abstraction, over lwip.  See, for example, the damage they cause with #2987: it is neither obvious nor documented that `espconn_gethostbyname` passes the espconn structure as the callback argument, and even if it were, it's also not obvious that this could be cast back to the `lmqtt_userdata` structure of which it is the first member.  We should just be calling `dns_gethostbyname` ourselves and passing the appropriate lua object to the callback.

Am I missing something and espconn is actually useful?  If not, I think I will propose that we attempt to excise it from our tree.

nwf, 2021-01-31:
Still true, sadly.

HHHartmann, 2022-04-19:
Is it still true?