nodemcu/nodemcu-firmware #3007: Memory still allocated after package.loaded set to nil, following ds18b20 example

nedoskiv, 2020-01-09:
Hello

It was been a while since I was using wonderful nodemcu firmware and seem lot of stuff changed, some modules moded from C to lua, like DS18B20 for example. So I compiled one wire support from main branch, downloaded ds18b20.lua and follow example in DS18B20 documentation section:

`local t = require("ds18b20")
local pin = 3 -- gpio0 = 3, gpio2 = 4

local function readout(temp)
  if t.sens then
    print("Total number of DS18B20 sensors: ".. #t.sens)
    for i, s in ipairs(t.sens) do
      print(string.format("  sensor #%d address: %s%s",  i, ('%02X:%02X:%02X:%02X:%02X:%02X:%02X:%02X'):format(s:byte(1,8)), s:byte(9) == 1 and " (parasite)" or ""))
    end
  end
  for addr, temp in pairs(temp) do
    print(string.format("Sensor %s: %s °C", ('%02X:%02X:%02X:%02X:%02X:%02X:%02X:%02X'):format(addr:byte(1,8)), temp))
  end

  -- Module can be released when it is no longer needed
  t = nil
  package.loaded["ds18b20"] = nil
end

t:read_temp(readout, pin, t.C)`

It works well, expect that around 14kb of free heap is gone, if I execute it again it do not allocate additional heap, but still looks like code is in memory somewhere and not released.

Did I miss something?

vsky279, 2020-01-09:
The module when loaded creates a variable `ds18b20` with reference to the module itself (code: `_G[modname or 'ds18b20'] = M`).
If you add `ds18b20=nil` after `package.loaded["ds18b20"] = nil` you will get your memory back.

Although I don't know exactly why we need to create such a variable.

nedoskiv, 2020-01-10:
Thank you, it is much better that way, only 2kb heap are lost (allocated for something)

closed 2020-01-13 by marcelstoer

TerryE, 2020-01-26:
> only 2kb heap are lost

I think that you will find that the Lua registry system doesn't shrink the number of slots, so not all of these resources are reclaimed by Lua GC.

nedoskiv, 2020-01-26:
Well i haven't use nodemcu for a while, lot of stuff changed. Not sure I can catch up now :) , but still it is one of the best choices for esp8266