nodemcu/nodemcu-firmware #3009: DS18B20.lua fail from time to time

nedoskiv, 2020-01-14:
I got 2 files executed each after other from a timer, every 5 seconds:

`local t = require("ds18b20")
local pin = s.ow1 -- gpio0 = 3, gpio2 = 4

local function readout(temp)
	local a={}
	if t.sens
		then
			a.ts=#t.sens
			for addr, temp in pairs(temp)
				do
					a[string.format("%s", ('%02X%02X%02X%02X%02X%02X%02X%02X'):format(addr:byte(1,8)))]=temp
			end
		else
			a.ts=0
	  -- Module can be released when it is no longer needed
	end
	  t = nil
	  package.loaded["ds18b20"] = nil
	  ds18b20=nil
	  ok, ow1 = pcall(sjson.encode, a)
	  ok=nil
end
t:read_temp(readout, pin, t.C)`

other file is similar, except that pin is different. It works fine except that from time to time I got following error:

PANIC: unprotected error in call to Lua API (ds18b20.lua:112: bad argument #2 to '?' (invalid function))

nedoskiv, 2020-01-14:
Thinking of it increase timer duration to 10 seconds to see if it fail again from time to time, perhaps sometime reading ow bus take too long and new execution come and something go wrong. Will leave it this night to see if fail again.

KT819GM, 2020-01-14:
Is this the only code you are running? ow is slow but definitely not that slow. I'm using this module for quite some time together with I<sup>2</sup>C devices querying it like every 2.5 secs without issues. Maybe @TerryE will take his eye on exact error you are getting.

vsky279, 2020-01-14:
I'm running your code for more than 70 minutes, every 5 seconds without any memory leak. I don't get the error you state. I cannot reproduce it.

nedoskiv, 2020-01-15:
That was the problem, looks like from time to time scanning ow bus take too long

closed 2020-01-15 by nedoskiv