nodemcu/nodemcu-firmware #3011: ESP32 I2C hardware timeout due to clock stretching

jmdasnoy, 2020-01-17:
### Summary
I have experienced I2C hardware timeouts due to clock stretching
I have solved the issue and thought I might share some insights with the community.

The title would spoil any attempt to write this as a high-suspense investigation report so here are the essentials.
### Situation: 
ESP-32, with the latest, standard NodeMCU firmware connected by I2C to a Bosch BNO055 IMU.
I am using the i2c module with the `i2c.SW` interface (software based bitbanging, master mode only, synchronous operation).

It's convenient for validation and testing of my I2C commands and making sure I read the BNO055 datasheet correctly :-)
With a I2C bus speed set to `i2c.SLOW` (100 kHz), the ESP32 and the BNO055 are communicating happily and reliably.

### Expectation:
The I2C bus is slow and ties down the CPU in synchronous mode.
So why not delegate the work to one of the ESP32 hardware I2C subsystems ?
This should be simple, specify `i2c.HW0/1` in `i2c.setup,` add the appropriate `i2c.transfer` call with a call back function to handle the results of the I2C operation whenever they are available.
### Issue:
It just doesn't work.
The call back function returns nonsense/random data and a false ACK flag.

Using `i2c.transfer` without a call back doesn't work better, it just fails with a Lua error and a timeout message.
That's surprising because I call `i2c.transfer` without a timeout parameter so it should default to infinite...
### Diagnostic:
The ESP32 hardware I2C interfaces have a timeout parameter and that's a good feature for reacting to I2C bus faults.
The ESP-SDK driver sets this to a default value corresponding to 8 I2C SCL clock cycle lengths.
More precisely, with a 80 Mhz ESP clock and a 100KHz I2C bus speed, this value is 8*80/0.1=6400 ESP clock cycles, i.e. 80 µS.

The BNO055 is a busy chip and can ask to slow down I2C communication by using clock stretching.
There is no documented upper bound to this so it could cause a timeout on the ESP.

You can set any I2C bus speed via `i2c.setup` and you don't have to stick to the predefined values of `i2c.SLOW` etc
A slower I2C bus speed means a higher timeout value.
In my case, an I2C bus speed of 30000 (30 Khz) does the trick and I am getting reliable I2C interaction again.
However, checking the ACK flag of the call back function is still highly recommended !
### Workaround:
Leave the firmware as it is and possibly add a cautionary note in the documentation of the i2c module.
I can do that, but read further...

### Suggestions:
#### Add a parameter to `i2c.setup` to modify the default timeout value
I find it a pity to have to use a slower I2C bus speed, just because one I2C peripheral chip may need some extra time to return results.
I would rather keep the higher bus speed and adapt the timeout value.

I have extended the `i2c.setup` function to accept a fifth, optional parameter as a (integer) timeout multiplier.
I had to reduce the bus speed by three previously to get reliable operation, so now a x4 timeout multiplier should be enough.

I can now call
`i2c.setup( i2c.HW0, pinSDA, pinSCL, i2c.SLOW, 4 )`
The timeout value is now 25 600 CPU clock cycles and I get reliable operations !

#### Adapt `i2c.setup` to return a value
By the way, `i2c.setup` does not return any value, contrary what the documentation states.
I have also adapted this function to return the I2C bus speed for the `i2c.SW` interface or the timeout value for the `i2c.HW0/1` interfaces.

#### Adapt `i2c.read` to standard I2C multibyte read practice
 By the way also, standard I2C practice for a multi byte read is to have the master (receiver) ACK'ing each byte except the last.
The ESP-SDK does this but the the glue routines from lua double-guess this with the end result of ACK'ing the last byte.
Not really according to specifications, but it still works, most likely because the I2C slave will stop sending data after seeing an bus STOP condition.

For compliance's sake, I have also adapted the `i2c.read` function to that effect.
### Conclusion
I hope this comments can help any others with similar issues.
If it is of interest to the community, I can contribute my code and the updated documentation, via, I suppose, a GitHub fork, adapt and pull request.

Let me know what you think of it.

nwf, 2021-01-14:
I seem to have missed this the first time around.  I think this is a sensible change to the i2c API and would like to see it happen.

The ESP8266 also has a hardware I2C peripheral AFAICT; do similar concerns exist over there, not just on the ESP32?

jmdasnoy, 2021-01-14:
@nwf thanks for your encouraging comments. I will prepare a PR to this effect.
I understand that this issue only impacts the ESP32.

The documentation for the 8266 i2c states in v:release

> ESP8266 chip does not have hardware I²C, so module uses software I²C driver.

> This module supports: ...- Clock stretching (slow slave device can tell the master to wait)

PR #3377 (2021-01-22): Fix ESP32 i2c timeout
Fixes #3011
- [x] This PR is for the `dev-esp32` branch rather than for the `release` branch.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [x] I have thoroughly tested my contribution.
- [x] The code changes are reflected in the documentation at `docs/*`.

Fix issues as described in issue 3011
Rewrote documentation with tested real-life examples
closed 2021-01-22 by marcelstoer