nodemcu/nodemcu-firmware #3017: unable to connect to 802.11n only ap

532910, 2020-01-25:
1. I've just switched my APs to 802.11n only mode. (`option require_mode 'n'` in openwrt)
2. An old NodeMCU 2.2.0.0 works fine.
3. Fresh NodeMCU 3.0.0.0 unables to connect. AP says: `Station does not support mandatory HT PHY - reject association` enven with `wifi.setphymode(wifi.PHYMODE_N)`

532910, 2020-01-25:
It's due to espressif / ESP8266_NONOS_SDK
v3.0.2 works fine
please update

nwf, 2020-02-02:
You are welcome to open a PR to bump our sdk version.

marcelstoer, 2020-02-03:
In which case #3008 would become obsolete.

532910, 2020-02-09:
> You are welcome to open a PR to bump our sdk version.

As soon as I'll understand what was technically wrong with  #3026

532910, 2020-02-09:
Which part of

```
ifeq ("$(RELEASE)","latest-3.0")
  SDK_VER        := 3.0.0
  SDK_FILE_SHA1  := NA
  SDK_ZIP_ROOT   := ESP8266_NONOS_SDK-release-v$(SDK_VER)
  SDK_FILE_VER   := release/v$(SDK_VER)
else ifeq ("$(RELEASE)","master")
  SDK_VER        := master
  SDK_FILE_SHA1  := NA
  SDK_ZIP_ROOT   := ESP8266_NONOS_SDK-$(SDK_VER)
  SDK_FILE_VER   := $(SDK_VER)
else
# SDK_VER        := 3.0
# SDK_FILE_VER   := v$(SDK_VER)
  SDK_FILE_VER   := e4434aa730e78c63040ace360493aef420ec267c
  SDK_VER        := 3.0-e4434aa
  SDK_FILE_SHA1  := ac6528a6a206d3d4c220e4035ced423eb314cfbf
  SDK_ZIP_ROOT   := ESP8266_NONOS_SDK-$(SDK_FILE_VER)
endif
```

should be updated, the first or the last one?

closed 2021-03-19 by stale[bot]