nodemcu/nodemcu-firmware #3020: tmr:start() does not restarts

532910, 2020-01-28:
### Expected behavior
https://nodemcu.readthedocs.io/en/master/modules/tmr/#tobjstart
says that `tmr:start()` **restarts** timer. It's not true:

### Actual behavior
`tmr:start()` will **restart** timer (start it count from zero)

### Test code
```Lua
> t = tmr.create()
> t:register( 3000, tmr.ALARM_SEMI, function( t ) print( 'alarm' ) end)
> t:start()
> -- wait 2 seconds
> t:start()
> -- alarm message will be printed 3 seconds after the first t:start() call, not the second one as expected.
```
### NodeMCU startup banner
NodeMCU 3.0.1.0, build type: integer
modules: file,gpio,net,node,rtctime,sntp,tmr,uart,wifi
build 2020-01-27 17:39 powered by Lua 5.1.4 on SDK 3.0.2(824dc80)

### Hardware
ESP8266 wemos D1 (not mini)

HHHartmann, 2020-01-29:
This seems to be an unclear documentation. 
The second start in your example should return false as the timer is already running.
To achieve what you want you will need to stop the timer, and then start it again.

532910, 2020-01-30:
So:
1. documentation should be corrected
2. tmr:restart() should be added

If there are no objections, I'm ready to PR. The question is: should it be two different pull requests or the one common will be fine?

532910, 2020-01-31:
@marcelstoer, @TerryE can I make one single PR for the two points above or it should be two separate pull requests?

TerryE, 2020-01-31:
@532910, I that both you and Gregor make valid points on this one   We should thing about valid use cases where this should apply, for example if the application is resetting a watchdog.  Do we insist on stopping the old timer first?  Whatever, the code should not fail silently.  It should return an error if the usage in incorrect.

Let's agree this first.

HHHartmann, 2020-01-31:
I could also imagine to change tmr.stop to restart the timer if it is already running. But that might break existing code.

532910, 2020-01-31:
you mean tmr.start, right?

HHHartmann, 2020-02-01:
Yes. Must have been still half asleep

fikin, 2020-02-01:
> documentation should be corrected
> tmr:restart() should be added
+1

imho clear api, clearer documentation and no visible backward compatibility risk.

HHHartmann, 2020-02-01:
So should we go for tmr.restart() and better documentation of tmr.start() then?

TerryE, 2020-02-01:
Or even one line Lua function to do the same stop / start, so I do think that adding an extra function to avoid this is a bit OTT.

Whatever, the code does need changing to return a correct error code.

HHHartmann, 2020-02-06:
Start already returns a boolean to indicate an error. At least according to the documentation. So testing if  it works as documented and clarification in the documentation should suffice here.

532910, 2020-02-09:
I'm for bringing `start` to its documentation: restart timer if it's already started.

marcelstoer, 2020-03-22:
@HHHartmann Is this still on your list?

HHHartmann, 2020-04-10:
I'v been thinking of a good solution here to satisfy @532910 s needs and not to break existing code.

If we use an optional parameter for `tmr:start(bool restart)` which would be `false` by default. That will not blow up the interface (too much).
Also the code should return `false` if false/no_param is give and the timer is already running to match the documentation.

One PR should be fine for code and documentation.

532910, 2020-04-10:
sounds good

PR #3111 (2020-08-29): Implementing optional parameter `restart` to tmr:start()
Fixes #3020.

- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [x] I have thoroughly tested my contribution.
- [x] The code changes are reflected in the documentation at `docs/*`.

Implements the solution discussed in #3020 as wrapped up by @HHHartmann 

Also removes unecessary check of lua_newuserdata result (also in ads1115 module).
closed 2020-08-29 by marcelstoer