nodemcu/nodemcu-firmware #3021: disable startup banner

532910, 2020-01-28:
Please add support for disable startup banner

### Workarounds
comment out `print_version(L);` in `app/lua/lua.c`

marcelstoer, 2020-01-29:
Why would you want to disable it? Feel free to create a PR.

TerryE, 2020-01-29:
The ROM bootloader generates a lot of output at 74,880 baud which looks like 1.2 Kb of crap at 115,200 baud, followed by the ~350 bytes of the extended version message, plus info messages on whether the LFS region is loaded or not.

The info message can be a little distracting, but given that the bootloader output is still going to be output to UART1, then suppressing it is only of minor benefit.  The main advantage of keeping it is that we created this extended info because this information is really helpful for pinning down bugreps and bug reporters rarely included this before we had it.  Perhaps could have a `node.textinfo()`or the like which generated this string on demand instead, but even so, I am not sure this is worthwhile.

KT819GM, 2020-01-30:
Wanted to write similar comment like @TerryE about bootloader output if the case is to keep silent boot because of some parsing on the other end of transmission. But maybe it's for an additional level of security? if so, it could be useful for production devices. Some ```#ifdef``` action could be applied on user_config.h to disable any version info?

marcelstoer, 2020-01-30:
Anyone is welcome to create a PR that makes the banner optional (with the default being `enabled` of course).

HHHartmann, 2020-01-30:
@marcelstoer But why do you close this issue?

marcelstoer, 2020-01-30:
#1010

532910, 2020-01-31:
@TerryE, I'm talking about a control to disable this banner, not about changing the current behaviour and disable it by default.

I'm agree it's very helpful for bugreps, but it may be unwanted for non-interactive serial devices.

https://github.com/nodemcu/nodemcu-firmware/pull/3026

532910, 2020-01-31:
@marcelstoer, I understand #1010, but still think it's odd to close it. This way no one will fix it. Do not treat it like a your responsibility. Think about it like a public todo list. It's not done, so should be unchecked (opened).

TerryE, 2020-01-31:
Marcel, given that @532910 is offering to do the fix, I also think that it is a little premature to close this.

@532910, what is your response to my point about there already being 1.2Kb of 'crap' sent to the UART, so what is really so bad about another ~350 bytes?  You only see it when debugging anyway.

I think that there should be a consensus on need before we consider changing the Lua core code.

On some of my apps I connect to my ESPs using the UART instead of WiFi, and with these `init.lua` goes straight into the app.  The controlling RPi issues JSON commands over the UART and gets JSON responses.  Any line not starting with `{` is ignored and hence the boot message is also ignored.

532910, 2020-01-31:
@TerryE, I'm sorry, I missed the note about 1.2Kb of 'crap' ):
BTW, could you confirm it also sends some crap to UART1 (GPIO2, TXD1) at boot-up?

The real problem that I have is the serial screen. So I need a separate pin to enable it only after boot-up. It will be cool just to send no 'crap' to it.

TerryE, 2020-01-31:
> The real problem that I have is the serial screen.

What do you mean? What are you using to read the TXD0 and TXD1 pins?  As for TXD1, output comes from three sources:  The ROM bootloader at 74880 baud
```
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 31860, room 16 
tail 4
chksum 0x74
load 0x3ffe8000, len 2460, room 4 
tail 8
chksum 0x67
load 0x3ffe899c, len 8, room 0 
tail 8
chksum 0x9a
csum 0x9a
```
the SDK 3.0 before handover to NodeMCU
```
system SPI FI size:4, Flash size: 4194304
boot not set
ota1 not set
ota2 not set
rf cal sector: 11
freq trace enable 0
rf[112] : 00
rf[113] : 00
rf[114] : 01

SDK ver: 3.0.1-dev(fce080e) compiled @ Apr 16 2019 15:02:05
phy ver: 1145_0, pp ver: 10.2
```
and then the  NodeMCU startup at whatever you've configured the frequency to be (by default 115,200)

532910, 2020-01-31:
The Futaba 20x2 dot matrix VFD M202MD07HB is connected to TXD1. TXD0 is used for debug & development.

PR #3026 (2020-02-09): No banner
Fixes #3021.

Make sure all boxes are checked (add x inside the brackets) when you submit your contribution, remove this sentence before doing so.

- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [x] I have thoroughly tested my contribution.
- [x] The code changes are reflected in the documentation at `docs/*`.

\<Description of and rationale behind this PR\>
closed 2020-02-09 by marcelstoer