nodemcu/nodemcu-firmware #3024: Lots of "[-Werror=…]" in NmraDcc.c

mk-pmb, 2020-01-30:
### Expected behavior

Firmware compiles, even with [my config](https://github.com/mk-pmb/nodemcu-firmware-wifigpio-pmb/tree/master/esp8266.app.include)

### Actual behavior

Lots of `[-Werror=…]` in `NmraDcc.c` in [my build log](https://gist.github.com/mk-pmb/795504c9f8fd5e166ecb3720f7b86a2b), then `cc1: all warnings being treated as errors`, then `Makefile:380: recipe for target '.subdirs' failed`

### Test code
https://github.com/mk-pmb/nodemcu-firmware-wifigpio-pmb/runs/415915753

mk-pmb, 2020-01-30:
The lines with warnings all seemed to be calls to `NODE_DBG`, so I [tried with #define DEVELOP_VERSION](https://github.com/mk-pmb/nodemcu-firmware-wifigpio-pmb/commit/9699575cc072146b1c708b8c2d1b6ddfb8b57da3), and at least it [changes the error](https://github.com/mk-pmb/nodemcu-firmware-wifigpio-pmb/runs/416499394) to:

```text
CC app/mbedtls/app/espconn_secure.c
espconn_secure.c:43:17: error: 'SSL_BUFFER_SIZE' undeclared here (not in a function)
   {NULL, ESPCONN_SECURE_DEFAULT_SIZE, 0, false, 0, false},
                 ^
../../../Makefile:422: recipe for target '.output/eagle/debug/obj/espconn_secure.o' failed
```

KT819GM, 2020-01-30:
It's so hard to find issues you are getting because you are using non standard building process and possibly not standard toolchain. I'm not getting any build warnings (all warnings are treated as error) with standard dev build and ```esp8266-linux-x86_64-20181106.0``` toolchain.

mk-pmb, 2020-01-30:
Under the hood my build process uses the docker image [`marcelstoer/nodemcu-build:latest`](https://github.com/marcelstoer/docker-nodemcu-build). It's an approach to actually increase reproducibility by implementing a CI build in the cloud. If you have ideas for how to make it more standard, I'm happy to try and apply them.

KT819GM, 2020-01-30:
To try I've just compiled firmware successfully on Mac using @marcelstoer docker image and local dev branch (1c83f02) with these randomly added modules (DCC included)

[user_modules.h.zip](https://github.com/nodemcu/nodemcu-firmware/files/4134621/user_modules.h.zip)

p.s. I don't wan't to tell that someone is doing something wrong. I just can't reproduce error on local, 'standard' machine. Will try on linux 'native' compile, maybe it will throw me an error and will let you know on gitter.

mk-pmb, 2020-01-30:
I discovered the dev branch has changed `app/include/user_config.h`, so I'll adapt my config and retry. If it works I'll make a PR to help separate experts-only-config from usual user config.

mk-pmb, 2020-01-30:
Fixed. Thos strange errors were just our firmware's way of saying "Your config file is outdated."
Dupe of #2990.

closed 2020-01-30 by mk-pmb