nodemcu/nodemcu-firmware #3025: There are two source files licensed under the GPL

pjsg, 2020-01-31:
I notice that there are two files `modules/am2320.c` and `driver/NmraDcc.c` that are licensed under the GPL. This is not compatible with the MIT license that the rest of the nodemcu firmware is licensed under.

I think that we either need to remove them or see if the original author will relicense them under the MIT license.

marcelstoer, 2020-01-31:
Great point, thanks! One more thing we need to pay attention to when reviewing PRs. We could also add another checkbox to the GitHub PR template.

AM2320 was contributed by @hvegh in the early days.
That driver file was just recently added by @vsky279 for the DCC decoder. It hasn't landed on master yet.

vsky279, 2020-01-31:
> I think that we either need to remove them or see if the original author will relicense them under the MIT license.

That would be sad but I got your point. I try to ask authors to relicense the project under MIT. Let's hope for the best.

pjsg, 2020-01-31:
I agree it would be sad for the DCC module -- I got quite excited when I saw it as I was thinking of stuff to use it for on my layout. There is another DCC source (though in C++) but it doesn't look as full featured as this one. However, it is under a very permissive license.

vsky279, 2020-01-31:
> By the way -- what interface are you using to connect to the track?

I am using the simplest extract of the https://github.com/IowaScaledEngineering/ard-dccshield, [v1.3 Schematic](https://github.com/IowaScaledEngineering/ard-dccshield/raw/master/pg/ard-dccshield-v1.3-e23d249/ard-dccshield.pdf). Basically only the part around the 6N137 optocoupler. I have not tried to implement the DCC ACK part of the circuit.

vsky279, 2020-01-31:
A reaction to my question specified that the used license is actually LGPL license.

I digged a bit into the license problematics and my understanding is that from license point of view it would be ok to include verbatim version of the [NmraDcc.cpp](https://github.com/mrrwa/NmraDcc/blob/master/NmraDcc.cpp) and build a module around it. Then this project's license wouldn't be impacted. My understanding is that this is the way `u8g2` graphic library is included.

Is my understanding correct?

That means that any adjustment to the file would need to be accepted in the initial repository. I tend to think that it worth trying. Though success is not guaranteed.

pjsg, 2020-01-31:
I've been thinking of putting together a small PCB to be plugged into to a
wemos d1 mini that would implement the interface circuitry from the
schematic that you are using. It would have the jumpers to allow powering
from the track, and two screw terminals for the connection to the track.
Would you be interested in those? Assembled or PCB only?

vsky279, 2020-01-31:
I am not using WeMos so not really interested. But thank you :-).
I'm using 12E and I'm about to make a quite small PCB with the optocoupler. I am using DCC rail for powering it but I use an external step-down circuit (+rectifier). The voltage drop is quite high for a linear stabilizer. My first version PCB used the linear stabilizer and it was not ideal.

nwf, 2020-02-02:
IANAL, IANYL, TINLA... but.. there's no real problem here, save that any binaries built with either of those modules become subject to the GPL, requiring anyone who ships them to also ship source.

kiwi64ajs, 2020-02-03:
> > I think that we either need to remove them or see if the original author will relicense them under the MIT license.
> 
> That would be sad but I got your point. I [try to ask authors](https://github.com/mrrwa/NmraDcc/issues/33) to relicense the project under MIT. Let's hope for the best.

Hi Guys,
I'm not keen to relicense this library to MIT as I feel it is too relaxed and also there are many contributors who I'd have to get permission from to make this change. I'm currently doing this for another Open Source project and its been quite tedious and problematic.

However, I just checked the original files and they are NOT licensed correctly/consistently. The Repo and included LICESE file are "Lesser GPL" whereas the actual text header in the NmraDcc[.ch] files is the GPL and not Lesser GPL text it should have been.

It was always the intent for this library to have the "Lesser GPL" license, but I must have copy/pasted the wrong "GPL" header and not the "Lesser GPL" header into the actual source files - doh!

I have just committed the "Lesser GPL" header corrections to both files:
    [NmraDcc.h](https://github.com/mrrwa/NmraDcc/blob/master/NmraDcc.h)
    [NmraDcc.c](https://github.com/mrrwa/NmraDcc/blob/master/NmraDcc.c)

Hopefully this should facilitate easier integration into your project.

Regards
Alex Shepherd

PS. Good to see this library being made accessible from LUA code. I have wondered for many years how hard/easy it would be to make it callable from LUA or Micro Python. I've recently (year ago) used LUA on a work project where is was the method for providing a means to execute "Custom User Code" in some commercial weigh scales.

marcelstoer, 2021-01-31:
@vsky279 I just wanted to correct the license header in our `NmraDcc.c(pp)` copy. In doing so noticed the upstream version got a couple of bugfixes in the meantime we might want to pull in as well: https://github.com/mrrwa/NmraDcc/commits/master/NmraDcc.cpp (everything since Feb 3, 2020 inclusive). Would you mind aligning our copy?

marcelstoer, 2021-01-31:
With a solution in sight for NmraDcc could we solve the am2320 issue by amending its documentation? IANAL either but I think https://github.com/nodemcu/nodemcu-firmware/issues/3025#issuecomment-581134879 is correct and, hence, a note in the docs could address this.

commit 6643cb46 (2021-02-14): Add note about GPL (#3400)

Fixes #3025
PR #3396 (2021-03-12): DCC module update to latest NmraDcc.cpp version
Updates DCC module to latest NmraDcc.cpp and fixes #3025.

- [x] This PR is for the `dev` branch rather than for the `release` branch.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [x] I have thoroughly tested my contribution.
- [x] The code changes are reflected in the documentation at `docs/*`.

This is the update of the DCC module to latest version of [https://github.com/mrrwa/NmraDcc/commits/master/NmraDcc.cpp]( [... cut]
commit c8f0ec8a (2021-12-30): Add note about GPL (#3400)

Fixes #3025
closed 2021-03-12 by marcelstoer