nodemcu/nodemcu-firmware #3028: Handling duplicated keys in a ROTable

TerryE, 2020-01-31:
### Current Behaviour

The standard Lua architecture enforces key uniqueness in table entries.  NodeMCU adds compile-time constant table implementation know as ROTables which can be statically compiled into Lua modules as vector structures and stored in Flash Memory rather than RAM.  Because the Lua `next()` exploits this uniqueness by finding the previous entry and then chaining to the next field, _if_ the ROTable contains duplicated entries then the enumeration algo goes into an infinite loop.

`luac.cross` also generates a ROTable index to the functions in a compiled LFS, but it currently does not do a uniqueness check on module names.

### Proposed Changes

My thought is that we shouldn't try to add runtime detection of duplicate keys, but rather document that keys in declared ROTable _must_ be unique.

`luac.cross` should also check for repeated module names within an LFS and raise an error on detection of a duplicate, so that module entries are unique.
 
### Possible future issue

At the moment the module name is based on the basename of the lua file name, so for example `lua_examples/luaOTA/_init.lua` creates the module `_init` which can be executed by `require "_init"` or `LFS._init()`.  With the increased use of LFS modules, the chance of name clash increases.  Perhaps we need to think about a convention for namespace separation.

nwf, 2020-02-02:
I think your proposed changes are sufficient.

The uniqueness constraint on ROTable keys is, presumably, post-linker?  That is, it's fine for two modules to declare "foo" keys so long as the C strings are commoned up in the .rodata section, yes?

TerryE, 2020-02-02:
You can have as many occurrences of a key as you want, e.g. `__index`, but only one per table.

TerryE, 2020-04-26:
`luac.cross` will throw a compile error if two modules have a common name, e.g. `_init`.  I suggest we treat any namespace issues as a future enhancement.

closed 2020-04-26 by TerryE