nodemcu/nodemcu-firmware #3030: nodemcu-build.com produces broken builds

bykof, 2020-02-01:
### Expected behavior

I click a build with dht, file, gpio, http, i2c, net, node, tmr, u8g2, uart, wifi packages. 
It builds properly and the given build is working.

### Actual behavior

When I install the build and try to send my code to esp, it prints: 
```
> wifi.setmode(wifi.STATION)
stdin:1: attempt to index global 'wifi' (a nil value)
```

**It's really strange that all builds are 404kb. I have an older build here from 27. January which was 476kb with exactly the same libraries**

### Test code
Go on [nodemcu-build.com](nodemcu-build.com)
Check `dht`, `file`, `gpio`, `http`, `i2c`, `net`, `node`, `tmr`, `u8g2`, `uart`, `wifi`.
For u8g2 options use: `ssd1306_i2c_128x64_noname` and `ssd1306_128x64_noname`

### NodeMCU startup banner

#### The broken build:

```
NodeMCU 3.0.0.0 built on nodemcu-build.com provided by frightanic.com
  branch: master
  commit: 71a182caa7841cbb478ed90ede526dc881943c80
  release: 
  release DTS: 202001061321
  SSL: false
  build type: float
  LFS: 0x0
  modules: dht,file,gpio,http,i2c,net,node,tmr,u8g2,uart,wifi
 build 2020-02-01 19:59 powered by Lua 5.1.4 on SDK 3.0.1-dev(fce080e)
```

#### The build who worked:

```
NodeMCU 3.0.0.0 built on nodemcu-build.com provided by frightanic.com
  branch: master
  commit: 71a182caa7841cbb478ed90ede526dc881943c80
  release: 
  release DTS: 202001061321
  SSL: false
  build type: float
  LFS: 0x0
  modules: dht,file,gpio,http,i2c,net,node,tmr,u8g2,uart,wifi
 build 2020-01-27 18:27 powered by Lua 5.1.4 on SDK 3.0.1-dev(fce080e)
```
### Hardware
`ESP8266` with Wifi and Chip: `ESP8266EX`

marcelstoer, 2020-02-01:
There's something fishy going on and I'll take a look. Thanks for diligent report.

However, the cloud builder is my personal project and I'll therefore close this issue here

closed 2020-02-01 by marcelstoer

bykof, 2020-02-01:
Okay thank you very much @marcelstoer!
How I know, when this bug is fixed? Could you leave a comment here when it's fixed?
I would be much obliged to you.

marcelstoer, 2020-02-02:
I had to revert a change I introduced in the backend. However, I still don't fully understand what the root cause of the changed behavior is.