nodemcu/nodemcu-firmware #3037: debounce should be a part of nodemcu

532910, 2020-02-12:
NodeMCU provides a lot of modules for different peripherals, but there are not simple push button support.

```Lua
local moduleName = ...
local M = {}

local pin

function M:setpin( p )
        pin = p
end

function M:init( p )
        M:setpin( p )
        gpio.mode( pin, gpio.INT )
end

local function debounce( func )
        local last = 0
        local delay = 150 * 1000

        return function( level, when, count )
                local delta = bit.band( when - last, 2^31-1 )
--              if delta < 0 then delta = delta + 2^31 end
                if delta < delay then return end
                last = when
                return func()
        end
end

function M:pressed( func )
        gpio.trig( pin, "down", debounce( func ) )
end

function M:released( func )
        gpio.trig( pin, "up", debounce( func ) )
end

_G[moduleName] = M
return M
```

532910, 2020-02-12:
1. It should a feature request, rather than a bug report.
2. not sure how well is it, but it works for me
3. possibly delay should be configurable
4. it depends on bit module, it can be get around with a commented string, but lua5.3 will bring binary operations

marcelstoer, 2020-02-16:
Back in 2015 when I started with IoT & NodeMCU I published two gists for debouncing one of which uses the same strategy as this one:
- https://gist.github.com/marcelstoer/59563e791effa4acb65f
- https://gist.github.com/marcelstoer/75ba30a4aec56d1b3810

Feel free to offer this as a PR but keep the upcoming changes in #3034 in mind.

532910, 2021-06-03:
I still believe it's actual.

TerryE, 2021-09-26:
@532910  sergio, I feel that this is something that the committers won't address, so if you want to develop a solution  and provide a PR for this, then by all means reopen the issue.

closed 2021-09-26 by TerryE