nodemcu/nodemcu-firmware #3038: rst cause:2 on module initialization from the command line

532910, 2020-02-12:
I wrote a module for Futaba M202MD07HB VFD serial text screen.
When I call it from the `init.lua` it works fine. But trying to do the same manually from the command line causes a reset.

### Actual behavior
```
ets Jan  8 2013,rst cause:2, load 0x40100000, len 28348, roomtail 12
chksum 0xa3
ho 0 tail 12 room 4
load 0x3ffe8000, len 2600, room 12 
tail 12
chksum 0x0d
ho 0 tail 12 room 4
load 0x3ffe8a28, len 8, room 12 
tail 8
chksum 0xc0
```

### Test code
```Lua
m202 = require( 'm202' )
m202:init( 1, 31250, 'dim' )
```

truncated m202.lua module, only init routines are left:
```Lua
local moduleName = ...
local M = {}

local DIM_MAX = 5

local uartID = nil
local dim = DIM_MAX
local dimFile = nil

local cmds = {
        ['dim']               = 0x04,
        ['clear']             = 0x0d,
        ['reset']             = 0x1f
}
local dims = { 0x00, 0x20, 0x40, 0x60, 0x80, 0xff }

function M:write( ... )
	uart.write( uartID, ... )
end

function M:cmd( c, ... )
	M:write( cmds[c], ... )
end

function M:dim()
	M:cmd( 'dim', dims[dim+1] )
end

function M:dimSet( d )
	dim = d
	if dimFile then
		file.putcontents( dimFile, dim )
	end
	M:dim()
end

function M:initDim( _dimFile )
	dimFile = _dimFile
	local d
	local f = file.open( dimFile, 'r' )
	if f then
		d = tonumber( f:read() )
		f:close()
	end
	if d~=0 and d~=1 and d~=2 and d~=3 and d~=4 and d~=5 then
		d = dim
	end
	M:dimSet( d )
end

function M:init( id, baud, _dimFile )
	uartID = id
	uart.setup( id, baud, 8, uart.PARITY_NONE, uart.STOPBITS_1, 1 )
	M:cmd( 'reset' )
	if _dimFile then
		M:initDim( _dimFile )
	end
end

_G[moduleName] = M
```

It's a LUA_NUMBER_INTEGRAL build with the following modules:
```
% grep -v '^//\|^$' user_modules.h                 
#ifndef __USER_MODULES_H__
#define __USER_MODULES_H__
#ifndef LUA_CROSS_COMPILER
#define LUA_USE_MODULES_BIT
#define LUA_USE_MODULES_FILE
#define LUA_USE_MODULES_GPIO
#define LUA_USE_MODULES_NET
#define LUA_USE_MODULES_NODE
#define LUA_USE_MODULES_RTCTIME
#define LUA_USE_MODULES_SNTP
#define LUA_USE_MODULES_TMR
#define LUA_USE_MODULES_UART
#define LUA_USE_MODULES_WIFI
#endif  /* LUA_CROSS_COMPILER */
#endif  /* __USER_MODULES_H__ */
```

### NodeMCU startup banner
NodeMCU 3.0.1.0 
        branch: 
        commit: 
        release: 
        release DTS: 
        SSL: false
        build type: integer
        LFS: 0x0
        modules: bit,file,gpio,net,node,rtctime,sntp,tmr,uart,wifi
 build 2020-02-01 07:39 powered by Lua 5.1.4 on SDK 3.0.2(824dc80)

### Hardware
ESP8266 wemos D1 (not mini)

marcelstoer, 2020-02-16:
> When I call it from the `init.lua` it works fine.

Does this indicate that it's likely not a firmware issue then?

532910, 2020-02-16:
Marcel, is it reproducable for you?
Forgot to say, you don't need to have or connect the VFD module.

nwf, 2020-02-23:
Please give an example interaction, not just the reboot.  What have you run at the Lua prompt prior to the reboot happening?

532910, 2020-02-23:
> What have you run at the Lua prompt prior to the reboot happening?

the **Test code**

532910, 2021-06-03:
Why bot is allowed to close issues?

532910, 2021-06-03:
Reopen please.

nwf, 2021-06-03:
It certainly appears that development of NodeMCU has... substantially tapered off, if not died completely.  I fear if you're waiting for answers you're going to be waiting a while.

532910, 2021-06-03:
I'm not waiting for an answer, but this is a bug that should be fixed regardless of time.

TerryE, 2021-09-26:
@532910  as @nwf says this is a one-developer issue of "my app doesn't work".  To be frank this falls into the won't fix category.  Sorry

closed 2021-09-26 by TerryE