nodemcu/nodemcu-firmware #3039: Builtin LED is LOW when setting to HIGH and vice versa

trentmillar, 2020-02-13:
### Expected behavior

on-board LED should be off when `digitalWrite(LED_BUILTIN, LOW);` 
... on when `digitalWrite(LED_BUILTIN, HIGH);` 

### Actual behavior

if `LOW` the LED is on, if `HIGH` the LED is off

### Test code
```
void setup() {
  // initialize digital pin LED_BUILTIN as an output.
  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, LOW);
}

// the loop function runs over and over again forever
void loop() { 
}
```
### NodeMCU startup banner
Not sure how to provide this when flashing via the Arduino IDE.

### Hardware
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz

https://keyestudio.com/new-keyeastudio-nodemcu-lua-esp8266-esp-12f-wifi-module-1m-usb-cable-development-board-compatible-with-networking-p0456-p0456.html

##### Steps

1. open the Arduino IDE
2. Go to files and click on the preference in the Arduino IDE
3. copy the below code in the Additional boards Manager
- http://arduino.esp8266.com/stable/package_esp8266com_index.json
4. click OK to close the preference Tab.
5.  go to Tools and board, and then select board Manager
6. Navigate to esp8266 by esp8266 community and install the software for Arduino.

udif, 2020-02-13:
Are you sure driving the pin with LOW should make it on?
Typically LEDS are powered by Vcc on one side and are turned on by the I/O pin going low (because traditionally TTL logic could sink much more current (i.e. current going into a low pin) than sourcing it (a high pin driving out current).
With CMOS it seems that sink and source current limits are the same (just checked the Esp8266 datasheet), so no special reason to keep hooking LEDs that way, other than as a habit.
I went ahead and checked the nodemcu schematics and indeed you need to drive the pin low for the led to turn on.

closed 2020-02-13 by trentmillar

marcelstoer, 2020-02-14:
> open the Arduino IDE

You picked the wrong project.