nodemcu/nodemcu-firmware #3040: Our LwIP is way out of date

nwf, 2020-02-16:
Our LwIP appears to correspond to upstream's effcb90fdf29e6475614c84d327bcf2722f854b6
a commit made in December of **2010**, but with some local patches added.  We should:

* Figure out what changes we have made to LwIP that lack upstream analogues,
  if any, and move to have those upstream.
* Move to mainline LwIP
* Remove `espconn_secure` in favor of LwIP's upstream `altcp_tls`.

rvalle, 2020-03-24:
@nwf did you look into this? I think #2931 could be related. We see sine HTTP requests failing and we can trace it down to different use of TCP segments by different browsers.

nwf, 2020-03-24:
Unfortunately, I have looked into it and LwIP straddles the boundary between the open and closed aspects of the SDK.  https://github.com/pfalcon/esp-open-sdk/issues/263 provides a reasonable, if depressing, overview of the situation.

https://github.com/d-a-v/esp82xx-nonos-linklayer looks like a sensible shim between the proprietary SDK bits and a more modern LwIP, but I have not had time to look into it much beyond theoretical discussion.

Espressif themselves have done a more modern LwIP, https://github.com/espressif/esp-lwip, but only as part of the IDF/RTOS world, so not us.  This might be further indication that living out on the NONOS branch is not a long-term survival strategy for NodeMCU, but I think we have not wanted to make the jump due to the RTOS increasing heap usage (by how much I cannot say; if someone has a link to more information handy, that'd be great).

pjsg, 2020-09-21:
To me, the https://github.com/d-a-v/esp82xx-nonos-linklayer repo looks like the right code to start with. I especially like that they have managed to get ipv6 working -- and while I don't expect nodemcu to support it as a result of this migration, it then would at least be feasible -- it is mostly the lua interfaces that would need to be changed.

TerryE, 2020-09-22:
I endorce @nwf and @jsqg working on this but I won't have much bandwidth to support this until I have this next tranche of LFS changes in and working.  BTW, as I noted elsewhere I've got some yard-works going on at the moment and keeping an eye on this is soaking up a lot of my free time for this coming week.

closed 2022-04-30 by stale[bot]