nodemcu/nodemcu-firmware #3046: dbg_printf does not print %llx correctly

nwf, 2020-02-23:
### Expected behavior

64-bit hex strings

### Actual behavior

32-bit hex strings

### Workaround

Use `printf` instead.

TerryE, 2020-04-10:
Nathaniel, if you look at @pjsg's implementation dbg_printf() was always intent as a lean and mean low stack cost implementation.  Hence it excludes most of these enrichments.  I am not sure if it worth replacing this.

pjsg, 2020-04-10:
The goal of this dbg_printf was to implement all the standard types and not have any buffer overflows and not use lots of stack. It doesn't implement a lot of stuff, but it seems that adding support for %llx would not be that hard... Do we support int64_t?

nwf, 2020-04-10:
We do support `int64_t` and it's used extensively in `mbedtls` and `sntp` (and a little in `tmr`) for example.  It's not hugely important that `dbg_printf` support it, but it would be nice.

closed 2021-06-20 by stale[bot]