nodemcu/nodemcu-firmware #3049: Docker Build Makefile fails on line 320

NicolSpies, 2020-03-09:
### Expected behavior
Using the docker process that I have been using in the last year, I expected a successful bin image build after pulling latest dev commit and `docker pull marcelstoer/nodemcu-build`
### Actual behavior
```
....
  inflating: ESP8266_NONOS_SDK-e4434aa730e78c63040ace360493aef420ec267c/third_party/include/ssl/ssl_private_key.h  
  inflating: ESP8266_NONOS_SDK-e4434aa730e78c63040ace360493aef420ec267c/third_party/include/ssl/ssl_ssl.h  
  inflating: ESP8266_NONOS_SDK-e4434aa730e78c63040ace360493aef420ec267c/third_party/include/ssl/ssl_tls1.h  
 extracting: ESP8266_NONOS_SDK-e4434aa730e78c63040ace360493aef420ec267c/third_party/include/ssl/ssl_version.h  
 extracting: ESP8266_NONOS_SDK-e4434aa730e78c63040ace360493aef420ec267c/third_party/include/user_config.h  
mv: cannot move '/opt/nodemcu-firmware/sdk//ESP8266_NONOS_SDK-e4434aa730e78c63040ace360493aef420ec267c' to '/opt/nodemcu-firmware/sdk//esp_iot_sdk_v3.0-e4434aa': Operation not permitted
Makefile:320: recipe for target '/opt/nodemcu-firmware/sdk/.extracted-3.0-e4434aa' failed
make: *** [/opt/nodemcu-firmware/sdk/.extracted-3.0-e4434aa] Error 1
make: Leaving directory '/opt/nodemcu-firmware'
````

### Test code
```Lua
run --rm -it -v c:/nodemcu:/opt/nodemcu-firmware marcelstoer/nodemcu-build build
```

NicolSpies, 2020-03-09:
This issue seems to happen if the Windows 10 folder permissions are not set correctly to provide full folder control.

closed 2020-03-09 by NicolSpies

marcelstoer, 2020-03-09:
In any case it would have been something for https://github.com/marcelstoer/docker-nodemcu-build/.