nodemcu/nodemcu-firmware #3052: tls no funciona

Kmiz07, 2020-03-12:
he probado todos los ejemplos existentes para nodemcu esp8266 sobre tls y ssl y no funciona ninguno.
Imagino que ya tienen constancia de ello.
Entonces..... Porque continuan ofreciendo los modulos?
Si me equivoco, Podrian enseñarme algun ejemplo que funcione para abrir una web https de cualquier servidor?

I have tried all the existing examples for nodemcu esp8266 on tls and ssl and none works.
I imagine they already have proof of this.
So ..... Why do you continue to offer the modules?
If I'm wrong, could you show me an example that works to open a https web from any server?

vsky279, 2020-03-12:
Hi, could you provide your testing code? What modules are you using? What build are you using?

Kmiz07, 2020-03-12:
He probado con varias compilaciones diferentes, y con los ejemplos que hay
en https://nodemcu.readthedocs.io
Las web http sin problemas, pero https no funciona.
Conoces algún código que consiga un get a alguna página https?

I have tried several different compilations, and with the examples at
https://nodemcu.readthedocs.io
 The http web without problems, but https does not work.
 Do you know any code that can get a https page?

vsky279, 2020-03-12:
You need to quote the minimum code to reproduce the issue and state the build you are using. You should also follow the template to report bugs.

Kmiz07, 2020-03-12:
my current compilation is:
NodeMCU 3.0.0.0 built on nodemcu-build.com provided by frightanic.com
	branch: master
	commit: 71a182caa7841cbb478ed90ede526dc881943c80
	release: 
	release DTS: 202001061321
	SSL: true
	build type: float
	LFS: 0x20000
	modules: ads1115,bit,crypto,dht,file,gpio,http,i2c,mdns,net,node,ow,sjson,sntp,spi,tmr,uart,websocket,wifi,tls
 build 2020-02-09 21:11 powered by Lua 5.1.4 on SDK 3.0.1-dev(fce080e)

Although as I have already said, I have tried other compilations with the same result.
one of the examples i have tried is:

function pruebahttp()
http.get("https://www.google.com/", nil, function(code, data)
    if (code < 0) then
      print("HTTP request failed")
    else
      print(code, data)
    end
  end)

another example:

srv = tls.createConnection()
srv:on("receive", function(sck, c) print(c) end)
srv:on("connection", function(sck, c)
  -- Wait for connection before sending.
  sck:send("GET / HTTP/1.1\r\nHost: google.com\r\nConnection: keep-alive\r\nAccept: */*\r\n\r\n")
end)
srv:connect(443,"google.com")

I suppose there is some way to make it work, but the examples in the documentation do not make it work.

marcelstoer, 2020-03-13:
What you currently get back from https://www.google.com/ is ~13KB - too much to fit onto the heap.

https://nodemcu.readthedocs.io/en/latest/modules/http/ -> SSL/TLS support

closed 2020-03-13 by marcelstoer

Kmiz07, 2020-03-13:
Yo pedía un ejemplo que si funcionase. No podéis darme ninguno?

Kmiz07, 2020-03-13:
Me parece muy mal que hayan cerrado el post.
Yo pedía un ejemplo que funcione para poder ver en qué fallo. He mostrado
el ejemplo de Google.com, pero he probado muchos otros.
He mirado mucha información y no he encontrado ningún ejemplo que funcione.
Almenos en lua.
Por eso pedía me mostraran alguno que funcionará correctamente, pero cada
vez estoy más convencido de que no funciona. No me gusta programar con
Arduino pero al final no tendré otra opción en este caso.

I find it very bad that they have closed the post.
 I was asking for an example that works so I can see what failure.  I've
shown the Google.com example, but I've tried many others.
 I have looked at a lot of information and have not found any example that
works.  At least in lua.
 So I asked to be shown one that will work correctly, but I am increasingly
convinced that it does not work.  I don't like programming with Arduino but
in the end I won't have any other option in this case.

vsky279, 2020-03-13:
I understand your disappointment. But reporting an issue is used when your report a bug or you want to discuss some new feature for example. It is not to be used when you want to get an advice.

The example you gave does not demonstrate any bug. That is why the issue was closed. It just needs more memory than you actually have. It's like trying to move a cow in a passenger car.

Try to connect to some other server. For example https://ifttt.com works fine for me. It makes also more sense to connect to this site with a device like ESP8266 than to google.com.

marcelstoer, 2020-03-13:
This works

```lua
http.get("https://nodemcu-build.com/jumbotron-narrow.css", nil, function(code, data)
    if (code < 0) then
      print("HTTP request failed")
    else
      print(code, data)
    end
  end)
```

And so does https://get.geojs.io/v1/ip.json or https://opendata.netcetera.com/smn/smn/OTL. If you care to study the documentation you will understand why the same code works for some URLs but not for others.

Kmiz07, 2020-03-14:
It was only this that he asked for.  Thank you

Kmiz07, 2020-03-14:
Hi, I have tried the example that told me it works. The answer is:

*E:M 1592E:M 1592E:M 1592E:M 1592E:M 1592E:M 1592E:M 1592HTTP client:
Disconnected with error: -1HTTP client: Connection timeoutHTTP request
failed*

as you see, in my compilation it doesn't work. I have now used a new dev build:

NodeMCU 3.0.0.0 built on nodemcu-build.com provided by frightanic.com
	branch: dev
	commit: c116d9d25f5d15e2b1d0d52ff4c8ebdfd18f75c3
	release:
	release DTS: 202003062124
	SSL: true
	build type: float
	LFS: 0x0 bytes total capacity
	modules: crypto,file,gpio,http,net,node,sjson,softuart,tmr,uart,wifi,tls
 build 2020-03-13 00:52 powered by Lua 5.1.4 on SDK 3.0.1-dev(fce080e)

[image: image.png]