nodemcu/nodemcu-firmware #3055: Support for Homie MQTT convention

steelman, 2020-03-19:
### Homie MQTT convention
[Homie](https://homieiot.github.io/specification/) is a convention for IoT devices to announce themselves on MQTT brokers that makes it easy for controllers (e.g. home automation software like OpenHAB) to discover the devices.

### Justification
Homie compliant devices are easier to support by different controllers.

An API built on top of the MQTT module to add some topics automatically and configure others with less amount of code would enable easier development.

### Workarounds
The convention may be implemented "by hand" using the mqtt module directly.

pastukhov, 2020-03-20:
Yes, IMHO it should be made with lua code

marcelstoer, 2020-05-28:
This shouldn't be built into the MQTT module but can be offered as a Lua module on top of it. PRs are certainly welcome. As per #1010 I am closing this for now.

closed 2020-05-28 by marcelstoer