nodemcu/nodemcu-firmware #3056: The ESP32 function gpio.read(pin) function is NOT working properly.

ru4reel, 2020-03-21:
### Expected behavior
pin=19 gpio.config({gpio=pin, dir=gpio.OUT}) gpio.write(pin,1) p = gpio.read(pin) print(p)
RESULT = 1
and
gpio.write(pin,0) p = gpio.read(pin) print(p)
RESULT = 0

### Actual behavior
pin=19 gpio.config({gpio=pin, dir=gpio.OUT}) gpio.write(pin,1) p = gpio.read(pin) print(p)
RESULT = 0
and
gpio.write(pin,0) p = gpio.read(pin) print(p)
Also RESULT = 0

### Test code
pin=19
gpio.config({gpio=pin, dir=gpio.OUT})
gpio.write(pin,1)
p = gpio.read(pin) print(p)
-- gpio.write(pin,0)
-- p = gpio.read(pin) print(p)

### NodeMCU startup banner
[0;32mI (73) boot: Chip Revision: 1[0m
[0;32mI (73) boot_comm: chip revision: 1, min. bootloader chip revision: 0[0m
[0;32mI (41) boot: ESP-IDF v3.3.1 2nd stage bootloader[0m
[0;32mI (41) boot: compile time 17:50:08[0m
[0;32mI (41) boot: Enabling RNG early entropy source...[0m
[0;32mI (46) boot: SPI Speed      : 40MHz[0m
[0;32mI (50) boot: SPI Mode       : DIO[0m
[0;32mI (54) boot: SPI Flash Size : 4MB[0m
[0;32mI (58) boot: Partition Table:[0m

### Hardware
ESP32 Do-it ver 1

chilipeppr, 2020-03-21:
I don't think you can do a read, which is a gpio.IN, on a pin when you have
it in gpio.OUT mode.

ru4reel, 2020-03-21:
Hi chilipeppr 
You give me an idea.
So I tried dir=gpio.IN_OUT
Then it works now.
Thank you very much.

closed 2020-03-22 by ru4reel