nodemcu/nodemcu-firmware #3057: Add MCP23017 port expander support as a lua module

plomi-net, 2020-03-21:
### Missing feature
Hello, everybody,
I would like to contribute a few things to this great project.

I saw that there are already issues for the MCP23017 Port Expander. Unfortunately I didn't find any Lua or C modules in the current dev branch.

Without looking into the issues, I just created a module in Lua for the MCP23017.
Now I would like to learn more and adapt the module to become part of this project.

### Justification
This module can help other IoT developers to easily provide more GPIO pins, for example to control a relays card. 

### Workarounds
I am not sure which workaround represents a good solution.

Kind regards,
Marcel

marcelstoer, 2020-03-22:
Contributions are welcome. Consider #3034.

jmd13391, 2020-03-22:
@plomi-net, Have you see this?

[MCP23008 Module](https://nodemcu.readthedocs.io/en/master/lua-modules/mcp23008/)

I am currently using this Lua Module to extend the ESP8266 GPIO with great success.

plomi-net, 2020-03-24:
Hi @jmd13391, @marcelstoer 

Thank you very much for your comment.
I had seen the module for the MCP23008 chip and that inspired me to create a module for the MCP23017.

For my project I need 16 channels, theoretically 128 pins would be possible with 8 MCP23017. 
I could also have used two MCP23008.

I will refactor my module in the next few days and then provide it.

plomi-net, 2020-04-10:
Hi,
now some time has passed.

I have attached the program code with a pull request.
I hope the refactor succeeded and helps someone to extend the nodemcu.
If anyone has any comments or feedback, please let me know.

thanks a lot

PR #3197 (2020-10-25): add mcp23017 Lua module
Fixes #3057

Make sure all boxes are checked (add x inside the brackets) when you submit your contribution, remove this sentence before doing so.

- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [x] I have thoroughly tested my contribution.
- [x] The code changes are reflected in the documentation at `docs/*`.

Add MCP23017 port expander support as a lua module. See issue #3057  [... cut]
closed 2020-10-25 by marcelstoer