nodemcu/nodemcu-firmware #3065: UART Flow Control

rvalle, 2020-04-10:
### Missing feature
UART Flow Control Configuration
### Justification
NodeMCU may ingest Serial Data that requires long time to process. For example, sending it over the network.  NodeMCU has no way to tell the UART sender to wait until the previous submission has finished to post more data. Flow Control is the right way to solve this issue.
I can see Flow Control in the underlying APIs / Platform, but for some reason the setting seems not to be available on the API.
### Workarounds
Applications could reimplement flow control themselves. For example. Sending custom messages over UART, instead of using XON/XOFF over the wire, or using other GPIOs instead of RTS/CTS.

TerryE, 2020-04-10:
On most H/W implementations, ESP modules follow the Arduino practice of tying RTS DTS to the boot GPIOs.  This means that ESP tools can hard reboot the modules programmatically.  However, the downside s that you can not use RTS/DTS for H/W flow control.  Your workaround is really the only option that we've got. Sorry.

nwf, 2020-04-10:
In principle, we need only modulate the connection's `CTS` pin, since we can probably assume that the host computer is going to keep up.  It looks like that's `GPIO13` on the ESP8266.  A careful hand may be able to grab the corresponding signal on the USB<->UART chip on a typical NodeMCU module.

TerryE, 2020-04-11:
Not sure about that. Need to check pinouts, but aren't they connected to RST + GPIO 0? You can reset most modules and optionally put it into the ROM bootloader by commanding RTS + CTS. This relies on disabling FIFO-triggered H/W flow control.

Even if partly doable, this seems a lot of work +  technical difficulties to overcome for something that significantly less than 1% of users will adopt.

nwf, 2020-04-11:
This is mostly for @rvalle's reference, should they wish to pursue something in this space.

Assuming https://raw.githubusercontent.com/nodemcu/nodemcu-devkit-v1.0/master/NODEMCU_DEVKIT_V1.0.PDF describes the design everyone adopts, the reset mechanism is driven by the USB UART's `DTR` (to `GPIO 0`) and `RTS`/`RTR` (to `nRST`), cross-wired so that the 00 and 11 states both result in `GPIO 0` and `nRST` both being pulled high through their associated pull-up resistors.  (That is, this design depends upon the host computer being able to transition from `RTS=0 DTR=1` to `RTS=1 DTR=0` quickly enough to hit the sample taken by the bootloader.)  The four DCE->DTE control signals (`RI`, `DCD`, `DSR`, and `CTS`) are exposed by the USB UART but are not connected to the ESP8266 (nor anywhere else on the board).

Reading the ESP8266 manual (https://www.espressif.com/sites/default/files/documentation/esp8266-technical_reference_en.pdf, section "11.3.10 Flow Control"), it's apparent that the ESP8266 UART pins are labeled as if it were DTE, not DCE, which is confusing!  Thus, the USB UART's CTS pin (that is, the host's sense of whether or not the ESP8266 is ready to receive) must be connected to the ESP8266's RTS pin, `GPIO 15`, not `GPIO 13` as I said above.  I apologize for the error.

Conveniently, the ESP8266's UART understands asymmetric flow control, so we can have it blissfully assume that it always sees an asserted `CTS` (i.e., what would be the USB UART `RTS` signal).

It looks like you'll want to have C call
```C
PIN_FUNC_SELECT(PERIPHS_IO_MUX_MTDO_U, FUNC_U0RTS);
UART_SetFlowCtrl(0, USART_HardwareFlowControl_RTS, 0x7F);
```

If you (or anyone else) pursue this, I would be inclined to accept a patch into the tree along with documentation detailing the hardware modification.  The appropriate place to expose this is as a new parameter to `uart.config()` and new result from `uart.getconfig()` (with corresponding changes to `platform_uart_setup` and `platform_uart_get_config).  The implementation of `uart.alt()` (esp. `platform_uart_alt`) may need some adjustment as well.

rvalle, 2020-04-15:
@TerryE  and @nwf many thanks for your replies and loads of information.

Unfortunatedly I inherit a HW design where the ESP was thought as a "modem" adapter to wifi for another microcontroller, and the serial 0 is the main communication link with only RX/TX been wired.

That leaves me no other option but XON/XOFF for flow control. After looking at the documentation I was it supported by lower layer APIs, and I wondered if it is just a non-exported parameter of perhaps another configuration existed.

The use case is OTA, not only for the ESP but also for this other microcontroller (AVR). At first a tried to stream the firmware from Network to Serial and then to the a SPI memory where this FW resides. Wifi is just way too fast compared to the write speed of the SPI memory... which seems to be the main bottleneck. 

In case anybody finds a similar use case. Finally I deceide to cache the download in the ESP filesystem wich I still find huge, as updates are only 30kb and there are 3Mbytes. Then the AVR reads the file in chucks one request a a time.

It would be nice to try with Xon/Xoff and see if that works. And that would translate in also flow-controlling the TCP stream... I have used Xon/Xoff in the past with sucess even to run TCP over a 2 wire long serial cable... many years ago, but worked like a charm.

nwf, 2020-04-16:
When you say "supported by lower-level APIs", what do you mean?  I didn't see mention of it in the ESP SDK guide, for example?

In principle, you could hook the `UART_RXFIFO_FULL_INT` interrupt state machine to provide fairly
low-level XON/XOFF driving, but it doesn't look like NodeMCU has any code to do that.

IMHO, you're best off, if you can, restricting yourself to bursts of <128 bytes (the size of the ESP8266's fifo) and waiting for positive acknowledgement from the software running.  Even a very minimal framing protocol would do just fine, if you have control of software on both sides of the stack.
FWIW, this is the approach I take in the test framework I'm working on: write up to 128 bytes, wait for the ESP to echo what I sent it, wait for a prompt, repeat.  It's not the greatest interface, and it's somewhat slower than it could be, but it doesn't overrun buffers.

rvalle, 2020-04-16:
@nwf I just did a quick search on the codebase for XOFF and this came up:

```
typedef enum {
    NONE_CTRL,
    HARDWARE_CTRL,
    XON_XOFF_CTRL
} UartFlowCtrl;
```
In what appears to be an UART driver C file.

I, perhaps wrongly, assumed that the implementation is already there in C but not exported to the LUA api.

nwf, 2020-04-16:
That's an interesting find.  If the boot ROM is doing something funny, I am unaware of it, but that hardly means it's not!

TerryE, 2020-04-16:
No, the driver doesn't currently include s/w flow control. 

IIRC, the AVR includes a H/W FIFO on the UART, probably 128 bytes so a SLIP-style protocol will work fine. Emit fixed length records with CRC trailer, and the flow control is the receiver doing an ACK or NAK response. A lot easier then us trying to extend the UART driver for a single request.

closed 2021-06-20 by stale[bot]