nodemcu/nodemcu-firmware #3068: A multitude of `mqtt` module bugs.

nwf, 2020-04-10:
There are some long-standing sources of sadness in our `mqtt` module.

* The data structures contain many redundant or unused fields, and some fields are used in confusing or surprising ways.

* The generic names `msg_enqueue` and friends have somehow not tripped us up yet, but they surely will as time goes to infinity.

* The timer fires once per second rather than at the end of the timeout.

* The `mqtt` module maintains its own notion of connection timeout (#2576, #2674) which is silly; the network stack knows better than we do whether it is making progress.

* `:close()` pushes the "orderly shutdown" message and then immediately calls disconnect.  This may break LwIP, by sending on top of another send, but that aside it is also wrong: we should wait for the shutdown message itself to be sent by the network stack before hanging up, otherwise it appears to be a disorderly shutdown if the stack tears down the connection before the send happens.

* Rx is not corked while a send is outstanding; that means that we might violate the LwIP/espconn constraint to never send two messages without getting the `sent` callback after the first.

* We don't pin the lmqtt_userdata structure in the registry across DNS and LwIP callbacks.  Calling `:close()` at an inopportune moment can cause a use-after-free (and probably crash) on the C side.  There are implications here as pertains to #3062 as well.

* Boy howdy do we do a lot of allocations and aren't particularly good at checking them and making sure the state machine either keeps going or hangs up the connection.

Fixing these will take extensive overhauling, and help would be appreciated.  I will keep my work public on https://github.com/nwf/nodemcu-firmware/tree/dev-active

jpeletier, 2020-05-11:
Hi @nwf,

I rewrote the entire MQTT module for ESP32 in #2629. However, it takes advantage of the ESP-IDF SDK MQTT stock implementation, so in ESP32 it is merely an adapter of that to Lua.

If it is full of bugs and old code, may be it makes sense to scrape the entire thing and replace with something that works, like [this implementation](https://github.com/tuanpmt/esp_mqtt). We could even link it in as a submodule to piggyback on that repo somehow.

What do you think?

nwf, 2020-05-11:
I'm not really thrilled with the code you linked to, either, and see in it some similarity to the code already in tree. :(

marcelstoer, 2020-05-12:
>  some similarity to the code already in tree.

That's because it befathered our module way back then 😉 

-> https://github.com/nodemcu/nodemcu-firmware/blob/dev/app/mqtt/mqtt_msg.h#L3

jpeletier, 2020-05-14:
😅

Would it make sense link to this library as a submodule, so we can at least get their latest fixes?

marcelstoer, 2020-05-14:
That's a valid point we keep coming back to, see #1411. 

That "upstream" module hasn't seen any changes in close to three years. We have probably fixed more bugs in the meantime than they have.

nwf, 2020-05-14:
At this point, I am taking a fairly "scorched earth" rewrite stance, making sure that changes continue to pass the limited test harness I have written.  Contributions to the test harness https://github.com/nwf/nodemcu-firmware/blob/dev-active/tests/test-mqtt.expect would be especially welcome since there are a lot of opportunities for messages in MQTT and I'm sure I'm only covering a small fraction of possible interleavings and orderings.  Notably, improving the test harness also helps ensure that whatever we do in terms of the on-module implementation hasn't regressed!

closed 2021-06-20 by stale[bot]