nodemcu/nodemcu-firmware #3078: Error Handling in Lua

TerryE, 2020-04-28:
#3075 introduced the new `luaL_callx()` to facilitate improved [Panic Handling](https://gist.github.com/TerryE/646407100785fb00c2cbddbb0e8a2b61#panic-handling).  I want to use this issue to discuss error handling in more general terms and why the focus on panic handling. 
-  **Caught Errors**.  The Lua architecture allow the application so establish an error handler via the `pcall()` and `xpcall()`, thus creating a _protected environment_ for executing a code hierachy.  In such cases the error is returned to the application and the application code itself determines how to process or log the error.
-  **Uncaught Errors**.  Any Lua code executed outside such a protected environment results in a Lua panic, which in standard Lua coding typically terminates the application.  In the case of NodeMCU, this triggers a processor restart.  

One of the main issues that catches new developers is that whilst the interactive thread established a default protected environment to catch and print errors interactively, any Lua callback runs as separate execution thread and therefore is not protected; any errors here will panic and reboot the processor.  The issue that I want to consider here is whether this is the correct behaviour for NodeMCU and if not then how we can improve this in a way which doesn't break existing applications.

We have three broad mechanisms for reporting errors in the firmware:

1.  **Print direct to the UART**.  We adopt this approach during Lua startup because this is the simplest and most robust mechanism that avoids dependencies on services that might not be yet running.
2.  **Print the error using the standard `print()` function**. Prior to SDK 3.0 releases, we always did (1) but this meant that errors generated during `node.output()` where reported only on the UART.  `print()` now sends output to the `STDOUT` which is in turn emptied to the UART or the `node.output()` reader using the [Pipe](https://nodemcu.readthedocs.io/en/dev/modules/pipe/) module, and this enables error reporting in a terminal session to occur through the session.
3. **Use a separate panic process manger** to process uncaught errors.  The problem with (2) during panics is that a simple "print error and restart" will queue the error message to STDOUT then restart the processor before the printing task has executed.  At a minimum, the restart operation should wait until the STDOUT pipe is empty (or a maximum time elapsed) before restarting the ESP.

In case (3) the default panic process should still print the error, but at least wait until the STDOUT pipe is empty or a maximum time elapsed before restarting the ESP.  We should also provide a `node.setonerror()` call to enable the application to override this default (say to add some form of error logging).

### Scope of Work

So I suggest that a good next PR would be to tidy up such error handling:
-  Review Lua code for complaince to above (1)-(3) separation.
-  Modify `node.restart()` functionality to wait until STDOUT or N sec elapsed before restarting ESP.  Add extra `delay` boolean parameter to enable this mode and document this. _**Q**: Should this wait until errors flushed be thedefault?_
-  Add `node.setonerror()` call and document.
-  Modify all module CBs to use `luaL_pcallx()`
-  Modify `debug.debug()` function to read from STDIN pipe, and output to `print()` (Maybe a separate PR.)

TerryE, 2020-04-28:
> ~~Modify `debug.debug()` function to read from STDIN pipe, and output to `print()` (Maybe a separate PR.)~~

This one needs a bit of explanation.  One of the ways (in standard Lua at least) to debug is use `debug.debug` as an `xpcall()` error handler ([see PiL 8.5](https://www.lua.org/pil/8.5.html)).  So if the error occurs 5 Lua call frames down, then the stripped down `debug.debug()` interpreter runs at call frame 6.  

The issue here with NodeMCU is that SDK tasks always start at call stack frame 0 and grow / release the frames until a return from stack frame 0 completes the task, allowing the next callback task to again execute from call-frame zero.  We would need the interpreter to run at stack frame 6 not 0.  This is going to be really difficult to achieve without botching the Lua architecture.  This needs quite a bit more though before I could consider doing this.  In the meantime we have a lot more valuable low hanging fruit to pick.

nwf, 2020-04-30:
One of the points of fragility in the C module is that `luaL_ref` can throw on OOM (if it fails to expand the Lua table that is the registry, e.g.).  Can we catch that exception in C, too?

TerryE, 2020-05-01:
> One of the points of fragility in the C module

@nwf ,  I hold a dissenting view here.   The underlying fragility is that is that people write apps that exhaust memory.  If you've run out of memory to the point that the application can't even grow registry (which should be a fairly small table if the application is correctly written), it is in this   Once an app exhausted RAM state, then it is f\*\*ked; it isn't going to recover. The best thing to do is to throw an error so that error reporting can give a decent traceback.

nwf, 2020-05-01:
An example is the proposed #2854 PR: the constructor began with a call to `malloc` and then `luaL_ref`'d a callback.  I *think* the solution there is to carefully stage the creation of a `userdata` with a `__gc` metamethod that unhooks things and to immediately install any `luaL_ref`'d result into said `userdata`, and I've suggested that change.

As the code was written, a failure of `luaL_ref` would have exacerbated an OOM situation by leaking the `ping_t` structure allocated at the beginning.

TerryE, 2020-05-02:
@nwf, Thanks Nathaniel.  I will do a review of the module and feed in comments.

TerryE, 2020-05-06:
> One of the points of fragility in the C module is that `luaL_ref` can throw on OOM
@nwf, I've been considering this issue.  This "throw on memory error" is the same for all C API calls which can allocate collectable resources such as extra stack, strings, tables or additional table entries and userdata.   In general resources created on the stack are cleaned up by the Lua CG after exiting the routine (and using a userdata this way is actually the simplest way of mallocing temporary data).  

The issue with creating resources in the registry is that these will persist if the routine throws an error for whatever reason, in that registry entries will persist even if an error is thrown, leading to memory leakage of dead resources.  Something that I propose to clean up in #1028, but you and I  should agree some best practice patterns for coding this.  But another peeve of mine if that a lot of our code tests for error values and error logic on return even though these call throw an error and will _never_ return an error value.

TerryE, 2020-05-08:
`node.setonerror()` (with no function argument) set the default option, which is to print the error, but also kick off a 2sec timer to restart the processor.  The delay is to ensure that error can print or be sent to output redirection.  If you are debugging and watching UART0, then you also have the option of doing a `node.setonerror(print)` which will print the error without restarting.

TerryE, 2020-05-10:
One of the complications here is that we need a unified error architecture across both the Lua 5.1 and 5.3 VMs so that the rules are the same for modules and independent of which Lua version used. I've had this challenge in other functional areas as well such as table and function handling.  A consequence of this is that for from Lua 5.1 being in "frozen" support, it is receiving a constant "bleed-in" of Lua 5.3 goodies.

TerryE, 2020-05-11:
@nwf, @HHHartmann,  I've set up `node.setonerror()` as discussed above, and also review all of the `lua_call()` uses in the modules. The majority just do the call and then immediately exit, so we can just do a global substitute in these for `luaL_pcallx()`.  There are a few (``,``, ``, ``, ``, `wifi`) which do a bit of cleanup after the call which need to be guarded by the pcallx status test as this now returns instead of an immediate panic.

I can include all of these changes as a single batch PR or alternatively just change one module on this PR (e.g. `tmr`)so that you can evaluate its behaviour and then do all of the others as a separate PR.  What are your thoughts?

TerryE, 2020-05-12:
Well this seems to working fine.  No more undiagnosed tracebacks in CBs and optionally being able to turn off panic reboot.
```Lua
> file.putcontents('f.lua', 'function f() g() end')
> dofile'f.lua'
> node.setonerror(print)
> tmr.create():alarm(1000,0,f)
> f.lua:1: attempt to call global 'g' (a nil value)
stack traceback:
	f.lua:1: in function <f.lua:1>

> node.setonerror()
> tmr.create():alarm(1000,0,f)
> f.lua:1: attempt to call global 'g' (a nil value)
stack traceback:
	f.lua:1: in function <f.lua:1>

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 31864, room 16 
...
```

nwf, 2020-05-12:
@TerryE: You may as well land the whole bunch in a single PR and we can go from there.  I *think* we are currently obligated to not merge anything to `dev` so that we can cut `master` soon (@marcelstoer, yes?).

marcelstoer, 2020-05-12:
> we are currently obligated to not merge anything to `dev` so that we can cut `master` soon

https://github.com/nodemcu/nodemcu-firmware#releases
https://github.com/nodemcu/nodemcu-firmware/issues/3066#issuecomment-626370485

TerryE, 2020-05-12:
@marcelstoer we are on the same page. A target date would be nice.

TerryE, 2020-05-13:
@wf wrote in offline email:

> @TerryE: You may as well land the whole bunch in a single PR and we can go from there. I think we are currently obligated to not merge anything to dev so that we can cut master soon.

I will formulate the PR on this basis.  Also since I am making changes to lauxlib and the modules to add the `luaL_pallx()` calls and l, I will also add the `luaL_unref2()` API as discussed [here in #1028](1028#issuecomment-625813440) and tidy up `tmr.c` as a worked example.

nwf, 2020-05-13:
Adding the `luaL_reref` API would also be nice; it would let us tidy up a lot of places. :)

TerryE, 2020-05-13:
Already done :smile:

TerryE, 2020-05-13:
@nwf, IIRC you are working on `mqtt`, so I haven't done the `luaL_pallx()` changes to this module

nwf, 2020-05-13:
That's fine; I can take those. :)

TerryE, 2020-05-14:
@nwf One note whilst I think on.  We don't want to convert all `lua_call()` instances, just those (and in the majority) that are executed via a callback, that is those where the execution path uses `lua_getstate()` to acquire the lua state.  The remainder (for example in `sjson`) are invoked from the Lua application and may already be protected by a `pcall()` so we don't want to turn these into a panic.

TerryE, 2020-05-14:
@nwf I have come across an issue in my testing and that is when the error is triggered memory exhaustion, for example:
```Lua
> a={}; function f() for i=1,3000 do a[i]=0 end end 
> f()
E:M 32784
E:M 32784
Lua error: 	not enough memory
> print(#a) a=nil  -- still got a in memory
2048
> collectgarbage();  print(node.heap())
40784
> tmr:create():alarm(500,0,f) -- now run out mem in a CB
> E:M 32784
E:M 32784
> -- this should have rebooted but it didn't
> a= nil; collectgarbage(); return node.heap()
40760
```
The issue is that the error handler is running at the deepest stack frame and I think that I need to run the GC before I try to create the closure for the post and do the reboot.

TerryE, 2020-05-14:
OK, that works.
```Lua
> a={}  collectgarbage();  print(node.heap())
40752
> tmr:create():alarm(500,0,f)
> E:M 32784
E:M 32784
out of memory

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)
...
```
It's a feature of Lua that event handlers are called on "out of memory" conditions, but at least `node.setonerror() is still honoured, so if you do the same with it set to print than you've lost the call stack but you can still try to diagnose what triggered the exhaustion.

pnkowl, 2020-05-16:
Given the above discussion of error handling, would it not be appropriate to add additional clarity to the module reference documentation?  When I read this, 

> Redirects the Lua interpreter output to a callback function. Optionally also prints it to the serial 
console.

should I not believe what is says?  There is no fine print, no mention or reference is made to what really happens.

I did search for "telnet" in issues.  

I believe I had already "_read_" this post (#3078), and it did not seem to apply -- that is to say, how could a module like telnet be offered without the caveat stated boldly that, in too many cases, errors will not be show?  

How could node.output() be offered without some referece to this issue?  

Now I certainly understand that fixing an issue is way more satisfying than documenting where things fall short of our expectations, but I believe that the documentation should accurately reflect real life behavior, and not what we wish or for what we have plans.  How to avoid the tl;dr? Perhaps have the module/method documentation page contain "td;lr" link(s) to separate pages which contain the caveats?  

BTW, I do not feel that pointing telnet and node.output() to this page satisfies the need.  Telnet, for example, might benefit, by having a description of each class of errors and whether they are printed or not, and an example of each class.  Alternatively, a companion pcall wrapper (executes the pcall, then manages the error handling/printing) could be referenced.  I could take a crack at all of these in isolation, but the current style of docs does not cater to this kind of thing... so more experienced folks should guide...

This all being said, the difference between having telnet in its present form versus not having it at all, is day and night.

TerryE, 2020-05-16:
@pnkowl, time to do an ow-oooom or whatever.  If you search the Issue and PR history then you will find 9 open and 57 closed threads which raise and discuss various aspects of this.  Sorting out this sort of architectural feature is hard.  Doing excellent documentation is hard.  We (and this means mostly me in this case) have been sorting these step by step, and that involves at times up to maybe 8 hrs development effort a day unpaid and pro bono.

An open issue is where we collaborate and get feedback when identifying and fixing issues.  The PR is where we post the change, and the "The code changes are reflected in the documentation at `docs/*`" checkbox is there for a reason.

In this case, the early implementations of output redirection were a bit of a botch. The runtime core handled redirection of `stdout` but `stderr` was just sent straight to the UART. Remember that using the net interface relies on heavy use of callbacks, so a lot of other features had to be put in place so that `stderr` content could also be reliably sent to the redirector.

This is community project, so all I can suggest here is that once the PR has been integrated into `dev`, then review the documentation and if you feel that you could improve this then raise the issue / PR and do this yourself.

pnkowl, 2020-05-16:
> then review the documentation and if you feel that you could improve this then raise the issue / PR and do this yourself.

Summary
- Whether it be dedicated volunteers or folks getting paid, rarely are things as we would like.  I am okay with "a bit of a botch" as long as the core reference docs reflect the actual state of the system.  
- in the dev branch docs, the node.output() "caution" not to print within the callback, is gone.
- in the master branch docs, the caution is present
- should it be expected that print is allowed in the callback now?  Based on what I see above, that seems unlikely (but what do I know)
- when doing some pcall investigation, I had the urge to add a print within the callback, but being forwarned, I did not (thanks)
- why take time to remove a caveat?   Why remove a helpful thought that may lead to a useful connection?
- if I were to be making PRs that have meaning to me, they would be to add more clarity, more hyperlinks, more examples for those that come next and have no pretentions of becoming experts. 
- before investing much time, it would be comforting to know that at least my sensibilities are aligned with the overall docs and development strategy.  I do not want to be in conflict over this.  At present, I do not get warm fuzzies that this type of PR would be kindly received.

Details follow (tl;dr)

So this is what I have proactively done as a result of the offered suggestion.

I decided that the PR should contain
- an additional "caution" regarding stderr in node.output of the docs and perhaps a helper function utilizing pcall that would surface the errors with the least extra typing and complication.
- a hyperlink to #3078 as the connection to more information

I then headed out to the docs (and started in master), did the fork, found the location to anchor the changes then realized dev is the place for PRs

So off to dev branch docs

Guess what?
- the "caution" block is gone
- the telnet example is gone
- reference to the telnet.lua exists with no hyperlink

With my limited skills I wade through History, Blame and Raw and can find no explanation as to why the "caution" block is gone (but my skills are limited).

This is not the only occurance of this type of removal in the node module.  

Perhaps a print in the callback is allowed now.  So
- I try "print in the callback" in master and it panics. So far, so good.
- I setup to try dev branch and telnet is broken (without LFS too), so I update issue #3108

But the basic question still remains.  How do new folks (or any folks for that matter) attach the tips, tricks, wisdom, and clarity where it is most likley to create meaningful connections from problems to solutions, especially for those who are, in fact, not experts, when it appears that at least some of these very connections are being systematically removed?

pnkowl, 2020-05-16:
> time to do an ow-oooom or whatever

the meaning of this escapes both me and google.

relax?

marcelstoer, 2020-05-16:
You can never have too much documentation. So, any correct and meaningful additions are welcome.

> it appears that at least some of these very connections are being systematically removed?

Thanks for bringing it to our attention but what makes you think they are _systematically_ removed? How about (documentation) bugs that just went unnoticed so far?

Git blame on `node.md` (for `node.output()`) leads straight to Terry's "Lua 5.1 to 5.3 realignement" PR #2836. It changed over a thousand lines of code and documentation. 

https://github.com/nodemcu/nodemcu-firmware/commit/14c1b8fd542beabba68646dab93f4862b364a4b3#diff-45fdccb9fe7b1f8ae504702b0a034d6e is the specific diff for the dropped admonition block. Whether the removal was intentional or not only Terry can explain.

TerryE, 2020-05-16:
In previous versions of NodeMCU, `node.output()` was calling was calling the Lua core recursively and there are internal health warnings not to do this, so it was a miracle that it worked at all.  Now output spools to a pipe which is emptied on a subsequent task which is posted to empty the pipe.  Actually printing inside a output CB does work, though this can easily cause a recursion loop and this will exhaust memory, so yes we probably should have kept the warning.  Our challenge is that we have a team of 1 doing all of this work.

pnkowl, 2020-05-17:
@ marcelstoer,

> what makes you think they are systematically removed?

Okay, so maybe those could be interpreted as "fighting words".  Let me try again. 

I see the core documentation as having 3 basic functions (1) to tell it like it is (2) to bridge from the past (3) to foster the interconnection of all the disparate pieces, all using the best technology available.

It seemed to me, that the dev branch documentation had greater brevity than master.  This is not a bad thing in itself, but it seemed to me that important connections had been lost.  It also seems that there were places where notes and cautions disappeared.  Notes and cautions are the evidence that blood has been spilled and IMO should not be discarded without cause or more blood will be spilled in the future.

So to be objective about this, let me take a brief walk down through the node module
- https://nodemcu.readthedocs.io/en/master/modules/node/#nodeoutput
- https://nodemcu.readthedocs.io/en/dev/modules/node/#nodeoutput

Differences
- node.input().  removed: Attention, This function only has an effect when invoked from a callback. Using it directly on the console does not work.
- node.output() removed this Caution. Do not attempt to print() or otherwise induce the Lua interpreter to produce output from within the callback function. Doing so results in infinite recursion, and leads to a watchdog-triggered restart.
- node.output() materially changed in a non backward compatible fashion
- telnet example removed and a non linked reference added.  
- node.startupcommand() added
- several spelling fixes

Please understand that I can be very directed in my activities.  Although it seems obvious now that node.output is significantly changed, it was not obvious hours ago.  It really should be made obvious.

Additionally, it would be helpful to have a set of actions required by end users when the new master appears, and for those attempting dev now.  It is not the same old thing but will, if all is done right, possibly break your application.  I have not looked very hard, but does this list exist? 

Verdict: "systematic" now seems over the top given the evidence presented.  

If requested, I could review all the docs, however, copy/paste/diff is likely not the most efficient way to accomplish this.  That being said, this seemed to yield a much smaller set than I expected

> It changed over a thousand lines of code and documentation.

Blame seems to have many more differences (as seen by the untrained), but that may be the nature of the beast (not side by side, but more a classic unix diff)

pnkowl, 2020-05-17:
@TerryE 

> Our challenge is that we have a team of 1 doing all of this work.

So how do we change that?  

Most of us have neither the skill, breadth of knowledge, nor the temperament to do the high value tasks.  That being said, I believe there must be a path to a system state where a non trivial number of time consuming and necessary tasks can be carved out that mortals can do, without taking more of your time, hearding all the new volunteer cats trying to help, than is saved,

Where does a volunteer start?  For example, I believe I know how to author a PR to put the cautions back and find the new telnet sample and make the link.

TerryE, 2020-05-19:
BTW, the above comments had absolutely nothing to do with this issue so I have moved them onto a separate one: #3101

chathurangawijetunge, 2020-05-28:
hi when will node.setonerror() be available in custom build online...?

nwf, 2020-05-28:
@chathurangawijetunge There's no better answer than "when it's ready".  @TerryE is moving heaven and earth with his changes to the Lua core of NodeMCU and we're all eagerly awaiting his work landing on `dev`.

TerryE, 2020-05-28:
@chathurangawijetunge, I've got this all working on my working branch, but we want to do the next cut to master before adding this PR.

TerryE, 2020-06-16:
PR has been merged.  On to next functional tranche :smile:

PR #3163 (2020-06-16): Implement panic call handling for all modules
Fixes #3078.  (Also #3141 and @KT819GM Gitter comment at  Jun 12 15:04)

Make sure all boxes are checked (add x inside the brackets) when you submit your contribution, remove this sentence before doing so.

- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [x] I have tested my contribution.
- [ ] The code changes are reflected in the documentation at `docs/*` [... cut]
closed 2020-06-16 by TerryE