nodemcu/nodemcu-firmware #3079: dev build breaks since add LiquidCrystal lua module (#2974)

HHHartmann, 2020-04-28:
All build steps pass

### Actual behavior
Build breaks due to Lua cheks and
since Lua 5.3  4 also the windows build breaks

### Test code
See https://travis-ci.org/github/nodemcu/nodemcu-firmware/builds/679883840

marcelstoer, 2020-04-29:
Weird...I don't see any CI build info in #2974. I'm sure we would otherwise have asked @seregaxvm to address this prior to merging.

seregaxvm, 2020-04-29:
Looks like there's a CI configuration issue. CI build is not run when submitting a new PR. At least when I do it.

marcelstoer, 2020-04-29:
> Looks like there's a CI configuration issue.

Definitely, yes. We see all the CI builds at https://travis-ci.org/github/nodemcu/nodemcu-firmware/pull_requests (i.e. they do run) but GitHub doesn't show the results anymore in the respective PR view.

Thanks for the quick fix!

marcelstoer, 2020-04-29:
@HHHartmann I created a separate issue for the Windows build: #3082

PR #3081 (2020-04-29): fix liquidcrystal luacheck warnings
Fixes #3079.

- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [x] I have thoroughly tested my contribution.
- [x] The code changes are reflected in the documentation at `docs/*`.

Fix liquidcrystal `unused argument` luacheck warnings.
closed 2020-04-29 by marcelstoer