nodemcu/nodemcu-firmware #3089: platform_gpio_register_intr_hook does not update interrupt hook pins

galjonsfigur, 2020-05-06:
While I was testing and cleaning up `softuart` module I discovered that serial receive from multiple pins does not work and only first registered object can receive. After some debugging I found that function `platform_gpio_register_intr_hook` should update interrupt hook but It doesn't do it. Because of that, other modules that uses this function (for example `rotary` module) also has this problem. I don't have any encodes around but is seems like only first registered rotary encoder would work (according to the documentation registering and using 3 encoders should be possible)

It seems like adding  the following lines before the `return` [here](https://github.com/nodemcu/nodemcu-firmware/blob/master/app/platform/platform.c#L306)
```
      ETS_GPIO_INTR_DISABLE();
      // This is a structure copy, so interrupts need to be disabled
      platform_gpio_hook = oh;
      ETS_GPIO_INTR_ENABLE();
```
should fix this issue.

### Expected behavior
`platform_gpio_register_intr_hook` should update interrupt hook

### Actual behavior
`platform_gpio_register_intr_hook` does not update interrupt hook

### Test code
```Lua
rotary.setup(0,1,2,3)
rotary.setup(1,4,5,6)
-- Only the first registered one seems to work
rotary.on(0, rotary.ALL, function (type, pos, when)
  print "Position=" .. pos .. " event type=" .. type .. " time=" .. when
end)
rotary.on(1, rotary.ALL, function (type, pos, when)
  print "Position=" .. pos .. " event type=" .. type .. " time=" .. when
end)
```
### NodeMCU startup banner

NodeMCU 3.0.0.0 
        branch: dev
        commit: 88a33afd6e750d9be3d9e47ba7b3920cd73829da
        release: 2.0.0-master_20170202 +509
        release DTS: 202005011052
        SSL: false
        build type: float
        LFS: 0x0 bytes total capacity
        modules: adc,bit,dht,file,gpio,i2c,mqtt,net,node,ow,rotary,softuart,spi,tmr,uart,wifi
 build 2020-05-06 13:52 powered by Lua 5.3.5 on SDK 3.0.1-dev(fce080e)
cannot open init.lua: 

### Hardware
NodeMCU Devkit V0.9 with ESP-12E

pjsg, 2020-05-06:
Good catch.

galjonsfigur, 2020-05-07:
Forgot to ask - should I make a PR to fix it or leave it up to you @pjsg ?

pjsg, 2020-05-07:
You should make a PR... Thank you

TerryE, 2020-05-09:
Having looked at this code, what this issue really exposes is that the approach to registering and registering interrupt hooks needs to have temporal update integrity because the GPIO ISR is currently concurrently to this update process.  

Given that this is an very infrequent (typically zero or a few calls during startup)  then the simplest and cleanest method of achieving this is to make `platform_gpio_hook` a pointer to the `struct gpio_hook` and append the `gpio_hook_entry` vector to this so that the gpio hook block is a single malloc unit.  This really simplifies the code since the register routine can simply malloc the new block, fill it in, update platform_gpio_hook and then free the old one.  Since word writes are atomic in fact we don't even need to mask GPIO interrupts.  

This removes a lot of complexity rather than adding to it, which is a good thing, IMO.

Incidentally another issue is that this registration task can fail and so returns a boolean success status.  However _all_ of the current modules that do this registration `rotary`, `NmraDcc`, `hx711` and  `softuart` treat this as a `void` and assume success. :grimacing:

marcelstoer, 2020-05-10:
Sorry Terry, missed this comment. I just merged the PR (already approved by @pjsg ).

TerryE, 2020-05-10:
@marcelstoer, that's fine. I can raise this as part of the module tidy up m

PR #3091 (2020-05-10): Fix for #3089 - update hook after updating gpio bits
Fixes #3089 

- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [x] I have thoroughly tested my contribution.
- [x] The code changes are reflected in the documentation at `docs/*`.

This little patch should fix problem that occurs  when modules like `rotary` or `softuart` hooks to gpio ISR more than one time.

If anyone could test if using more than on [... cut]
commit 2e58471e (2020-05-10): Update hook after updating GPIO bits (#3091)

Fixes #3089
closed 2020-06-10 by nwf