nodemcu/nodemcu-firmware #3092: Build Failure when uncommenting user_config.h[#define DISABLE_STARTUP_BANNER]

jmd13391, 2020-05-08:
### Expected behavior
Firmware build completes successfully with startup banner is disabled.

### Actual behavior
Uncommenting user_config.h:#define DISABLE_STARTUP_BANNER causes a NodeMCU firmware build failure. Simply restoring the comment produces a successful build.

user_config.h:
=-=-=-=
.
.
.
// At start-up firmware details like:
//
// NodeMCU 3.0.1.0
//         branch:
//         commit:
//         release:
//         release DTS:
//         SSL: false
//         build type: integer
//         LFS: 0x0
//         modules: file,gpio,net,node,rtctime,sntp,tmr,uart,wifi
//  build 2020-01-27 17:39 powered by Lua 5.1.4 on SDK 3.0.2(824dc80)
//
// will be printed to serial console.  While it's mandatory for bug reports
// and good for development, it may be unwanted for non-interactive serial
// devices.

#define DISABLE_STARTUP_BANNER
.
.
.
=-=-=-=

[Windows 10 Pro (10.0.18363), Docker for Windows (2.2.0.5)]

docker run --rm -it -e "IMAGE_NAME=test1" -v "//c/Users/Joe/Google Drive/test1/nodemcu-firmware":/opt/nodemcu-firmware marcelstoer/nodemcu-build build
.
.
.
CC app/lua/lua.c
In file included from lua.c:11:0:
../lua53/lua.c:117:13: error: 'print_version' defined but not used [-Werror=unused-function]
 static void print_version (lua_State *L) {
             ^
cc1: all warnings being treated as errors
../../Makefile:422: recipe for target '.output/eagle/debug/obj/lua.o' failed
make[2]: *** [.output/eagle/debug/obj/lua.o] Error 1
make[2]: Leaving directory '/opt/nodemcu-firmware/app/lua'
../Makefile:380: recipe for target '.subdirs' failed
make[1]: *** [.subdirs] Error 2
make[1]: Leaving directory '/opt/nodemcu-firmware/app'
Makefile:380: recipe for target '.subdirs' failed
make: *** [.subdirs] Error 2
make: Leaving directory '/opt/nodemcu-firmware'

### Test code
n/a

### NodeMCU startup banner
NodeMCU 3.0.0.0 built with Docker provided by frightanic.com
	branch: dev
	commit: 88a33afd6e750d9be3d9e47ba7b3920cd73829da
	release: 2.0.0-master_20170202 +509
	release DTS: 202005011052
	SSL: false
	build type: float
	LFS: 0x10000 bytes total capacity
	modules: bit,file,gpio,i2c,net,node,tmr,uart,wifi
 build 2020-05-08 08:37 powered by Lua 5.1.4 on SDK 3.0.1-dev(fce080e)

### Hardware
Amica ESP8266 12E Development Board

TerryE, 2020-05-08:
The workaround is to add a `#ifdef DISABLE_STARTUP_BANNER` around the static function declaration.  I'll add this to my next PR.

marcelstoer, 2020-05-09:
> I'll add this to my next PR.

I personally prefer to have either a direct commit to `dev` or a dedicated PR referencing this issue. This will allow us to more easily track changes.

I'll fix this if you like.

jmd13391, 2020-05-10:
@TerryE @marcelstoer

> The workaround is to add a #ifdef DISABLE_STARTUP_BANNER around the static function declaration.

I believe the workaround is actually to add a `#ifndef DISABLE_STARTUP_BANNER` around the static function declaration.

jpeletier, 2020-05-11:
There you go!

PR #3103 (2020-05-12): ESP8266: fix compilation error when disabling startup banner
Fixes #3092 
Make sure all boxes are checked (add x inside the brackets) when you submit your contribution, remove this sentence before doing so.

- [X] This PR is for the `dev` branch rather than for `master`.
- [X] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [X] I have thoroughly tested my contribution.
- [X] The code changes are reflected in the documentation at `docs/*`.

Fixes #3092
closed 2020-05-12 by TerryE