nodemcu/nodemcu-firmware #3096: SPIFFs File Corruption

pnkowl, 2020-05-09:
### Expected behavior
- file contents never change as a result of other file adds

### Actual behavior
- files go missing
- files are truncated
- files are altered

Supplied cases show this occurs when all SPIFFs pages have been written once.
For a 4MB module, this usually takes awhile.

When no SPIFFs pages are free, but are either used or deleted, a garbage collection
operation must be performed to free the deleted pages.  Sometimes this results in
- truncated files (overwritten pages)
- overwritten index or object lookup pages (data is there but filename page is gone)
- modified pages (data file bytes are changed at the beginning or end of a page boundary)

### Test code
Provide a [Minimal, Complete, and Verifiable example](http://stackoverflow.com/help/mcve) which will reproduce the problem.
```
image and file
esp8266_21_4M_06.bin  -- esp8266 4M image with SPIFFS okay, all that is important is SPIFFS layout
20200505_test009.txt  -- add this 32kbyte file to SPIFFS which then reveals an issue with SPIFFS garbage collection?
esp8266_21_4M_07.bin  -- resultant image with SPIFFS corruption

Basic process
1. flash "06" image to 4M esp8266
2. upload 32kbyte file to SPIFFS (FTP or serial, corrupts either way)
3. valid data is overwritten (see below for details)
```
### NodeMCU startup banner
You MUST include the firmware startup banner to describe the version you are using. We reserve the right to immediately close any bug that doesn't.

Nodemcu-build.com provided by frightanic.com
        branch: master
        commit: 3d917850180f67adc7f2c6b5d00f27c152e7194c
        release:
        release DTS: 202002231746
        SSL: false
        build type: float
        LFS: 0x40000
        modules: adc,bit,cron,dht,enduser_setup,file,gpio,mqtt,net,node,ow,pwm2,rtcmem,rtctime,sntp,struct,tmr,uart,wifi
 build 2020-05-01 13:41 powered by Lua 5.1.4 on SDK 3.0.1-dev(fce080e)

### Hardware
Describe which ESP8266/ESP32 device you use and document any special hardware setup
required to reproduce the problem.

  Wemos D1 mini (4MB), aka esp-21 ref. below
  Nodemcu V0.9 (4MB), aka esp-01 ref. below

### Other Resources Used

FTP
  https://nodemcu.readthedocs.io/en/master/lua-modules/ftpserver/
  with several added hooks
  use default user and pwd: "test","12345"

Serial Interface Software
  Lua Loader 0.91

Flasher, Unloader
  esptool.py

Local LUA code
  as listed below to help reveal issue but does not cause it

### Detailed Notes

How to find the "issue" locations?  
[... 204 lines cut ...]
   but should otherwise work fine.  If this is a problem,
   please comment accordingly
```
---------------------------------------------------------------------------------
Attachments
```
SPIFFS files used
  20200505_test023.txt   1048576 bytes
  20200505_test006.txt   1048576 bytes
  20200505_test007.txt    327680 bytes
  20200505_test008.txt    262144 bytes
  20200505_test009.txt     32768 bytes  appears to break SPIFFS
  LLbin.lua                  639 bytes
  ide_c_local.lua            677 bytes
  ide_wifi_cfg.lua           654 bytes
  init_0.lua                1177 bytes

Results Files relevant and captured (others avialable if needed)
  esp8266_21_4M_06.bin  still some untouched pages left, all okay, no garbage collection yet
  esp8266_21_4M_07.bin  32k byte add with FTP (wifi), captures corrupted state
  esp8266_21_4M_08.bin  32k byte add with lualoader (serial), captures corrupted state 

```

[ticket files.zip](https://github.com/nodemcu/nodemcu-firmware/files/4602528/ticket.files.zip)

TerryE, 2020-05-09:
I have just tried:
```Lua
function wf(n,r)
  local f,s = file.open(n..'.txt','w'), ('0123456789abcdef'):rep(64)
  for i=1,32 do f:write(s) end
  file.remove(r..'.txt')
end
function cf()
  local t = tmr.now(); wf(c, c-6); c=c+1
  print("%u %8.2f" % {c, (tmr.now() - t)/1000000})
  if c<l then node.task.post(cf) end
end
c = 0 l = 500 cf()
```
and this chunders away happily recycling the FS and doing GC without any obvious errors.

TerryE, 2020-05-09:
What does 
```Lua
for k,v in pairs(node.getpartitiontable()) do print("%s=0x%x" % {k,v}) end
``` 
printout?

pnkowl, 2020-05-09:
`For the SPIFFS almost "no free and never touched page" case (no corruption yet)....
> for k,v in pairs(node.getpartitiontable()) do print("%s=0x%x" % {k,v}) end
stdin:1: attempt to perform arithmetic on a string value
stack traceback:
	stdin:1: in main chunk
> for k,v in pairs(node.getpartitiontable()) do print(k,v) end
lfs_size	262144
spiffs_addr	770048
lfs_addr	507904
spiffs_size	3424256`

pnkowl, 2020-05-09:
Single page writes do not seem to cause the problem.  I repeated the presented test case with smaller files.  Corruption occurred if I approached the cliff from my stated starting point (image 06) with three 8k byte files each.  I then reduced this to two 8ks, followed by one 4k, followed by two 2ks and no corruption occurred.  Then followed with one 8k for good measure and all was fine.  It appears the edge case is when a larger file (say 8k or more) forces the gc.

pnkowl, 2020-05-09:
If I recall, block size is 8k (the eraseable unit).

TerryE, 2020-05-09:
Try aligning the FS on a 1Mb boundary:
```Lua
node.getpartitiontable{spiffs_addr=0x100000,spiffs_size=0x300000}
```
This will set the SPIFFS partition to be the last 3Mb and reboot the module.

pnkowl, 2020-05-09:
Thanks for the try.  No Go Though

Here is a more credible MVCE of the issue
1. restore firmware: python c:\esptool\esptool.py -b 115200 --port COM5 write_flash --flash_freq 80m 0x000000 esp8266_21_4M_06.bin
2. for k,v in pairs(node.getpartitiontable()) do print(k,v) end
3. node.setpartitiontable{spiffs_addr=0x100000,spiffs_size=0x300000}
4. file.format()
5. upload f0509d.lua 
6. dofile("f0509d.lua")
7. f0509(10,99) -- what it does: writes a 32k file, checks it, loops to the next file,... do until done, then check them all again (if it gets that far).  Start and stop file name indices need to be 2 digits, that is, in the range of (10,99).  With a small firmware/LFS build 90x32k may not be enough, so, as needed execute again (appends the 90 created files with another 32k each)
8, still fails for me on both my test ESP8266s

Find attached a serial console listing of the above process and f0509d.lua.
[200509.zip](https://github.com/nodemcu/nodemcu-firmware/files/4604387/200509.zip)

TerryE, 2020-05-09:
Thanks. I'll give it a try

pnkowl, 2020-05-10:
The prior case (200509.zip) comes pretty close to filling SPIFFS.  Here is a more representative case of real life: several files close to the beginning of the FS, then a bunch of deleted pages and finally several larger files (representing LFS uploads).  Few "free" pages are left (plus a few used pages and a whole bunch of deleted pages in the middle).

process flow
```
1. node.setpartitiontable{spiffs_addr=0x100000,spiffs_size=0x300000}
2. file.format()
3. restart (small changes make material differences in gc operation, do this step)
4. f0509d.lua upload (unsure if this file's presence in SPIFFS makes a difference in the outcome)
5. dofile("f0509d.lua")
6. f0509(10,91)
7. function myremove(i0,i1) for i=i0,i1 do a0="<"..i..">"; b="20200509_test"..i..".txt"; print("remove",b,file.remove(b)) end end
8. myremove(12,88)
9. f0509(89,93)
10. for k,v in pairs(filelist()) do l = string.format("%-15s",k) print(l.."   "..v.." bytes") end
11. function f(a) i=0; local src = file.open(a, "r"); if src then local line; repeat line = src:read(); if line then l=string.len(line); i=i+l; print(l,i); if(i>1953300) then print(line) end end until line == nil src:close(); src = nil end end
12. f("20200509_test90.txt")
13. for k,v in pairs(node.getpartitiontable()) do print(k,v) end
14. =file.fsinfo()
```
Of greatest interest in the attached listing
```
> f0509(89,99)
create	20200509_test89.txt
check	20200509_test89.txt	65536
create	20200509_test90.txt
check	20200509_test90.txt	12288  *** should be 64k like its neighbors ***
create	20200509_test91.txt
check	20200509_test91.txt	65536

> for k,v in pairs(file.list()) do l = string.format("%-15s",k) print(l.."   "..v.." bytes") end
20200509_test10.txt   32768 bytes
20200509_test11.txt   32768 bytes
20200509_test89.txt   65536 bytes
20200509_test90.txt   65536 bytes  *** FS thinks the file is 64k ***
20200509_test91.txt   65536 bytes
20200509_test92.txt   32768 bytes
20200509_test93.txt   32768 bytes

> function f(a) i=0; local src = file.open(a, "r"); if src then local line; repeat line = src:read(); if line then l=string.len(line); i=i+l; print(l,i); if(i>1953300) then print(line) end end until line == nil src:close(); src = nil end end
> f("20200509_test90.txt")
1024	1024
1024	2048
1024	3072
1024	4096
1024	5120
1024	6144
1024	7168
1024	8192
1024	9216
1024	10240
1024	11264
1024	12288  *** inspect again, file is way less than 64k
> 
```

pnkowl, 2020-05-10:
Oops, forgot the attachement.
[200509b.zip](https://github.com/nodemcu/nodemcu-firmware/files/4605078/200509b.zip)

TerryE, 2020-06-06:
See also #3147.

closed 2021-06-20 by stale[bot]