nodemcu/nodemcu-firmware #3097: Some modules _must_ be included in the firmware

TerryE, 2020-05-09:
### Expected behaviour

`app/include/user_modules.h` defines which modules are to included in the build.  Commenting out the relevant defines removes the modules from the build.  The firmware should then build and run correctly, with the specified list of modules included.

### Actual behaviour

Some core Lua modules (`coroutine`, `debug`, `string`, `math`, `table`) and core NodeMCU modules (`node`, `uart`) _must_ be in the build for the runtime to work.  We shouldn't be able to disable the define for these.

### Suggested Fix.

-  Move the `LUA_USE_MODULES_NODE` and `LUA_USE_MODULES_UART` defines from the `user_modules.h` file into the respective C files.  
-  Or if we don't want to break the CloudBuilder use a `#ifndef` gated `#define` in the module to force its inclusion. @marcelstoer, what is your preference here?

#### Note

Some other modules that you might think are mandatory (`file`, `net`, `wifi`) can be omitted and the firmware will build, though where the developer can use it is another issue.

marcelstoer, 2020-05-09:
It occurred to me occasionally that I should enable `node` and `uart` by default (already the case) _without an option to remove the selection_. Just never done that yet. 

> Or if we don't want to break the CloudBuilder 

Please don't. I also need/want to support the frozen 1.5.x branch and I'd hate to have my build scripts treat that branch differently.

jpeletier, 2020-05-11:
I have been looking at this and found the following:

`LUA_USE_MODULES_NODE`, `LUA_USE_MODULES_UART` and `LUA_USE_MODULES_FILE` are defined in `user_modules.h` but never read in the code. This means their value is ignored and their respective modules are always compiled in regardless of `user_modules.h`

I could not find entries in `user_modules.h` for `coroutine`, `debug`, `string`, `math` or `table`

So maybe this bug doesn't exist?

TerryE, 2020-05-12:
@jpeletier Javier not ignored but hidden deeply.  This is Johny Mattsson's make magic.  Have a look at the macros in `module.h` and `grep "_selected" app/mapfile` -- for esp8266 builds at least.  Having the define set changes section name of the `ROTable_entry` generated by the `NODEMCU_MACRO()` declaration and this is use in the link to determine which modules are included in the build.

closed 2021-06-20 by stale[bot]