nodemcu/nodemcu-firmware #3105: gpio.serout() fails

Ubiveritas, 2020-05-12:
### Expected behavior
gpio.serout() should toggle PIN with different pulse width by a series of given durations as long as the overall duration does not exceed a limit, then call the callback function

### Actual behavior
With given 27 durations an output is generated and the callback function is called as expected. With a given series of 28 durations, no output is generated (observed with the logic analyzer) and callback function is not called. Overall duration of all single toggle width is around 7,5 ms (measured for successfull output).

### Test code
```Lua
pin = 1 --GPIO05
zero=250 --us
one=500 --us
gpio.mode(pin,gpio.OUTPUT,gpio.PULLUP)
print('Start sequence')
gpio.serout(pin,gpio.HIGH,{zero,zero,zero,zero,one,one,one,one,zero,zero,zero,zero,zero,zero,zero,zero,zero,zero,zero,zero,zero,zero,zero,zero,zero,zero,zero,zero},1,function() print("done") end) -- will not complete
```
### NodeMCU startup banner
NodeMCU 3.0.0.0 built on nodemcu-build.com provided by frightanic.com
	branch: dev
	commit: 6e5edf434005f2bd4ad380fd448bc8653075a009
	release: 
	release DTS: 202005101836
	SSL: false
	build type: float
	LFS: 0x0 bytes total capacity
	modules: bit,bme280,file,gpio,gpio_pulse,http,i2c,net,node,pwm,rotary,rtctime,tmr,uart,wifi,ws2812,ws2812_effects
 build 2020-05-11 16:07 powered by Lua 5.1.4 on SDK 3.0.1-dev(fce080e)

### Hardware
NodeMCU Lua Lolin V3 Modul with ESP8266 12F
Logic Analyzer attached to PIN1 sampling with 50kHz

vsky279, 2020-05-12:
Hi @Ubiveritas,
I cannot reproduce the issue you are reporting.
I am using the same module setup and recent `dev` branch.
My hw is ESP8266 ESP-01 so I use GPIO02, PIN 4.
I needed to increase times by 5 so the LED blink is observable by eyes. I don't assume shorter interval would change anything as providing callback function means the `serout` is going asynchronous.
It works with parameter length 28 as well as 27. With 27 intervals the final state of the GPIO is high which is correct for odd number of intervals.

Try the test with LED (and longer interval). Maybe your logic analyzer is not wired correctly?

Ubiveritas, 2020-05-14:
Hi @vsky279,

thanks for trying to reproduce this case. I tried to verify that it works with your setup (Output at PIN4 with 5 times the duration and exactly the same comment at the end of the line). This works, but I found out that when I add a few more durations, it has no output again and does not call the callback:

```lua
pin = 4 --GPIO05
zero=125000 --us
one=250000 --us
gpio.mode(pin,gpio.OUTPUT,gpio.PULLUP)
print('Start sequence')
gpio.serout(pin,gpio.HIGH,{zero,zero,zero,zero,one,one,one,one,zero,zero,zero,zero,zero,zero,zero,zero,zero,zero,zero,zero,zero,zero,zero,zero,zero,zero,zero,zero,zero,zero,zero,zero,zero,zero,zero},1,function() print("done") end)-- will not complete
```
When I reduce the number of durations it works again. I played around with length of "zero" and "one" durations but this does not have any influence.
I also disconnected all other hardware and the logic analyzer which did not change it. Logic analyzer has a high impedance input and shout not interfere.

Regards Ubiveritas, holding a PPL(A)

vsky279, 2020-05-14:
Hi @Ubiveritas, I tested your code and get this:
```
dofile("test3.lua")
Start sequence
> done
``` 
So I doubled the sequence, one has to wait, but in the end I get the `done` output again. This is indeed very strange.
What does your logic analyzer say? No signal?
Maybe trying with another piece of hw?

TerryE, 2020-05-14:
@vsky279, Lukas since you have the H/W to test this can I leave it to you to help track it down.  Email or WhatApp me is you need any support.

Ubiveritas, 2020-05-15:
Hi @vsky279,

I found out the reason: It is the length of the transfered line of code.
I use ESPlorer software v0.2.0-rc5 @ 115200 to save the code to the board and this fails when the total chunk length of a line (in this example the last line with gpio.serout()) exceeds a limit.
Finally I was able to add another duration or two, when I had changed the comment at the end of the line which led me to the solution. Breaking the command into several lines of code by newlines eliminated the fail. Searching the web and documentation I found no hint about limited line length.

Thanks for investigating.

Ubi

TerryE, 2020-05-15:
The issue here is that ESPlorer seems to use a different algorithm for sending binary files (which it does in fixed 126 byte chunks IIRC, and sending text files where it will send the line one at a time to the uart so for the line `print('Start sequence')`, it sends:
```Lua
w([==[print('Start sequence')]==]);
```
So each line contains the 14 byte bookend overhead of `w([==[]==]);\n`  Character bursts of more than 128bytes will overrun the UART H/W fifo.  This is a H/W limitation rather than anything to do with Lua, and what this means is that if any source lines are longer than ~110 bytes, then the transfer will fail, as will any use of the `[==[` + `]==]` delimiters.

So this is an ESPlorer download limitation that an issue for here.  I am therefore closing this issue.

Thanks @vsky279, @Ubiveritas

closed 2020-05-15 by TerryE

HHHartmann, 2020-05-15:
I'm pretty sure that of you use the upload function of ESPlorer where you select a file to upload works also for lua files. I suspect that this is a issue of you upload it from the editor only. Try to reduce the transmission speed. That may well help.