nodemcu/nodemcu-firmware #3106: cannot compile dev build with #define PMSLEEP_ENABLE

tibinoo, 2020-05-14:
I'd like to compile latest dev build and use wifi-suspend, i changed #define PMSLEEP_ENABLE in user_config.h, but compilation finihed with error:
Could anyone try it?

CC app/modules/wifi.c
wifi.c: In function 'wifi_pmSleep_suspend_CB':
wifi.c:441:5: error: too few arguments to function 'luaL_unref'
     luaL_unref(L, wifi_suspend_cb_ref); // remove suspend callback from LUA_REGISTRY
     ^
In file included from wifi.c:6:0:
../lua/lauxlib.h:75:14: note: declared here
 LUALIB_API void (luaL_unref) (lua_State *L, int t, int ref);
              ^
../../Makefile:422: recipe for target '.output/eagle/debug/obj/wifi.o' failed
make[2]: *** [.output/eagle/debug/obj/wifi.o] Error 1

TerryE, 2020-05-14:
@tibinoo, you should stick to posting guidelines, but in this case mea culpa.  Line 441 of wifi.c should read `luaL_unref(L, LUA_REGISTRYINDEX, wifi_suspend_cb_ref);` that is the second parameter is missing. you can apply this workaround manually, and I will fix this on my next PR.

(This used to use the Lua5.1 macro `lua_unref()` which has been dropped from the Lua 5.3 API so I went through and converted them all, but forgot to add the extra param in this case, but didn't pick up the error as this code is skipped unless `PMSLEEP_ENABLE` is defined.

HHHartmann, 2020-05-15:
I feel that this should be fixed before the master drop.
That's what the quiet period is for as I understand it.

marcelstoer, 2020-05-15:
> I will fix this on my next PR

Thanks Terry!

I know, if _you_ fix it you can do this in which ever way _you_ like. However, I and maybe others would really appreciate if you could push 1 commit per issue rather than folding changes for many issues into a single commit/PR. Can't you switch back to `dev`, fix this, push it, and then rebase your current working branch onto `dev`? That's usually absolutely painless and quick.

> I feel that this should be fixed before the master drop.

👍 fully agree

TerryE, 2020-05-15:
Fixed in 50fcd72

commit 50fcd72a (2020-05-15): Update wifi..c to fix #3106
closed 2020-05-15 by TerryE