nodemcu/nodemcu-firmware #3119: error: implicit declaration of function 'espconn_ssl_delete'

SourceLink, 2020-05-20:
When making a make in the local linux environment without any changes, the following problems occur:

```
CC app/mbedtls/app/espconn_secure.c
espconn_secure.c: In function 'espconn_secure_delete':
espconn_secure.c:381:10: error: implicit declaration of function 'espconn_ssl_delete' [-Werror=implicit-function-declaration]
  error = espconn_ssl_delete(espconn);
```

nwf, 2020-05-23:
You've not said what branch or git revision you're on.  That symbol was removed as of f5aa0af5d4aec3ce5b88d537ff949abd9c0d5721, and espconn_secure.c was updated at the same time.

SourceLink, 2020-05-25:
master branch

SourceLink, 2020-05-25:
It's just so weird.

The version I got using the following command:

```
git clone https://github.com/nodemcu/nodemcu-firmware.git
```

nwf, 2020-05-25:
Works for me, even with SSL turned on:

```
$ git clone https://github.com/nodemcu/nodemcu-firmware.git nodemcu-firmware-master
$ cd nodemcu-firmware-master
$ git describe
3.0-master_20190907-2-g3d917850
$ git diff
diff --git a/app/include/user_config.h b/app/include/user_config.h
index 4866e6a9..13e2fde3 100644
--- a/app/include/user_config.h
+++ b/app/include/user_config.h
@@ -93,7 +93,7 @@
 // The SHA1 and MD5 function are implemented in the ROM BIOS. The MD2 and SHA2
 // are by firmware code, and can be enabled if you need this functionality.
 
-//#define CLIENT_SSL_ENABLE
+#define CLIENT_SSL_ENABLE
 //#define MD2_ENABLE
 #define SHA2_ENABLE
 #define SSL_BUFFER_SIZE 4096
diff --git a/app/include/user_modules.h b/app/include/user_modules.h
index 0b28dbe7..618e488b 100644
--- a/app/include/user_modules.h
+++ b/app/include/user_modules.h
@@ -62,7 +62,7 @@
 //#define LUA_USE_MODULES_SWITEC
 //#define LUA_USE_MODULES_TCS34725
 //#define LUA_USE_MODULES_TM1829
-//#define LUA_USE_MODULES_TLS
+#define LUA_USE_MODULES_TLS
 #define LUA_USE_MODULES_TMR
 //#define LUA_USE_MODULES_TSL2561
 #define LUA_USE_MODULES_UART
$ docker run -u `id -u`:`id -g` --rm -ti -e IMAGE_NAME=$(git rev-parse --abbrev-ref HEAD) -v `pwd`:/opt/nodemcu-firmware docker-nodemcu-build build
[...]
make: Leaving directory '/opt/nodemcu-firmware'
$ ls bin
0x00000.bin  0x10000.bin  nodemcu_float_master.bin  nodemcu_float_master.map
```

Please check that your tree is clean.

SourceLink, 2020-05-29:
```
➜  nodemcu-firmware-master git:(master) git describe 
3.0-master_20190907-2-g3d91785
```

Local compilation still has that problem.

```
espconn_secure.c: In function 'espconn_secure_delete':
espconn_secure.c:381:10: error: implicit declaration of function 'espconn_ssl_delete' [-Werror=implicit-function-declaration]
  error = espconn_ssl_delete(espconn);
```

nwf, 2020-05-29:
Sigh.  I see.  Your compiler is being more stringent than mine for reasons unknown.  Either use `dev` until the next master cut (which should be any day now; see #3066) or, somewhat hackishly, add `extern sint8 espconn_ssl_delete(struct espconn*);` above `espconn_secure_delete` in `espconn_secure.c`.

This is further evidence in support of #3006, but thankfully this particular bit of destruction has already landed on `dev`.

closed 2020-05-29 by marcelstoer