nodemcu/nodemcu-firmware #3120: ledc.channel:reset() only working once

chipec84, 2020-05-20:
### Expected behavior
channel:reset() should reset the timer and trigger a new pwm when the pulse count reaches 200
### Actual behavior
By doing measurements with the oscilloscope, channel:reset() only triggers the reset once (at 200 pulse) but never again although le pulse count is working properly.

### Test code
```Lua
pinOut=15
pinPulseInput = 12
thr0 = 200
thr1 = -4
llim = -6
hlim = 220

pwm=ledc.newChannel({
  gpio=pinOut,
  bits=ledc.TIMER_13_BIT,
  mode=ledc.HIGH_SPEED,
  timer=ledc.TIMER_0,
  channel=ledc.CHANNEL_0,
  frequency=100,
  duty=4096
})

function onPulseCnt(unit, isThr0, isThr1, isLLim, isHLim, isZero)
  print("Pulse count:" .. pcnt:getCnt())
  if isThr0 then 
    pwm:reset()
    print("thr0 "..thr0.." triggered event") 
    pcnt:clear() 
  end 
end

pcnt = pulsecnt.create(0, onPulseCnt)

pcnt:chan0Config(
  pinPulseInput,
  pulsecnt.PCNT_PIN_NOT_USED,
  pulsecnt.PCNT_COUNT_DIS,
  pulsecnt.PCNT_COUNT_INC, 
  pulsecnt.PCNT_MODE_KEEP,
  pulsecnt.PCNT_MODE_KEEP, 
  llim, 
  hlim  
)

--filter spikes
pcnt:setFilter(1023)
-- Clear counting
pcnt:clear()
-- Set counter threshold for low and high
pcnt:setThres(thr0, thr1)

```
### NodeMCU startup banner
branch: dev-esp32
	commit: 189fe9646f1bf916d8127bd63ee7a453bb795443

### Hardware
ESP32-WROOM-32

chilipeppr, 2020-05-20:
Love that you're using the pulse counter library I wrote. It looks like you're trying to generate PWM signals that turn off at exactly a pulse count. I'm guessing you won't like how it works as the timing of the callbacks and processing through Lua to then turn off the PWM is going to be pretty slow for any accuracy. So maybe that's what you mean in your post? Instead, you should be using the RMT hardware to generate your exact output pulses instead of the pulsecnt approach. Depends on whether you are ok with a mushy solution or not.

If instead you just mean the pulse counter stops counting, I never saw that in my use of it. Are you ever turning your PWM signal back on such that you would hit threshold 0 again? From looking at the code you never turn PWM back on, so you would never get the isThr0 again.

chilipeppr, 2020-05-20:
Hmm. I might have misunderstood your question. Perhaps you are just referring to the ledc library having an issue on reset(), not anything to do with the pulsecnt library. In that case, I'm not sure :reset() does anything other than just start the PWM again, which would mean nothing in your code.

chipec84, 2020-05-21:
Thank you, The issue I have is only with ledc. PulseCount is working fantastic, you did a great job on that library. I am trying to make a light AC dimmer. there is a pulse count everytime I get a zero crossing point of the AC 230v 50HZ. I need to keep ledc in sync with that slightly moving 50HZ. All I want is to restart the 100HZ ledc pwm every 200 (will probably be 4 in the end) zero crossing to keep it in sync with the 230v AC. It seems to me ledc.channel:reset() is suppose to start a fresh PWM by reseting the timer.

chilipeppr, 2020-05-21:
Have you tried pause() and resume()?

chipec84, 2020-05-21:
yes I did, it does not fix my problem

marcelstoer, 2020-05-21:
> I am trying to make a light AC dimmer.

-> #3023, in the making, bingo?

chipec84, 2020-05-21:
Great news, when will it be released?
Regardless, don’t you think LEDC.reset() should work?

marcelstoer, 2020-05-21:
> when will it be released?

@jpeletier stated that he'd like to address the review feedback.

> don’t you think LEDC.reset() should work?

I leave that for @chilipeppr to address.

chilipeppr, 2020-05-21:
My gut says that this is an ESP-IDF thing, beyond the scope of NodeMCU. If you take a look at the code for the ledc module, it is literally just a pass-through call to Espressif's code. I'd look deeper at these docs https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/ledc.html 

Maybe try creating a new channel and deallocate the old one as a heavier way of recreating a timer? If that's too slow of an operation, maybe have a 2nd timer waiting in the wings already allocated but not started?

Here's the code from the module. It's not doing anything special.
```C
static int lledc_timer_rst( lua_State *L ) {
  ledc_channel * channel = (ledc_channel*)luaL_checkudata(L, 1, "ledc.channel");

  esp_err_t err = ledc_timer_rst(channel->mode, channel->timer);
  if(err != ESP_OK)
    return luaL_error (L, "reset failed, code %d", err);

  return 1;
}

static int lledc_timer_pause( lua_State *L ) {
  ledc_channel * channel = (ledc_channel*)luaL_checkudata(L, 1, "ledc.channel");

  esp_err_t err = ledc_timer_pause(channel->mode, channel->timer);
  if(err != ESP_OK)
    return luaL_error (L, "pause failed, code %d", err);

  return 1;
}

static int lledc_timer_resume( lua_State *L ) {
  ledc_channel * channel = (ledc_channel*)luaL_checkudata(L, 1, "ledc.channel");

  esp_err_t err = ledc_timer_resume(channel->mode, channel->timer);
  if(err != ESP_OK)
    return luaL_error (L, "resume failed, code %d", err);

  return 1;
}

static int lledc_stop( lua_State *L ) {
  ledc_channel * channel = (ledc_channel*)luaL_checkudata(L, 1, "ledc.channel");

  int idleLevel = luaL_checkint (L, 2);
  luaL_argcheck(L, idleLevel >= 0 && idleLevel <= 1, 1, "Invalid idle level");

  esp_err_t err = ledc_stop(channel->mode, channel->channel, idleLevel);
  if(err != ESP_OK)
    return luaL_error (L, "stop failed, code %d", err);

  return 1;
}
```

chipec84, 2020-05-21:
Thank you @chilipeppr, it seems ledc is definitely only passing infos straight down to idf.

I tried two things and both with no luck:
1. Stop the first pwm and immediately start another one on a different timer and channel.
2. Same thing but each pwm on a different pin

Every new pwm seem to start from where the previous one left off, no reset.

marcelstoer, 2020-05-21:
@chipec84 did you try bumping ESP-IDF to https://github.com/espressif/esp-idf/releases/tag/v3.3.2 (we're currently on 3.3.1)?

chilipeppr, 2020-05-22:
For what it's worth, you could try to take my RMT TX library, which I haven't pushed to this repo yet because the docs aren't done yet, and use that to generate your PWM with very fine-grained control. You could just set up your duration0, level0, and duration1, level1 to your 100Hz values and let it loop on its own. Then you can call stop and start in a cleaner way. Sure, you're using the amazing RMT hardware to do basic PWM, but if you weren't using it anyway, might as well.

https://github.com/chilipeppr/robot-actuator-esp32-v8/blob/master/firmware/src/components/modules/rmttx.c

Here's sample Lua code showing how to send info to a ws2812 LED using the module: https://github.com/chilipeppr/robot-actuator-esp32-v8/blob/master/lua/rmttx_ws2812_v2.lua

Here's sample Lua code showing how to send stepper motor signals with accelerate/decelerate using the module:
https://github.com/chilipeppr/robot-actuator-esp32-v8/blob/master/lua/rmttx_stepper_v3.lua

chipec84, 2020-05-22:
@marcelstoer , I updated to v3.3.2 and issue remains the same. (ledcl:reset() not triggering)
@jpeletier , I compiled dimmer.c module, it works like a charm! 

Thank you all for the inputs, the AC dimmer is now working. 

- My initial post was regarding ledc:reset() not triggering. Like @chilipeppr  mentioned, it definitely seems to be coming from the ESP-IDF. I leave it up to you guys if you wish to keep this issue open just to track it down.

marcelstoer, 2020-05-28:
@chipec84 thanks for testing. Can I ask you to help fellow developers by raising this at https://github.com/espressif/esp-idf/issues?

closed 2020-05-28 by marcelstoer