nodemcu/nodemcu-firmware #3125: 🙉NoteMcu nodemcu lua uart.write  ( Control air conditioning failure )

jiaplee, 2020-05-24:
**Control air conditioning**

FE FD 22 11 02 06 02 06 02 02 02 02 02 02 02 02 02 06 02 06 02 06 02 06 02 06 02 02 02 02 02 02 02 02 02 06 02 02 02 02 02 02 02 02 02 02 02 06 02 06 02 06 02 02 02 02 02 02 02 02 02 02 02 02 02 02 02 02 02 02 02 02 02 02 02 02 02 02 02 06 02 06 02 02 02 02 02 02 02 02 02 02 02 02 02 02 02 02 02 02 02 02 02 01 02 02 02 02 02 02 02 02 02 06 02 02 02 02 02 02 02 02 02 02 02 02 02 02 02 02 02 02 02 02 02 02 02 02 02 02 02 02 02 02 02 02 02 02 02 02 02 02 02 02 02 02 02 02 02 02 02 01 02 02 02 02 02 02 02 02 02 02 02 02 02 02 02 02 02 02 02 06 02 02 02 06 02 02 02 02 02 02 02 02 02 02 02 06 02 06 02 06 02 06 02 02 02 02 02 06 02 06 02

**my code  (Control failure)**

`
--uart.setup(1, 115200, 8, uart.PARITY_NONE, uart.STOPBITS_1)
-- uart.setup(0, 115200, 8, uart.PARITY_NONE, uart.STOPBITS_1)
text={}
text[1]=0xFE
text[2]=0xFD
text[3]=0x22
text[4]=0x11
text[5]=0x02
text[6]=0x06
text[7]=0x02
text[8]=0x06
text[9]=0x02
text[10]=0x02
text[11]=0x02
text[12]=0x02
text[13]=0x02
text[14]=0x02
text[15]=0x02
text[16]=0x02
text[17]=0x02
text[18]=0x06
text[19]=0x02
text[20]=0x06
text[21]=0x02
text[22]=0x06
text[23]=0x02
text[24]=0x06
text[25]=0x02
text[26]=0x06
text[27]=0x02
text[28]=0x02
text[29]=0x02
text[30]=0x02
text[31]=0x02
text[32]=0x02
text[33]=0x02
text[34]=0x02
text[35]=0x02
text[36]=0x06
text[37]=0x02
text[38]=0x02
text[39]=0x02
text[40]=0x02
text[41]=0x02
text[42]=0x02
text[43]=0x02
text[44]=0x02
text[45]=0x02
text[46]=0x02
text[47]=0x02
text[48]=0x06
text[49]=0x02
text[50]=0x06
text[51]=0x02
text[52]=0x06
text[53]=0x02
text[54]=0x02
text[55]=0x02
text[56]=0x02
text[57]=0x02
text[58]=0x02
text[59]=0x02
text[60]=0x02
text[61]=0x02
text[62]=0x02
text[63]=0x02
text[64]=0x02
text[65]=0x02
text[66]=0x02
text[67]=0x02
text[68]=0x02
text[69]=0x02
text[70]=0x02
text[71]=0x02
text[72]=0x02
text[73]=0x02
text[74]=0x02
text[75]=0x02
text[76]=0x02
text[77]=0x02
text[78]=0x02
text[79]=0x02
text[80]=0x06
text[81]=0x02
text[82]=0x06
text[83]=0x02
text[84]=0x02
text[85]=0x02
text[86]=0x02
text[87]=0x02
text[88]=0x02
text[89]=0x02
text[90]=0x02
text[91]=0x02
text[92]=0x02
text[93]=0x02
text[94]=0x02
text[95]=0x02
text[96]=0x02
text[97]=0x02
text[98]=0x02
text[99]=0x02
text[100]=0x02
text[101]=0x02
text[102]=0x02
text[103]=0x02
text[104]=0x01
text[105]=0x02
text[106]=0x02
text[107]=0x02
text[108]=0x02
text[109]=0x02
text[110]=0x02
text[111]=0x02
text[112]=0x02
text[113]=0x02
text[114]=0x06
text[115]=0x02
text[116]=0x02
text[117]=0x02
text[118]=0x02
text[119]=0x02
text[120]=0x02
text[121]=0x02
text[122]=0x02
text[123]=0x02
text[124]=0x02
text[125]=0x02
text[126]=0x02
text[127]=0x02
text[128]=0x02
text[129]=0x02
text[130]=0x02
text[131]=0x02
text[132]=0x02
text[133]=0x02
text[134]=0x02
text[135]=0x02
text[136]=0x02
text[137]=0x02
text[138]=0x02
text[139]=0x02
text[140]=0x02
text[141]=0x02
text[142]=0x02
text[143]=0x02
text[144]=0x02
text[145]=0x02
text[146]=0x02
text[147]=0x02
text[148]=0x02
text[149]=0x02
text[150]=0x02
text[151]=0x02
text[152]=0x02
text[153]=0x02
text[154]=0x02
text[155]=0x02
text[156]=0x02
text[157]=0x02
text[158]=0x02
text[159]=0x02
text[160]=0x02
text[161]=0x02
text[162]=0x01
text[163]=0x02
text[164]=0x02
text[165]=0x02
text[166]=0x02
text[167]=0x02
text[168]=0x02
text[169]=0x02
text[170]=0x02
text[171]=0x02
text[172]=0x02
text[173]=0x02
text[174]=0x02
text[175]=0x02
text[176]=0x02
text[177]=0x02
text[178]=0x02
text[179]=0x02
text[180]=0x02
text[181]=0x02
text[182]=0x06
text[183]=0x02
text[184]=0x02
text[185]=0x02
text[186]=0x06
text[187]=0x02
text[188]=0x02
text[189]=0x02
text[190]=0x02
text[191]=0x02
text[192]=0x02
text[193]=0x02
text[194]=0x02
text[195]=0x02
text[196]=0x02
text[197]=0x02
text[198]=0x06
text[199]=0x02
text[200]=0x06
text[201]=0x02
text[202]=0x06
text[203]=0x02
text[204]=0x06
text[205]=0x02
text[206]=0x02
text[207]=0x02
text[208]=0x02
text[209]=0x02
text[210]=0x06
text[211]=0x02
text[212]=0x06
text[213]=0x02

mytimer = tmr.create()
mytimer:register(5000, tmr.ALARM_AUTO, function() 
    uart.setup(0, 115200, 8, uart.PARITY_NONE, uart.STOPBITS_1)
    for i=1,213 do
         uart.write(0, text[i]  )
        -- uart.write(1, text[i]  )
    end
end)
mytimer:start()
`

**  try2  (Control failure)  **  

`
text={}
text[1]=254
text[2]=253
text[3]=34
text[4]=17
text[5]=2
text[6]=6
text[7]=2
text[8]=6
text[9]=2
text[10]=2
text[11]=2
text[12]=2
text[13]=2
text[14]=2
text[15]=2
text[16]=2
text[17]=2
text[18]=6
text[19]=2
text[20]=6
text[21]=2
text[22]=6
text[23]=2
text[24]=6
text[25]=2
text[26]=6
text[27]=2
text[28]=2
text[29]=2
text[30]=2
text[31]=2
text[32]=2
text[33]=2
text[34]=2
text[35]=2
text[36]=6
text[37]=2
text[38]=2
text[39]=2
text[40]=2
text[41]=2
text[42]=2
text[43]=2
text[44]=2
text[45]=2
text[46]=2
text[47]=2
text[48]=6
text[49]=2
text[50]=6
text[51]=2
text[52]=6
text[53]=2
text[54]=2
text[55]=2
text[56]=2
text[57]=2
text[58]=2
text[59]=2
text[60]=2
text[61]=2
text[62]=2
text[63]=2
text[64]=2
text[65]=2
text[66]=2
text[67]=2
text[68]=2
text[69]=2
text[70]=2
text[71]=2
text[72]=2
text[73]=2
text[74]=2
text[75]=2
text[76]=2
text[77]=2
text[78]=2
text[79]=2
text[80]=6
text[81]=2
text[82]=6
text[83]=2
text[84]=2
text[85]=2
text[86]=2
text[87]=2
text[88]=2
text[89]=2
text[90]=2
text[91]=2
text[92]=2
text[93]=2
text[94]=2
text[95]=2
text[96]=2
text[97]=2
text[98]=2
text[99]=2
text[100]=2
text[101]=2
text[102]=2
text[103]=2
text[104]=1
text[105]=2
text[106]=2
text[107]=2
text[108]=2
text[109]=2
text[110]=2
text[111]=2
text[112]=2
text[113]=2
text[114]=6
text[115]=2
text[116]=2
text[117]=2
text[118]=2
text[119]=2
text[120]=2
text[121]=2
text[122]=2
text[123]=2
text[124]=2
text[125]=2
text[126]=2
text[127]=2
text[128]=2
text[129]=2
text[130]=2
text[131]=2
text[132]=2
text[133]=2
text[134]=2
text[135]=2
text[136]=2
text[137]=2
text[138]=2
text[139]=2
text[140]=2
text[141]=2
text[142]=2
text[143]=2
text[144]=2
text[145]=2
text[146]=2
text[147]=2
text[148]=2
text[149]=2
text[150]=2
text[151]=2
text[152]=2
text[153]=2
text[154]=2
text[155]=2
text[156]=2
text[157]=2
text[158]=2
text[159]=2
text[160]=2
text[161]=2
text[162]=1
text[163]=2
text[164]=2
text[165]=2
text[166]=2
text[167]=2
text[168]=2
text[169]=2
text[170]=2
text[171]=2
text[172]=2
text[173]=2
text[174]=2
text[175]=2
text[176]=2
text[177]=2
text[178]=2
text[179]=2
text[180]=2
text[181]=2
text[182]=6
text[183]=2
text[184]=2
text[185]=2
text[186]=6
text[187]=2
text[188]=2
text[189]=2
text[190]=2
text[191]=2
text[192]=2
text[193]=2
text[194]=2
text[195]=2
text[196]=2
text[197]=2
text[198]=6
text[199]=2
text[200]=6
text[201]=2
text[202]=6
text[203]=2
text[204]=6
text[205]=2
text[206]=2
text[207]=2
text[208]=2
text[209]=2
text[210]=6
text[211]=2
text[212]=6
text[213]=2

for i=1,213 do
 uart.write(0, text[i]  )
end

`

**execute scirpt  , Can't control  , But this can be controlled**

[image]

**I connect via USB serial port and I can receive the data**

[image]

nwf, 2020-05-24:
This is not a bug report.  Please try again following the template given, and also try to post *shorter* code segments that exhibit the failure in question.

closed 2020-05-24 by nwf

HHHartmann, 2020-05-25:
Looks like you have the code of a character here. You need to change it to a character itself before sending to serial. 

Please also have a look at #1010 as this seems to be a support question and not a bug in the firmware.

jiaplee, 2020-05-25:
> Looks like you have the code of a character here. You need to change it to a character itself before sending to serial.
> 
> Please also have a look at #1010 as this seems to be a support question and not a bug in the firmware.

This problem has troubled me for a long time, can you help me troubleshoot the problem? I really need your help.