nodemcu/nodemcu-firmware #3131: Bug With (New Div) LFS

chathurangawijetunge, 2020-05-29:
NodeMCU 3.0.0.0 built on nodemcu-build.com provided by frightanic.com
	branch: dev
	commit: 8f85c9b3e4a606f1731051af45c39302799ddb28
	release: 
	release DTS: 202005281716
	SSL: false
	build type: integer
	LFS: 0x40000 bytes total capacity
	modules: file,gpio,net,node,tmr,uart,wifi
 build 2020-05-29 03:54 powered by Lua 5.1.4 on SDK 3.0.1-dev(fce080e)
cannot open init.lua: 

=node.flashreload("scw.img")
LFS region updated.  Restarting.

after LFS loads it keeps rebooting with following (loop)

ets Jan  8 2013,rst cause:2, boot mode:(3,6)
load 0x40100000, len 27444, room 16 
tail 4
chksum 0xa7
load 0x3ffe8000, len 2560, room 4 
tail 12
chksum 0x13
ho 0 tail 12 room 4
load 0x3ffe8a00, len 8, room 12 
tail 8
chksum 0x6c
csum 0x6c

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 27444, room 16 
tail 4
chksum 0xa7
load 0x3ffe8000, len 2560, room 4 
tail 12
chksum 0x13
ho 0 tail 12 room 4
load 0x3ffe8a00, len 8, room 12 
tail 8
chksum 0x6c
csum 0x6c

chathurangawijetunge, 2020-05-31:
On my raspberry pi... immage works fine with master

TerryE, 2020-06-07:
Given the lack of reply, I suspect that this issue has been resolved,so I am closing it.

closed 2020-06-07 by TerryE