nodemcu/nodemcu-firmware #3136: Dev Branch node.output() results in unresponsive device

pnkowl, 2020-05-31:
### Expected behavior
Behaves similarly to master branch (after adjusting from string to pipe)
### Actual behavior
After first call, device is unresponsive 
### Test code (Dev Branch, file attached as mypipe1.lua)
Provide a [Minimal, Complete, and Verifiable example](http://stackoverflow.com/help/mcve) which will reproduce the problem.
```Lua
-- add code here
C={}
C.pipe=1
print("pipe=",C.pipe)
url="temp1.lua"
if file.open(url, "w") then
  file.write('print("temp1.lua was executed")')
  file.write('print("temp1.lua was executed, line 2")')
  file.close()
end

count=0
sen = "<start>\n"

if (C.pipe==0) then
  function s_output(str)
    if (str ~=nil) then
      sen = sen .. "|"..str
    else
      count=count+1
      -- sen=sen.."|" 
    end
  end
end

if (C.pipe==1) then
  function s_output(opipe)
    local temp=opipe:read(1300)
    if (temp~=nil) then
      sen = sen.."|"..temp
    else
      count=count+1
      -- sen=sen.."|" -- results in string overflow
    end
  end
end

node.output(s_output, 0) -- re-direct output to function s_output.

st, result = pcall(dofile, url)

for k,v in pairs(file.list()) do print(k,v) end

-- delay the output capture by 1000 milliseconds to give some time to the user routine in pcall()
ideTmr = tmr.create()
ideTmr:register(2000, tmr.ALARM_SINGLE, function()
  node.output()  -- back to serial
  -- s_output=nil -- no difference
  if result then
    outp = tostring(result):sub(1,1300)
    result = nil
    sen = sen .. outp
  end
  sen = sen .. "<done>\n"
  print("sen="..sen, "count=",count)
  --node.output(nil) -- no difference
end)

ideTmr:start()
print("done,",count)
```
### Response (Dev Branch)
```
> Uploading to ESP file mypipe1.lua...Success

> dofile("mypipe1.lua")
pipe=   1
sen=<start>
|temp1.lua was executed
temp1.lua was executed, line 2
mypipe.lua      856
mypipe1.lua     1318
temp.lua        31
temp1.lua       70
done,   0
> <done>
        count=  12641
=node.heap()
=node.heap()
```
Noteworthy
- no response from "=node.heap()" entered from the terminal
- only way to continue is to invoke an device hard reset (restart unresponsive from terminal)
- 12641 calls to s_output, all but 1 passes nil, which seems rather odd (6K per second)
- only 1 call to s_output which combines all the output: dofile with file.list()
### NodeMCU startup banner (Dev Branch)
You MUST include the firmware startup banner to describe the version you are using. We reserve the right to immediately close any bug that doesn't.
NodeMCU 3.0.0.0 built on nodemcu-build.com provided by frightanic.com
        branch: dev
        commit: 8f85c9b3e4a606f1731051af45c39302799ddb28
        release:
        release DTS: 202005281716
        SSL: false
        build type: float
        LFS: 0x40000 bytes total capacity
        modules: adc,bit,cron,dht,file,gpio,mqtt,net,node,ow,pwm2,rtcmem,rtctime,sntp,softuart,struct,tmr,uart,wifi
 build 2020-05-30 13:33 powered by Lua 5.1.4 on SDK 3.0.1-dev(fce080e)
cannot open init.lua:

### Test code (Master Branch, file attached as mypipe0.lua, works fine)
**The only difference is C.pipe=0 -- from 1**
Provide a [Minimal, Complete, and Verifiable example](http://stackoverflow.com/help/mcve) which will reproduce the problem.
```Lua
-- add code here
C={}
C.pipe=0
print("pipe=",C.pipe)
url="temp1.lua"
if file.open(url, "w") then
  file.write('print("temp1.lua was executed")')
  file.write('print("temp1.lua was executed, line 2")')
  file.close()
end

count=0
sen = "<start>\n"

if (C.pipe==0) then
  function s_output(str)
    if (str ~=nil) then
      sen = sen .. "|"..str
    else
      count=count+1
      -- sen=sen.."|" 
    end
  end
end

if (C.pipe==1) then
  function s_output(opipe)
    local temp=opipe:read(1300)
    if (temp~=nil) then
      sen = sen.."|"..temp
    else
      count=count+1
      -- sen=sen.."|" -- results in string overflow
    end
  end
end

node.output(s_output, 0) -- re-direct output to function s_output.

st, result = pcall(dofile, url)

for k,v in pairs(file.list()) do print(k,v) end

-- delay the output capture by 1000 milliseconds to give some time to the user routine in pcall()
ideTmr = tmr.create()
ideTmr:register(2000, tmr.ALARM_SINGLE, function()
  node.output()  -- back to serial
  -- s_output=nil -- no difference
  if result then
    outp = tostring(result):sub(1,1300)
    result = nil
    sen = sen .. outp
  end
  sen = sen .. "<done>\n"
  print("sen="..sen, "count=",count)
  --node.output(nil) -- no difference
end)

ideTmr:start()
print("done,",count)
```
### Response (Master Branch, works fine)
```
lua: cannot open init.lua
> =node.heap()
43808
> Uploading to ESP file mypipe0.lua...Success

> dofile("mypipe0.lua")
pipe=   0
sen=<start>
|temp1.lua was executed|
|temp1.lua was executed, line 2|
|mypipe1.lua|   |1318|
|mypipe0.lua|   |1318|
|mypipe.lua|    |856|
|temp.lua|      |31|
|temp1.lua|     |70|
|done,| |0|
|> <done>
        count=  0
=node.heap()
42200
> =node.heap()
42200
```
Noteworthy
- =node.heap() working fine
- device responsive
- multiple calls to s_output, one per "field"
### NodeMCU startup banner
You MUST include the firmware startup banner to describe the version you are using. We reserve the right to immediately close any bug that doesn't.
NodeMCU 3.0.0.0 built on nodemcu-build.com provided by frightanic.com
        branch: master
        commit: 3d917850180f67adc7f2c6b5d00f27c152e7194c
        release:
        release DTS: 202002231746
        SSL: false
        build type: float
        LFS: 0x40000
        modules: adc,bit,cron,dht,file,gpio,i2c,mqtt,net,node,ow,pwm2,rtcmem,rtctime,sntp,struct,tmr,uart,wifi
 build 2020-05-30 14:13 powered by Lua 5.1.4 on SDK 3.0.1-dev(fce080e)
lua: cannot open init.lua

### Hardware
Describe which ESP8266/ESP32 device you use and document any special hardware setup
required to reproduce the problem.
Wemos D1 Mini (ESP8266, 12F)
### Notes
1. May be related to #3118 as telnet uses node.output()
2. The code does yield a valid response the first time and then becomes unresponsive even before a second invocation is attempted.
3. I am laying odds that something is likely missing in my code, but in reviewing the new node.output and pipe(), I could find nothing out of place.  If this turns out to be true, perhaps additional helpful tips can be added to the docs.

[200530a2.zip](https://github.com/nodemcu/nodemcu-firmware/files/4707023/200530a2.zip)

TerryE, 2020-05-31:
@pnkowl,

> I am laying odds that something is likely missing in my code, but in reviewing the new node.output and pipe(), I could find nothing out of place.

Yes, you are missing something and yes there's something weird going on.  Let me do a little more exploring to bottom this out.

TerryE, 2020-06-01:
@pnkowl, you need to read the [`pipe.create()`](https://nodemcu.readthedocs.io/en/dev/modules/pipe/#pipecreate) documentation and specifically the description of the `CB_function` parameter:

>  optional reader callback which is called through the `node.task.post()` when the pipe is written to. If the CB returns a boolean, then the reposting action is forced: it is reposted if true and not if false. If the return is nil or omitted then the default is to repost if a pipe write has occurred since the last call.

The pipe functionality works intimately with the tasking system within the SDK, because it must interoperate with the `net` module which is built around this.  By default normal Lua tasks run at medium priority and any prints are added to the `stdout` pipe.  The first of these will post a task to call the CB (also at medium priority).  This CB task should either empty the pipe or if not then either post a follow on task (as is done by the `on:sent()` event when sending over the network) or ask the pipe to do this automatically by setting the appropriate return status.

Here is a minimal example of how this works:
```Lua
do
  chunk,stdout,i,s = {}  -- use globals for interactive access
  
  local function unspool(opipe)
      stdout, s, i = opipe, opipe:read(200), #chunk+1
      if s == nil then return nil end -- only reschedule if another print occurs
      chunk[i] = s
      uart.write(0,("opipe called %u(%u)\n,"):format(i,#s))
      return true -- reschedule to unspool next chunk
    end
  node.output(unspool, 0)
    
  for i=1,20 do print(("Line %3u is this"):format(i)) end
  tmr.create():alarm(500, 0, function()
    node.output() -- reset output
  end)
end
```
Note how the return status is used to reschedule the next pipe read or not.

As far as I can see, this is all working to spec.

The whole of the pipe spooling / unspooling might seem an unnecessary complexity, but it is really necessary. without this it was just far too easy to crash the TCP stack.

TerryE, 2020-06-02:
@pnkowl, I will leave this another day for you to respond, but ATM I see this as a case of application error and I plan to close the issue.

pnkowl, 2020-06-03:
@TerryE . Thanks for adding greater clarity to this.  

I missed the additional complexity needed to make my code work when reviewing the telnet usage example offered in the dev docs for node.output() at https://github.com/nodemcu/nodemcu-firmware/blob/dev/lua_examples/telnet/telnet_pipe.lua

Before dev is PRed to master it may be good to have an example (like above?) and provide notice (a "breaking changes" tag?) that the new node.output() will not work with existing master branch code and what steps are needed for users to modify their code.

I have not had a chance to take your above offering for a spin yet, but have spent the last several days working on another set of suggestions: (1) getting luac.cross working on RPi so that I can create LFS master branch images, (2) installing Win10 on a spare PC, (3) upgrading CMOS for hyperthreading and virtualization, (4) installing Docker,  (5) creating master branch firmware and LFS images with the same feature set as created by CloudBuilder and last months master branch luac.cross web service, (6) brief testing which did not find anything amiss.  Nice.  So progress is being made.  Docs made this possible.  Thanks.

It is fine if you close this.

closed 2020-06-03 by TerryE