nodemcu/nodemcu-firmware #3140: ws2812 driver doesn't wait for UART to drain

nwf, 2020-05-31:
### Expected behavior

`ws2812.write()` should completely send all the data before returning control to Lua, so that subsequent events do not overlap the bitstream.  Alternatively, the documentation should include a suitable delay for `tmr.delay`; the answer appears (to my relatively naive eye) to be a full FIFO (at 128 bytes) being sent at 3200000 baud, which takes 300 microseconds?

### Actual behavior

`ws2812.write()` returns after the tail of the bitstream is queued to the UART FIFO, and does not stick around to wait for the FIFO to drain out the pin.

### Test code

With GPIO 3 AND-gating the ws2812 bitstream (from GPIO 4), the following sequence will not transmit the entirety of `ledfb` before the stream is interrupted:

```Lua
ws2812.init(ws2812.MODE_SINGLE)
gpio.write(3,gpio.HIGH)
ws2812.write(ledfb)
gpio.write(3,gpio.LOW)
```

### Firmware revision

Verified experimentally under, and then again by inspection of, 5135ee441aee86129c37ce78dfca02053000cc2a .

### Additional commentary

Perhaps the ws2812 driver itself should know how to do the GPIO jostling for this kind of AND-gating, as an optional feature.  It's very handy because otherwise the ESP's startup spew out its UARTs gets interpreted by the ws2812 drivers and causes annoying visual artifacts.

Perhaps @Alkorin, being the author of this (very much nicer than bitbanging) UART transmission technique, has opinions?

pjsg, 2020-06-01:
I wouldn't want that wait behavior to be the default.. if you are just
using the ws2812 library then you probably want to get the benefit of the
overlapping... Maybe add a wait option to the write method...

nwf, 2020-06-01:
I'd settle for the overlap being documented, really.  A `sync` parameter also seems like a grand option.

pjsg, 2020-06-23:
Curiously enough, I find myself wanting something very similar with some other enhancements.

My use case is that I need to be able to attach a ws2812 to an esp8266 in such a way that it *does not* illuminate during boot (this is for power consumption reasons). As far as I can tell, this means that either I have to use another gpio to gate D2, or maybe I can switch UART0 to use D8 directly. In the latter case, doing the ws2812.write synchronously would allow me to switch back to the regular TX pin for other output. 

Also, in the case that I have to gate the output signal, I probably want to be able to control whether D2 output is inverted or not. 

Any other thoughts?

Alkorin, 2020-06-23:
I've reflashed an ESP8266 few days ago, so I'm back in business.

I can add a ws2812.wait() which will wait until all buffers are emptied or try to add an optional parameter to ws2812.write(). I will try to take a look this weekend.

nwf, 2020-06-23:
@pjsg  You have discovered, exactly, why I discovered this in the first place; I have a ws2812 array that I gate with an AND gate (a literal 7408 DIP, in fact) and the tail end of the signal was getting lost if I just did `ws2812.write(); gpio.write()`.
 
I was going to expose a `uart.fifodepth([uart],[dir])` call and just poll it from Lua after my `ws2812.write()` call and before doing the `gpio.write()`.  The maximum drain time for `ws2812` is, IIRC, less than 300 microseconds, so hardly worth worrying about, but it'd be nice to spin only as long as needed, nevertheless.

I think `uart.fifodepth()` is more general than `ws2812.wait()`, so perhaps we should just give the one-line loop in `ws2812.md` rather than update `ws2812.c` itself.

nwf, 2020-06-23:
@Alkorin Welcome back!

pjsg, 2020-06-24:
What do you think of the approach of using the alternate uart0 output pin?

nwf, 2020-06-24:
Seems like it could work, just be careful to put it back, yea?  :)  I don't know if it matters, but the alternate pin (pin13, MTDO) can't be pulled high during boot, it seems.

nwf, 2020-09-12:
I realized tonight that the 8266 `ws2812.c` driver has a bug: it should cork the UART or disable IRQs until the TX buffer is full, otherwise there's a chance that we won't have the tens of microseconds it takes to run an ISR equivalent of data queued to the TX FIFO and so will glitch the output waveform.  See #2936.

closed 2021-01-06 by nwf