nodemcu/nodemcu-firmware #3155: Pipe Object Unread String Length Limitation

NicolSpies, 2020-06-11:
### Expected behavior
It is expected that the `pobj:unread(s)` method not be limited by the length of the string to be unread.
### Actual behavior
The  `pobj:unread(s)` method works up to string lengths of 256 characters. With a string lengths longer than 257 characters, the following error is returned: `bad argument #-2 to 'read' (userdata expected, got no value)`. On very long strings, a panic and restart results.

A workaround in not to use  `pobj:unread(s)` but `pobj:write(s)` after a read.
### Test code
The `string` `'ttttttt.....ttttt'` has been shortened in the code sample for readability purposes.
```Lua
  pobj = pipe.create()
  pobj:write('tttttttt.......tttttttttt')

  data1=pobj:read()
  print(data1)

  data2=pobj:read()  
  if data2==nil then print("pobj empty after read") end  
  
  pobj:unread(data1)
  data3=pobj:read()
  print(data3)
```
### NodeMCU startup banner
```
NodeMCU 3.0.0.0 built with Docker provided by frightanic.com
	branch: dev
	commit: 50fcd72a5822b628708e4bd1c8f6ba04815289f3
	release: 2.0.0-master_20170202 +515
	release DTS: 202005151145
	SSL: true
	build type: float
	LFS: 0x30000 bytes total capacity
	modules: adxl345,bit,cron,crypto,encoder,file,gpio,http,i2c,net,node,ow,pipe,rtcmem,rtctime,sjson,sntp,softuart,struct,tls,tmr,uart,wifi
 build 2020-06-02 06:48 powered by Lua 5.1.4 on SDK 3.0.1-dev(fce080e)
```
### Hardware
Standard ESP12F

nwf, 2020-06-12:
The proposed workaround is not semantics-preserving, I think: `write` appends to the end of the pipe, while `unread` prepends to the front.

In general, "unread"-ing from a pipe is dubious, and I *believe* the intention is not to support general prepending as in double-ended queues, but quite specifically to put back a substring of what was just `:read` from the `pipe`, so your example, which reads `data1` and then reads `data2` and then tries to unread `data1` is, I think, outside the intended design.  Is out-of-order `unread` requisite for the bug to manifest?

NicolSpies, 2020-06-12:
@nwf , the issue reported relates to the limitation on the length of the string that can be unread. The test code is only for simple verification to confirm the issue reported.

TerryE, 2020-07-08:
I will fix this post #3193

PR #3295 (2020-10-23): Fix pipe unread (#3155)
Fixes #3155.

Make sure all boxes are checked (add x inside the brackets) when you submit your contribution, remove this sentence before doing so.

- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [x] I have thoroughly tested my contribution.
closed 2020-10-23 by nwf