nodemcu/nodemcu-firmware #3156: LFS under Lua 5.3

vsky279, 2020-06-11:
I'm sorry I don't have time to study the issue myself now. So maybe I am reporting something while the issue is on the other side of the keyboard. I guess this has some easy solution.

### Expected behavior
LFS example working under Lua 5.3

### Actual behavior
```Lua
node.flashindex("_init")()
Lua error: _init.lua:55: attempt to call a nil value (global 'getfenv')
stack traceback:
	_init.lua:55: in main chunk
	stdin:1: in main chunk
	[C]: in ?
	[C]: in ?
```

### Test code
Standard [LFS example](/nodemcu/nodemcu-firmware/tree/master/lua_examples/lfs) loaded in LFS.
```Lua
node.flashindex("_init")()
```
### NodeMCU startup banner
```
NodeMCU 3.0.0.0 built with Docker provided by frightanic.com
	branch: watertank
	commit: 78ab92f24021bb843e611f4daac0124649304f7f
	release: 2.0.0-master_20170202 +523
	release DTS: 202006112149
	SSL: false
	build type: float
	LFS: 0x10000 bytes total capacity
	modules: enduser_setup,file,gpio,http,net,node,rtcmem,rtctime,sjson,sntp,sun,tmr,uart,wifi
 build 2020-06-11 22:06 powered by Lua 5.3.5 on SDK 3.0.1-dev(fce080e)
```

### Hardware
ESP8266 ESP-12E

lmercucci, 2020-06-11:
the same problem, my init.lua calls node.flashindex ("_ init") (). Remove
it in version 5.3 it works, as if it had. In version 5.1 it is mandatory,
still tested both ...

nwf, 2020-06-12:
This is not a LFS issue; Lua 5.2 (and therefore 5.3) removed the `getfenv` (and `setfenv`) functions.  See http://www.lua.org/manual/5.2/manual.html#8 for details.

@TerryE will have to update his `_init.lua` example.

TerryE, 2020-06-12:
Lua has subversion releases, e.g 5.1.x and these are not breaking, but going from Lua 5.1 to 5.3 is itself a breaking change.  I've tried to implement as many of the 5.1 compatibility features as possible but as Nathaniel says the environment architecture was a breaking change on 5.2.

We need really need to go through all examples and modules including the `lfs` ones to mitigate this and try to provide backwards compatibility.  I'll do this in the next day or so. 

In the meantime, as a workaround, you will see that `LFS` is now implemented in the core and so `LFS.modname()`  etc. works without needing to run any _init function.

HHHartmann, 2020-06-12:
Will there also be a solution for loadfile/dofile?
I use them to allow LFS overrides from SPIFFS

TerryE, 2020-06-12:
If you want search lists, why not just stick to `require ()`?
And let me go to sleep! 🥴

vsky279, 2020-06-14:
It seems to me that simple modification should fix the issue:
```Lua
- local G=getfenv()
+ local G=_ENV or getfenv()
  G.LFS = setmetatable(lfs_t,lfs_t)
```
However I don't know what's wrong now as I get immediate restart with `node.flashindex()` (e.g. `node.flashindex("telnet")`.

TerryE, 2020-06-14:
Lukas,  That's because _G already has a metatable that you _don't_ want to overwrite.  See `luaopen_base()` in `app/lua53/lbaselib.c`.  All of the Lua base functions and the modules are in a `ROTable` called `ROM`, which is why you can do a `for k,v in pairs(ROM)`.  The open code effectively does a `setmetatable( _G, {__index=ROM})`.   That's how all all these functions and libraries are accessible through the default environment without taking up entries in a table in RAM.

 `ROM` already includes `LFS` so why try to replace it?

nwf, 2020-06-15:
I suspect @vsky279's objective is a replacement `_init.lua` that works on both 5.1 and 5.3, while the global symbol `LFS` being provided by the firmware itself is (AFAICT) new to 5.3?

TerryE, 2020-06-15:
>  the global symbol `LFS` being provided by the firmware itself is (AFAICT) new to 5.3?

Yes.  This means that new Lua developers can use LFS from the get-go without needing an `_init` module, but if we want to avoid a break here then all we need is some `ìf _VERSION == 'Lua 5.1' then` logic in the module.

vsky279, 2020-06-16:
As @nwf guessed, my goal was to have `_init.lua` compatible with both 5.1 and 5.3 Lua.
It's nice we can use LFS from scratch in 5.3. Though my understanding is that we still need to define `package.loaders[3]` to make `require` work with LFS.
`loadfile` and `dofile` should be redefined too.

So executing whole `_init.lua` but parts mentionned above under `if _VERSION == 'Lua 5.1' then` should do the thing.

See [_init.lua](https://gist.github.com/vsky279/5b7864e02d596286003d38b9dd422c91)

nwf, 2020-06-16:
FWIW, I think

```
package.loaders[3] = function(module) -- loader_flash
  local fn, ba = index(module)
  return ba and "Module not in LFS" or fn
end
```

can just be

```
package.loaders[3] = function(module) return LFS[module] end
```

and similarly for the other uses of `node.flashindex` (aka `index`) in your `_init.lua` mod.

Regardless of that, tho', please do feel free to open a PR.

PR #3168 (2020-09-03): Makes lfs/_init.lua compatible with Lua 5.3
Fixes #3156.

- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [x] I have thoroughly tested my contribution.
- [x] The code changes are reflected in the documentation at `docs/*`.

This commit makes [lua_examples/lfs/_init.lua](/nodemcu/nodemcu-firmware/blob/master/lua_examples/lfs/_init.lua) compatible with Lua 5.3.