nodemcu/nodemcu-firmware #3167: Spurious crashing when hx711 module included

twitchyliquid64, 2020-06-18:
Trying to interact with the flash (ie: uploading, reading file contents) results in the crash below. Even uploader tools are unable to upload file contents.

Able to trigger on `master` but only with `hx711` included as a module.

The issue is not present with `1.5.4.1` (but that version is so old all the network stuff is janky).

Here is a trace from trying to upload an `init.lua` file:

```
**[xxx@xxx]:~/projects/home/sleep> nodemcu-uploader upload init.lua --verify=raw
opening port /dev/ttyUSB0 with 115200 baud
Preparing esp for transfer.
Transferring init.lua as init.lua
Bad chunk response "
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 27880, room 16 
tail 8
chksum 0x3f
load 0x3ffe8000, len 2564, room 0 
tail 4
chksum 0x21
load 0x3ffe8a04, len 8, room 4 
tail 4
chksum 0x5e
csum 0x5e
�2r���N�Gl�,�.>�#
                 �b��.;&�'~b
                            l$
                              �

                               $ 
                                 �fn�nn���`~�
                                             �on���`~�b
                                                       �nn���`��n�r��n|�
                                                                        lll`b
                                                                             brl�nb�n
                                                                                     l`�rl�l�

                                                                                             l`�rl�l�

                                                                                                     l`�rl
                                                                                                          $�

                                                                                                            ll`rl$rl�bb
                                                                                                                       ǌb�bpr
                                                                                                                             �br

                                                                                                                                lb�n'nn
    lbl��
         ll
           ll�
              ��
                l$`�n�bb
                        ll
                          $�
                            b
                             �bb
                                l
                                 �bslr
NodeMCU 3.0.0.0 built on nodemcu-build.com provided by frightanic.com
	branch: master
	commit: 8d091c476edf6ae2977a5f2a74bf5824d07d6183
	release: 3.0-master_20200610
	release DTS: 202006092026
	SSL: false
	build type: integer
	LFS: 0x0 bytes total capacity
	modules: file,gpio,http,hx711,net,node,tmr,uart,wifi
 build 2020-06-17 18:30 powered by Lua 5.1.4 on SDK 3.0.1-dev(fce080e)
init.lua:1: unexpected symbol near 'char(1)'
> " a:20:65:74:73:20:4a:61:6e:20:20:38:20:32:30:31:33:2c:72:73:74:20:63:61:75:73:65:3a:32:2c:20:62:6f:6f:74:20:6d:6f:64:65:3a:28:33:2c:36:29:d:a:d:a:6c:6f:61:64:20:30:78:34:30:31:30:30:30:30:30:2c:20:6c:65:6e:20:32:37:38:38:30:2c:20:72:6f:6f:6d:20:31:36:20:d:a:74:61:69:6c:20:38:d:a:63:68:6b:73:75:6d:20:30:78:33:66:d:a:6c:6f:61:64:20:30:78:33:66:66:65:38:30:30:30:2c:20:6c:65:6e:20:32:35:36:34:2c:20:72:6f:6f:6d:20:30:20:d:a:74:61:69:6c:20:34:d:a:63:68:6b:73:75:6d:20:30:78:32:31:d:a:6c:6f:61:64:20:30:78:33:66:66:65:38:61:30:34:2c:20:6c:65:6e:20:38:2c:20:72:6f:6f:6d:20:34:20:d:a:74:61:69:6c:20:34:d:a:63:68:6b:73:75:6d:20:30:78:35:65:d:a:63:73:75:6d:20:30:78:35:65:d:a:8e:32:72:82:9e:12:92:4e:ec:47:2:12:6c:8c:12:2c:8e:12:2e:3e:1a:23:c:86:62:c4:c2:2e:3b:2:26:c7:12:27:7e:12:62:4:c:6c:24:c:c7:c:c:24:20:3:c:f2:66:6e:2:8c:6e:6e:2:c7:12:90:92:60:2:7e:90:12:13:c:8c:6f:6e:2:c7:12:90:92:60:2:7e:90:12:62:c:8c:6e:6e:2:c7:12:90:92:60:2:8c:1c:2:7:12:82:6e:c7:12:72:90:f2:6e:7c:ec:c:6c:6c:6c:60:2:1c:62:12:12:2:c:62:12:72:12:2:6c:e2:6e:62:82:6e:2:c:6c:60:2:8c:1c:72:12:6c:c4:6c:13:e4:c:c:c:6c:60:2:8c:1c:72:12:6c:c7:6c:12:e4:c:c:c:6c:60:2:8c:1c:72:12:6c:c:24:12:e4:c:c:6c:6c:60:2:72:6c:24:10:72:6c:8c:12:62:62:c:c7:8c:62:8c:62:70:72:c:12:9c:2:62:72:12:2:c:c:6c:12:62:7:f2:6e:27:12:6e:6e:10:2:c:2:6c:2:62:2:6c:8c:4:c4:c:6c:6c:c:6c:6c:ec:c:c4:e4:c:6c:24:60:2:80:2:6e:2:8c:12:62:62:c:6c:6c:c:24:e4:12:c:62:c:2:2:8c:12:62:62:c:6c:c:8c:62:1c:73:6c:72:14:d:a:4e:6f:64:65:4d:43:55:20:33:2e:30:2e:30:2e:30:20:62:75:69:6c:74:20:6f:6e:20:6e:6f:64:65:6d:63:75:2d:62:75:69:6c:64:2e:63:6f:6d:20:70:72:6f:76:69:64:65:64:20:62:79:20:66:72:69:67:68:74:61:6e:69:63:2e:63:6f:6d:d:a:9:62:72:61:6e:63:68:3a:20:6d:61:73:74:65:72:d:a:9:63:6f:6d:6d:69:74:3a:20:38:64:30:39:31:63:34:37:36:65:64:66:36:61:65:32:39:37:37:61:35:66:32:61:37:34:62:66:35:38:32:34:64:30:37:64:36:31:38:33:d:a:9:72:65:6c:65:61:73:65:3a:20:33:2e:30:2d:6d:61:73:74:65:72:5f:32:30:32:30:30:36:31:30:d:a:9:72:65:6c:65:61:73:65:20:44:54:53:3a:20:32:30:32:30:30:36:30:39:32:30:32:36:d:a:9:53:53:4c:3a:20:66:61:6c:73:65:d:a:9:62:75:69:6c:64:20:74:79:70:65:3a:20:69:6e:74:65:67:65:72:d:a:9:4c:46:53:3a:20:30:78:30:20:62:79:74:65:73:20:74:6f:74:61:6c:20:63:61:70:61:63:69:74:79:d:a:9:6d:6f:64:75:6c:65:73:3a:20:66:69:6c:65:2c:67:70:69:6f:2c:68:74:74:70:2c:68:78:37:31:31:2c:6e:65:74:2c:6e:6f:64:65:2c:74:6d:72:2c:75:61:72:74:2c:77:69:66:69:d:a:20:62:75:69:6c:64:20:32:30:32:30:2d:30:36:2d:31:37:20:31:38:3a:33:30:20:70:6f:77:65:72:65:64:20:62:79:20:4c:75:61:20:35:2e:31:2e:34:20:6f:6e:20:53:44:4b:20:33:2e:30:2e:31:2d:64:65:76:28:66:63:65:30:38:30:65:29:d:a:69:6e:69:74:2e:6c:75:61:3a:31:3a:20:75:6e:65:78:70:65:63:74:65:64:20:73:79:6d:62:6f:6c:20:6e:65:61:72:20:27:63:68:61:72:28:31:29:27:d:a:3e:20
Traceback (most recent call last):
  File "/home/xxx/.local/bin/nodemcu-uploader", line 10, in <module>
    sys.exit(main_func())
  File "/home/xxx/.local/lib/python2.7/site-packages/nodemcu_uploader/main.py", line 260, in main_func
    args.restart)
  File "/home/xxx/.local/lib/python2.7/site-packages/nodemcu_uploader/main.py", line 56, in operation_upload
    uploader.write_file(filename, dst, verify)
  File "/home/xxx/.local/lib/python2.7/site-packages/nodemcu_uploader/uploader.py", line 284, in write_file
    raise BadResponseException('Bad chunk response', ACK, resp)
nodemcu_uploader.exceptions.BadResponseException: Bad chunk response expected:`` != actual: `
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 27880, room 16 
tail 8
chksum 0x3f
load 0x3ffe8000, len 2564, room 0 
tail 4
chksum 0x21
load 0x3ffe8a04, len 8, room 4 
tail 4
chksum 0x5e
csum 0x5e
�2r���N�Gl�,�.>�#
                 �b��.;&�'~b
                            l$
                              �

                               $ 
                                 �fn�nn���`~�
                                             �on���`~�b
                                                       �nn���`��n�r��n|�
                                                                        lll`b
                                                                             brl�nb�n
                                                                                     l`�rl�l�

                                                                                             l`�rl�l�

                                                                                                     l`�rl
                                                                                                          $�

                                                                                                            ll`rl$rl�bb
                                                                                                                       ǌb�bpr
                                                                                                                             �br

                                                                                                                                lb�n'nn
    lbl��
         ll
           ll�
              ��
                l$`�n�bb
                        ll
                          $�
                            b
                             �bb
                                l
                                 �bslr
NodeMCU 3.0.0.0 built on nodemcu-build.com provided by frightanic.com
	branch: master
	commit: 8d091c476edf6ae2977a5f2a74bf5824d07d6183
	release: 3.0-master_20200610
	release DTS: 202006092026
	SSL: false
	build type: integer
	LFS: 0x0 bytes total capacity
	modules: file,gpio,http,hx711,net,node,tmr,uart,wifi
 build 2020-06-17 18:30 powered by Lua 5.1.4 on SDK 3.0.1-dev(fce080e)
init.lua:1: unexpected symbol near 'char(1)'
> 
```

I had a quick look through the blame for the hx711 module, but there was no obvious culprit.

nwf, 2020-06-18:
As `hx711` is a hardware driver, it's difficult for the rag-tag team of core developers to test reliably, so regressions are always a possibility.  The only thing that jumps out to me is that this driver creates a Espressif SDK "task" at initialization, so perhaps something's going wrong there.

Could you do some testing of intermediate releases between `1.5.4.1` and `3.0 20200610`?  That might help us narrow things down.

Are you able to build your own firmware images in addition to those from nodemcu-build.com?  If so, could you test disabling `GPIO_INTERRUPT_ENABLE`?

twitchyliquid64, 2020-06-19:
I don't think you need the actual hardware to trigger the crash, just need to have the module compiled in. Let me double-check this point and get back to you.

I am not able to build arbitrary versions unfortunately.

>  it's difficult for the rag-tag team of core developers to test reliably, so regressions are always a possibility.

Don't worry :) :heart: embedded C is wild lol. I'll try and help out as much as I can, but the layout of the SDK and the task system is pretty alien to me

twitchyliquid64, 2020-06-19:
Oh man, I'm not sure what exactly it was but its working now. I setup nodemcu-uploader from head, and I also yeeted 2200uF onto the 3.3V rail to rule out power instability.

With both those things in place, I havent been able to cause more issues.

Really sorry for the confusion!!

closed 2020-06-19 by twitchyliquid64

nwf, 2020-06-19:
No worries!  Glad to hear you got your issue resolved. :)