nodemcu/nodemcu-firmware #3169: string.format("%.0f")

vsky279, 2020-06-19:
### Expected behavior
```
=("%.0f"):format(3.141)
3
```

### Actual behavior
```
=("%.0f"):format(3.141)
3.141000
```

On desktop in Lua 5.1 and 5.3 formatting string `%.0f` works as expected:
```Lua
Lua 5.3.1  Copyright (C) 1994-2015 Lua.org, PUC-Rio
> print(("%.0f"):format(3.141))
3

Lua 5.1.5  Copyright (C) 1994-2012 Lua.org, PUC-Rio
> print(("%.0f"):format(3.141))
3
```
### Test code
as above

### NodeMCU startup banner
```
NodeMCU 3.0.0.0 built with Docker provided by frightanic.com
	branch: dev
	commit: 86f3419835b6ed92265f054cf2271795b0757606
	release: 3.0-master_20200610 +7
	release DTS: 202006182133
	SSL: false
	build type: float
	LFS: 0x10000 bytes total capacity
	modules: enduser_setup,file,gpio,net,node,rtcmem,rtctime,sjson,sntp,tmr,uart,wifi
 build 2020-06-19 11:06 powered by Lua 5.1.4 on SDK 3.0.1-dev(fce080e)
```
Applies also to Lua 5.3.

### Hardware
ESP8266 ESP-01

KT819GM, 2020-06-19:
Had same problem some time ago, did not bothered to check on pc version (thought my limited knowledge was the reason), so wen't with `math.floor` or with 5.3 `//` when doing division. Just checked example on `build 2020-06-19 06:37 powered by Lua 5.3.5 on SDK 3.0.1-dev(fce080e)` and got the same problem.

vsky279, 2020-06-19:
Yep, workaround is quite simple. I think `%f` uses rounding, so suggested solutions would need to be more sophisticated to replicate the functionality. It would be nice to implement it correctly.

TerryE, 2020-06-29:
Having stepped through the code with the debugger, the problem is `app/libc/stdio.c:L766     `if (prec == 0)  prec = 6;` This version does not differentiate between precision not specified which should default to 6 and an explicit `prec=0`.  This is a bug that dates back to the very earliest versions.  Well I've found it.  over to one of you guys to propose and test a fix.

KT819GM, 2020-07-06:
Gosh, started to look at it and it was solved like 15 min before :D Just for the record:

```c
if (prec == 0)
        prec = 6;
```
to

```c
if (!prec)
   {
       prec;
   }
```
Works also. Used this as reference: [Reference](http://ftp.uni-bayreuth.de/math/statlib/macanova/dos/djgpp/doprnt.c)

vsky279, 2020-07-06:
There's is a `++` operator missing in your comment in front of `prec`.
Anyway it seems that this modification applies to the `%g` formatting string
```
/* a precision of 0 is treated as a precision of 1. */
```
As per comment this is probably not the right fix for the `%f`.

TerryE, 2020-07-07:
@KT819GM  @vsky279 You guys have a ping-pong and ring the bell when you've got a solution. :+1: to you both.  Thanks

KT819GM, 2020-07-07:
> There's is a `++` operator missing in your comment in front of `prec`.
> Anyway it seems that this modification applies to the `%g` formatting string
> 
> ```
> /* a precision of 0 is treated as a precision of 1. */
> ```
> 
> As per comment this is probably not the right fix for the `%f`.

Hi, yes, `++` is missing, for a reason - `1.` and `1`. I've tested only one string from your example  
```Lua
for i=0,8 do print(("%."..i.."f"):format(pi)) end
```
and both variants return:
```Lua
3
3.1
3.14
3.142
3.1416
3.14159
3.141593
3.1415927
3.14159274
```
BUT, it does not mean that it's same in all situations and I suspect that you've tested your solution far better than me, so I see it as correct bugfix.

PR #3199 (2020-07-07): Fix of string.format("%.0f")
Fixes #3169.

- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [x] I have thoroughly tested my contribution.
- [x] The code changes are reflected in the documentation at `docs/*`.

This PR fixes the `string.format("%.0f")` to output float number without decimals after decimal point. At this same time it preserves 6 decimals when no format is specified (`%f`).

The default va [... cut]
closed 2020-07-07 by TerryE