nodemcu/nodemcu-firmware #3170: Stripdebug feature

TerryE, 2020-06-19:
### Background 
I added the [LCD patch](../blob/master/docs/lcd.md) back in 2015 when I first started optimising the Lua engine for NodeMCU. The issue that this was addressing was that leaving the debug metadata in compiled code pretty much tripled the RAM required to store a given compiled function. The way to get around this was to use `node.compile()` which stripped out all debug info, but this resulted in error trackback just returning lots of "?"   LCD had three broad strategies:
-  A packed run-length encoding was adopted for line number information and this dropped the overhead for this by roughly 15×.  OK, looking up line numbers became an O(n) operation but this was only done during error tracebacks anyway.
-  A 3-level stripping scheme was introduced: no debug info; just line number info; all debug info.
-  The `node.stripdebug()` allowed you to set a default strip level during initial load.

I have yet to add this to Lua5.3 and am now doing this as part of the tidy-up activities, but since I am standardising the APIs and making this consistent across all version, I wanted to expose what I am doing here for feedback.

### Proposed Functionality

-  The public `lua.h` interface is implemented by `lua_stripdebug(L, level)`.  This pops a value off the Lua stack This can be function or a positive integer (which means the function at that Lua call stack level, e.g. 1 is the parent calling function).  If this is a Lua function not in LFS, then this and any nested functions within it are stripped of debug information to the specified level.  If the value is `nil` then this sets the default strip level if `level` in the range 0 to 2 (-1 means don't change), and the default level is returned.
-  `node.stripdebug([f], level)` is the Lua callable implementation of this, so for example:
```Lua
  node.stripdebug(someFunc, 2)  -- strip all debug info from someFunc
  node.stripdebug(1, 2)  -- strip all but line number info from the function calling this.
  node.stripdebug(1)  -- set the default to strip no debug info on load
  local strip = node.stripdebug() -- a `nil` level means "don't change" so this returns the default strip level
```
-  node levels are 1,2,3 instead of 0,1,2 but that's for historic reasons and not worth a process break.
 
Also the restart default is `strip=2`, that is upvalue and local metadata are removed during load.  This mainly impacts the `debug` interface (which wasn't in our Lua 5.1 implementation anyway).  If developers need this info to retained then they need to include `node.stripdebug(1)` in their startup.

**Q**: do I add `false` / `true` for none and all stripping?

TerryE, 2020-06-19:
I've got all of this "unified API" working `app/lua` moving to `app/lua53` is a job for tomorrow.

nwf, 2020-06-20:
Ah ha!  This explains why Lua 5.3 was consuming more RAM than I expected.  Thank you for bringing LCD over.

What's the rationale / intended use of `node.stripdebug(n, m)` with `m` integer?

> Also the restart default is strip=2, that is upvalue and local metadata are removed during load.

That's C's 2, or Lua's 3?

> Q: do I add false / true

Might be best to avoid `true` and `false` because I think (Lua's) `2` will be more likely to be explicitly selected than `1` or the default `3` and it seems odd to have shorthands for the more rarely used flavors but not the more common one!

TerryE, 2020-06-20:
> Ah ha! This explains why Lua 5.3 was consuming more RAM than I expected. Thank you for bringing LCD over.

The main reasons for this (1) that we are still carrying debug info so _RAM_ based compiled code is a bit larger, and (2) the LuaGC is a bit lazier about collecting used RAM.  However TValues are 8 instead of 12 bytes so certainly LFS-based apps should have more RAM to play with.

I will be improving the LuaGC in this PR, and I've just been tracking down an interesting stripping bug.  The Proto field `sizeupvalues` is used for two purposes:
-  It dimensions the (debug) upvalues array, and
-  It is used to initialise the closure `nupvalues` field, so if the upvalues are stripped then the closure ends up with 0 upvals and and `GETTABUP` instruction which accesses an upval barfs with as segfault.

Fun, fun.

TerryE, 2020-06-20:
OK, the `Proto` has an array of `Upvaldec` each of which is a `TString *` + 2 bytes. When we strip Upvalues, all this does is to set the TString pointers to `NULL`.  We still need the Upvaldec vector.  This will allow the TStrings for the upvalue names to be GCed _if_ they are in the RWstrt and not used elsewhere. It doesn't save any space in the Proto hierarchy itself :unamused:

TerryE, 2020-06-20:
OK, I've got this all strip stuff working, but want to sleep on this

TerryE, 2020-06-21:
BTW, the `sizeupvalues` issue is specific to Lua 5.3 This is because by default the engine takes a lazy approach to GC of `Closure` records (the Lua function variable is a `Closure` which points to a `Proto`)  Each `Proto` has its last associated `Closure` which is not CGed by default.  If a new `Closure` is needed and the Upvalues are the same as this last stale one then it is reused.

This avoids Closure GC + create for patterns such as:
```Lua
for i=1, n do
  function somefunc(a,b) --[[ ... ]] end
  x[i] = somefunc(i,c)
end
```
We don't use this in NodeMCU Lua because (i) this needs a R/W field in the Proto which is a no-no for LFS and (ii) leaving Closures around is a _bad_ idea when you have limited RAM.  It's also pretty questionable IMO since the cost of checking to see if you can reuse the last Proto is more that creating a new one.  The only advantage is the saving in GC overheads.

TerryE, 2020-06-21:
> What's the rationale / intended use of `node.stripdebug(n, m)` with `m` integer?

See the `getfenv()` documentation. `node.stripdebug(2,1)`  will strip the debug from the function that called this one. Possibly a frill to far.

nwf, 2020-06-22:
> We don't use this in NodeMCU Lua because (i) this needs a R/W field in the Proto which is a no-no for LFS and (ii) leaving Closures around is a _bad_ idea when you have limited RAM. It's also pretty [...]

I'm sympathetic to the desire to avoid needlessly thrashing the GC, since that's a lot of work to be doing when we'd rather be running the mutator, but yea, we don't have a lot of space to play.  But... rather than allowing each Proto to cache a Closure, we could have a "global" (well, in `L`) LRU cache or a per-activation-frame pointer of some refcount-0-but-not-yet-free'd Closures that we could try to reuse.  Either could still dodge the alloc+GC in the loop you gave but would leave the Protos themselves RO.

TerryE, 2020-06-22:
@nwf it's great having someone with whom I can have an informed discussion so :+1: for this.  As you may recall 5.3 added an LRU cache for `CString` -> `TString` conversion, and I also folded `ROTable` access into the same LRU cache.  Before this ROTable access was slow as a dog (ROTable accesses could be up to 100× slower than RAM table ones), so slow in fact that doing stuff like `local node,file = node.file` was pretty essential.  Now ROTable access is still slower but maybe only 30-50% slower and this is in the noise.  These two caches make a big difference to performance.

When it comes to this mutator style optimisation, this was only added in 5.3 and reuse has a per-call _increased_ setup cost.  The trade off is that _if_ you do manage to reuse, then you can often avoid the GC of the old `LClosure` record. The current standard implementation is quite simple: each `Proto` has a R/W field containing the last `LClosure` created. This get cleared if the closure is GCed and if it hasn't yet been, then it is reused or discarded depending on whether the upvalues are the same or not.  This can work quite well in a large RAM model as the GC can afford to be quite laggy.  But for a tight RAM situation like in the NodeMCU case:
-  LFS is a far greater enabler so as you say we would need to move this caching off `Proto`.
-  We couldn't easily fold this into the string / ROTable cache so we would need another and this could start to get quite complicated.
-  Our GC strategy has to be more aggressive because we have so little RAM to play with, so the chances of the closure still being around are less. 

So I am not sure is we would get any material performance increase even if we did this, and i think that we need to be aware of the effect of diminishing returns here: this isn't going to be a case where it is worth spending a lot off effort, IMO.  One possible approach would be to cache the last Proto/Closure created in G(L) and check for reuse of that as this would be reasonable fast and small variation on the standard code.  But in terms of improving general runtime, I think this is going to have very small effect so I feel that we should focus on other low hanging fruit such as #3175.

nwf, 2020-06-22:
I agree that premature optimization is the root of a great deal of evil, so I suggest we just drop the Closure caching until we have some profiles that show us we should do something about it.

TerryE, 2020-07-04:
I feel that PR #3193 fully addresses the above discussion.  

However, we do have one outstanding issue which relates to setting the stripping options in `luac.cross` but I feel that this is best discussed in a `luac.cross` issue

closed 2020-07-04 by TerryE