nodemcu/nodemcu-firmware #3173: ESP32: Bump ESP-IDF to 3.3.2 LTS

marcelstoer, 2020-06-20:
We should adopt the latest fixes in the ESP-IDF LTS branch: 3.3.2.

I prepared a branch at https://github.com/nodemcu/nodemcu-firmware/commits/chore/bump-esp-idf but it appears we also need to adjust code. The [CI build broke](https://travis-ci.org/github/nodemcu/nodemcu-firmware/builds/700442610) with a compile error:

```
/home/travis/build/nodemcu/nodemcu-firmware/components/modules/net.c:981:22: error: invalid initializer

   ip_addr_t ipaddr = dns_getserver(numdns);
```

marcelstoer, 2020-06-25:
@landswipe  over at  #2779 you stated that you had successfully bumped ESP-IDF locally. Were you not getting the error I reported above?

thornley-david, 2020-07-03:
@marcelstoer yes I did, on my side this was fixed by changing: -

ip_addr_t ipaddr = dns_getserver(numdns);

to

const ip_addr_t *ipaddr = dns_getserver(numdns);

and adjusting the usage references to the ipaddr variable as pointer.

marcelstoer, 2020-07-03:
Oh, right. If you happen to have that ready would you mind creating a PR against our `chore/bump-esp-idf` branch?

thornley-david, 2020-07-04:
Submitted the fix as requested, sorry it took a while; I had to setup my whole development environment again after a wayward attempt as using WSL1/2.

PR #3195 (2020-07-04): Fixed esp-lwip breaking change on dns_getserver
Fixes #3173

The dns_getserver function's return value was changed to a const pointer in commit https://github.com/espressif/esp-lwip/commit/da2740fa8d56b9b9e8a10602f38df1ea4dbd9b74 on 2.0.3-esp branch.
PR #3196 (2020-07-08): Bump ESP-IDF to 3.3.2 LTS
Fixes #3173.

- [x] This PR is for the `dev-esp32` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [ ] I have thoroughly tested my contribution.
- [ ] The code changes are reflected in the documentation at `docs/*`.

Thank you @thornley-david for the support.
closed 2020-07-08 by marcelstoer